conda-forge/gazebo-feedstock #31: On Linux gzserver does not start as RenderSystem_GL Ogre plugin is not found

traversaro, 2020-10-04:
Using the latest miniforge ( Miniforge3-4.8.5-1) on Ubuntu 20.04 (via `WSL2`) and installing the `gazebo` package, starting `gzserver --verbose` fails with:
~~~
(base) traversaro@IITICUBLAP102:~$ gzserver --verbose
Gazebo multi-robot simulator, version 11.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.17.163.171
[Err] [RenderEngine.cc:455] Unable to load Ogre Plugin[/home/traversaro/miniforge3/lib/OGRE/RenderSystem_GL]. Rendering will not be possible.Make sure you have installed OGRE and Gazebo properly.
[Err] [RenderEngine.cc:681] EXCEPTION: unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.

[Err] [RenderingIface.cc:39] Failed to load the Rendering engine subsystem
unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.
[Err] [gazebo.cc:81] Unable to load sensors
[Err] [Server.cc:287] Unable to load gazebo
~~~

The problem is similar to the one discussed in https://github.com/conda-forge/ogre-feedstock/issues/15, but I guess that the Gazebo package forces the system to install the old ogre 1.10.12, while the problem was solved in https://github.com/conda-forge/ogre-feedstock/pull/16 for Ogre 1.12.5 . 

<br/>
Environment (<code>conda list</code>):
<details>

```
$ conda list
(base) traversaro@IITICUBLAP102:~$ conda list
# packages in environment at /home/traversaro/miniforge3:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
ace                       6.5.11               h84b9e52_1    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
asio                      1.14.1               he1b5a44_0    conda-forge
binutils                  2.35                 h2122c62_9    conda-forge
binutils_impl_linux-64    2.35                 h18a2f87_9    conda-forge
binutils_linux-64         2.35                heab0d09_28    conda-forge
boost                     1.74.0           py38hd103949_0    conda-forge
boost-cpp                 1.74.0               h9359b55_0    conda-forge
brotlipy                  0.7.0           py38h1e0a361_1000    conda-forge
bzip2                     1.0.8                h516909a_3    conda-forge
c-ares                    1.16.1               h516909a_3    conda-forge
c-compiler                1.1.1                h516909a_0    conda-forge
ca-certificates           2020.6.20            hecda079_0    conda-forge
cairo                     1.16.0            h3fc0475_1005    conda-forge
certifi                   2020.6.20        py38h32f6830_0    conda-forge
cffi                      1.14.3           py38h5bae8af_0    conda-forge
chardet                   3.0.4           py38h32f6830_1007    conda-forge
cmake                     3.18.2               h5c55442_0    conda-forge
conda                     4.8.5            py38h32f6830_1    conda-forge
conda-package-handling    1.7.0            py38h1e0a361_5    conda-forge
console_bridge            1.0.1                hc9558a2_0    conda-forge
cppzmq                    4.6.0                hc9558a2_1    conda-forge
cryptography              3.1              py38h766eaa4_0    conda-forge
curl                      7.71.1               he644dc0_8    conda-forge
cxx-compiler              1.1.1                hc9558a2_0    conda-forge
dbus                      1.13.6               he372182_0    conda-forge
eigen                     3.3.7             hc9558a2_1001    conda-forge
expat                     2.2.9                he1b5a44_2    conda-forge
ffmpeg                    4.3.1                h167e202_0    conda-forge
fontconfig                2.13.1            h1056068_1002    conda-forge
freeglut                  3.0.0             hf484d3e_1005    conda-forge
freeimage                 3.18.0               h775acd0_3    conda-forge
freetype                  2.10.2               he06d7ca_0    conda-forge
gazebo                    11.0.0              h32bde7a_11    conda-forge
gcc_impl_linux-64         7.5.0               hd9e1a51_17    conda-forge
gcc_linux-64              7.5.0               hf34d7eb_28    conda-forge
[... 165 lines cut ...]

     active environment : base
    active env location : /home/traversaro/miniforge3
            shell level : 1
       user config file : /home/traversaro/.condarc
 populated config files : /home/traversaro/miniforge3/.condarc
                          /home/traversaro/.condarc
          conda version : 4.8.5
    conda-build version : not installed
         python version : 3.8.5.final.0
       virtual packages : __glibc=2.31
       base environment : /home/traversaro/miniforge3  (writable)
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/traversaro/miniforge3/pkgs
                          /home/traversaro/.conda/pkgs
       envs directories : /home/traversaro/miniforge3/envs
                          /home/traversaro/.conda/envs
               platform : linux-64
             user-agent : conda/4.8.5 requests/2.24.0 CPython/3.8.5 Linux/4.19.104-microsoft-standard ubuntu/20.04 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```
</details>

wolfv, 2020-10-04:
We do have a rebuild of ogre 1.10 that properly enables the gl backend. Maybe we should move the previous versions to broken so that they cannot be installed.

BTW we're close to have Moveit, cartographer, OpenSlam and desktop-full on OS X, Windows and Linux (including arm64) and we will make sure that gazebo works fine before publishing a blog post, soon!

Tobias-Fischer, 2020-11-07:
Is that still an issue with the most recent ogre 1.10 build?

wolfv, 2020-11-07:
This works fine when sourcing share/gazebo-9/setup.sh 

That file sets some ogre_plugin_directory variable which seems to be necessary

Tobias-Fischer, 2020-11-30:
Hi @traversaro, could you please check whether this is still an issue with ` win-64/ogre-1.10.12-h71cedee_6`? I think we can close this, this particular issue should be fixed with that build.

Tobias-Fischer, 2020-11-30:
Actually, it's on WSL so `ogre-1.10.12-hb844067_6` should do the job

Tobias-Fischer, 2020-11-30:
Just tried this, the error still occurs. Oddly enough the file exists (however .so is missing, not sure whether that's a problem). I noticed that some fixes were not pulled in on unix - see https://github.com/conda-forge/gazebo-feedstock/pull/41 which could be a potential fix for this issue.

Tobias-Fischer, 2020-11-30:
Actually it works fine now - I must not have sourced the environment properly. Could you please confirm @traversaro?

closed 2020-11-30 by Tobias-Fischer

traversaro, 2020-11-30:
> 
> 
> Actually it works fine now - I must not have sourced the environment properly. Could you please confirm @traversaro?

I am on WSL2 and now I have a different error related to the X server (even if I have it on the Windows side and I should have it properly configured), so I guess the RenderSystem_GL problem was indeed solved!