conda-forge/gazebo-feedstock #34: We have to set GAZEBO_RESOURCE_PATH for gazebo 9

wolfv, 2020-11-05:
Gazebo starts when I do:

export GAZEBO_RESOURCE_PATH=/home/wolfv/miniconda3/envs/ros/share/gazebo-9

but fails (with logs using `--verbose`)

wolfv, 2020-11-05:
Argh, now I have this issue: [Wrn] [GuiIface.cc:119] JPEG parameter struct mismatch: library thinks size is 632, caller expects 664

Which I also had before and solved somehow ... I think that was related to freeimage.

wolfv, 2020-11-05:
found the old gitter chat regarding the jpeg issue: https://gitter.im/conda-forge/conda-forge.github.io?at=5e628cb14599eb28644c483f

I have pushed the necessary changes to freeimage (linking to cf libs instead of their own). I think we need to figure out why we're not getting the latest freeimage here.

wolfv, 2020-11-05:
Unfortunately there appears to be a conflict with Ogre and the fixed freeimage:

`mamba install freeimage=3.18.0=*_3 ogre=1.10.12=h6508522_5`

fails with 

Problem: package ogre-1.10.12-h6508522_5 requires openexr >=2.4.1,<2.5.0a0, but none of the providers can be installed

So it seems like we need to rebuild ogre with the new openexr ... or freeimage.

wolfv, 2020-11-05:
Yeah so ogre=1.10.12=h6508522_5 depends on openexr=2.4.1 and freeimage is on openexr=2.5.something. 

We need to push a new build of ogre 1.10.12 with the latest pinnings.

traversaro, 2020-11-06:
`GAZEBO_RESOURCE_PATH` should be set by the `setup.sh` script of Gazebo, if I am not wrong.

wolfv, 2020-11-06:
@traversaro, thanks! when activating a conda environment, certain activation scripts are ran (scripts in `$PREFIX/etc/conda/activate.d` (and vice-versa `$PREFIX/etc/conda/deactivate.d` for deactivation). It seems like we haven't added a script like that for gazebo yet.

wolfv, 2020-11-06:
What I want to say is that the script for gazebo should ideally execute `source $CONDA_PREFIX/share/gazebo-9/setup.sh` or similar.

wolfv, 2020-11-06:
after sourcing gazebo starts, and OGRE works as well! :tada:
I did have to manually upgrade to freeimage=3.18.0=*_3 though

traversaro, 2020-11-06:
Yes

> 
> 
> What I want to say is that the script for gazebo should ideally execute `source $CONDA_PREFIX/share/gazebo-9/setup.sh` or similar.

Yes, I thought indeed that it was already done, but apparently it is not the case. Note that Gazebo's upstream setup.sh and setup.bat are not relocatable, but probably this could be handled by conda tricks to make non-relocatable scripts relocatable. Also, the `setup.bat` script may need to be adapted to where conda install stuff, for my existing Gazebo builds I mantain a separate setup.sh script at https://github.com/robotology/robotology-superbuild-dependencies-vcpkg/blob/master/scripts/setup-gazebo.sh .

wolfv, 2020-11-06:
the setup.sh in this gazebo package automatically has paths replaced, so the relocatability on Linux looks good enough!
We've got to check windows, though.

traversaro, 2020-11-06:
> > What I want to say is that the script for gazebo should ideally execute `source $CONDA_PREFIX/share/gazebo-9/setup.sh` or similar.
> 
> Yes, I thought indeed that it was already done, but apparently it is not the case.

Unless I am missing something the script should be indeed be sourced, see:
* https://github.com/conda-forge/gazebo-feedstock/blob/master/recipe/activate.sh
* https://github.com/conda-forge/gazebo-feedstock/blob/master/recipe/activate.bat

wolfv, 2020-11-06:
yeah, not for 9.x :(
https://github.com/conda-forge/gazebo-feedstock/blob/v9.x/recipe/build.sh

traversaro, 2020-11-06:
I see, got it.

Tobias-Fischer, 2020-11-07:
Is there a reason to stick to Gazebo9 now that we focus mainly on Noetic? If we switch to G11, we could move this issue down in the priority list ..

Tobias-Fischer, 2020-11-07:
Or is the fix to simply copy over the activation scripts from G11?

wolfv, 2020-11-07:
Yeah i think the fix is to copy the stuff from 11. I can work on this next week

PeterMitrano, 2020-11-29:
I'm confused why we need these setups scripts to be sourced. From a standard gazebo install via apt, I can run gazebo without sourcing these, and afaik gazebo itself does not read these files. So why do we need to source them?

traversaro, 2020-11-29:
> From a standard gazebo install via apt, I can run gazebo without sourcing these, and afaik gazebo itself does not read these files.

If the env variable is not defined, Gazebo binaries hardcode the default values of most of this variables directly in their source code to the appropriate location in the system in which Gazebo was built (see https://github.com/osrf/gazebo/blob/6fd426b3949c4ca73fa126cde68f5cc4a59522eb/gazebo/common/SystemPaths.cc#L230). 
When Gazebo is built with apt, hard-coding location that are in `/usr` will work on any system, but when building on conda, that location will not be the same in different environments, and especially it will be different between the build environment and the enthronement in which Gazebo is installed. Conda does a lot of "magic" to fix this non-relocatable parts of binaries (see https://docs.conda.io/projects/conda-build/en/latest/resources/make-relocatable.html), but I don't think this is done at the moment for Gazebo. For this reason, in conda at the moment it is necessary to call this scripts. 

Note however that sourcing those scripts is required by the Gazebo docs itself (see http://gazebosim.org/tutorials?tut=components&cat=get_started) if you want to append any value to those environment variables: 
> If you want to modify Gazebo's behavior, e.g., by extending the path it searches for models, you should first source the shell script listed above, then modify the variables that it sets.

So I would say that it is a good idea to always source them. This is also what is done by `gazebo_ros`  helper scripts: https://github.com/ros-simulation/gazebo_ros_pkgs/blob/dd51b73ccc54a54ae41dd5c4901252a49ca51177/gazebo_ros/scripts/gazebo#L30 .

Tobias-Fischer, 2020-11-30:
Fixed by #40

PR #40 (2020-11-30): Activate scripts for gazebo 9

Checklist
* [x] Used a [personal fork of the feedstock to propose changes](https://conda-forge.org/docs/maintainer/updating_pkgs.html#forking-and-pull-requests)
* [x] Bumped the build number (if the version is unchanged)
* [ ] Reset the build number to `0` (if the version changed)
* [x] [Re-rendered]( https://conda-forge.org/docs/maintainer/updating_pkgs.html#rerendering-feedstocks ) with the latest `conda-smithy` (Use the phrase <code>@<space/>conda-forge-admin, please rerender</code> in a comment in this PR for automated rerendering)
* [ ] Ensured the license f [... cut]
closed 2020-11-30 by Tobias-Fischer