conda-forge/gazebo-feedstock #45: On Windows, can't insert online models.

danzimmerman, 2020-12-30:
#### Issue: 
Can't insert models from online sources into Gazebo 11.3 binary conda install on Windows.

Installed Gazebo binary by creating fresh `gz-simple` environment from YAML file [gz-simple.yml.txt](https://github.com/conda-forge/gazebo-feedstock/files/5755301/gz-simple.yml.txt)

(This environment was created with `mamba 0.76` if it matters, see https://github.com/osrf/gazebo/issues/2899, but same issues were present with working `conda` installs as detailed there)

#### Expected Behavior
I can insert models into the empty world from the `Insert` tab from [fuel.ignitionrobotics.org](https://fuel.ignitionrobotics.org) or [models.gazebosim.org](http://models.gazebosim.org) and they'll show up in the GUI.

#### Actual Behavior
I get the following errors in the terminal window and no model in the GUI:

[fuel.ignitionrobotics.org](https://fuel.ignitionrobotics.org) -> OpenRobotics -> Hospital Robot (or any other) results in:

```
[Err] [..\gazebo\common\FuelModelDatabase.cc:139] Missing model.config for model
```

[models.gazebosim.org](http://models.gazebosim.org) -> Universal Robotics UR10 robot arm (or any other) results in:

```
[Err] [..\gazebo\common\ModelDatabase.cc:607] Missing model.config for model ""
[Err] [..\gazebo\common\ModelDatabase.cc:671] Invalid model manifest file[""]
``` 

<br/>
Environment (<code>conda list</code>):
<details>

```
$ conda list

# packages in environment at C:\Users\<username>\miniconda3\envs\gz-simple:
#
# Name                    Version                   Build  Channel
boost-cpp                 1.72.0               h54f0996_4    conda-forge
bzip2                     1.0.8                he774522_0  
ca-certificates           2020.12.8            haa95532_0  
certifi                   2020.12.5        py38haa95532_0  
console_bridge            1.0.1                h7ef1ec2_0    conda-forge
cppzmq                    4.7.1                h276ee43_0    conda-forge
curl                      7.71.1               h2a8f88b_1  
dlfcn-win32               1.2.0                he025d50_0    conda-forge
eigen                     3.3.9                h2d74725_1    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
freeimage                 3.18.0               h33f27b4_0    conda-forge
freetype                  2.10.4               hd328e21_0  
gazebo                    11.3.0               hc4486a8_3    conda-forge
gettext                   0.19.8.1          h1a89ca6_1005    conda-forge
gts                       0.7.6                h39c6fae_2    conda-forge
icu                       68.1                 h0e60522_0    conda-forge
ilmbase                   2.5.3                h12d4b20_0    conda-forge
jpeg                      9d                   he774522_0    conda-forge
jsoncpp                   1.8.4                h74a9793_0  
krb5                      1.18.2               hc04afaa_0  
libclang                  11.0.0          default_h5c34c98_2    conda-forge
libcurl                   7.71.1               h2a8f88b_1  
libffi                    3.3                  h0e60522_2    conda-forge
libglib                   2.66.4               hcfe831f_1    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libignition-cmake2        2.6.1                h39d44d4_0    conda-forge
libignition-common3       3.9.0                h92846f5_1    conda-forge
libignition-fuel-tools4   4.2.1                h961be0a_1    conda-forge
libignition-math6         6.7.0                h13ae965_0    conda-forge
libignition-msgs5         5.3.0                h13ae965_2    conda-forge
libignition-tools1        1.0.0                h57928b3_3    conda-forge
libignition-transport8    8.1.0                h2130529_2    conda-forge
libpng                    1.6.37               h2a8f88b_0  
libprotobuf               3.13.0.1             h200bbdf_0  
libsdformat               9.3.0                hbeba28f_2    conda-forge
libsodium                 1.0.18               h62dcd97_0  
libssh2                   1.9.0                h7a1dbc1_1  
libzip                    1.7.3                hfed4ece_0    conda-forge
lz4-c                     1.9.3                h8ffe710_0    conda-forge
ogre                      1.10.12              h71cedee_6    conda-forge
openexr                   2.5.3                hab3b255_1    conda-forge
openssl                   1.1.1i               h2bbff1b_0  
pcre                      8.44                 ha925a31_0  
pip                       20.3.3           py38haa95532_0  
protobuf                  3.13.0.1         py38ha925a31_1  
pugixml                   1.10                 ha925a31_1    conda-forge
python                    3.8.3                he1778fa_2  
qt                        5.12.9               h5909a2a_2    conda-forge
qwt                       6.1.5                h265ddaf_0    conda-forge
ruby                      2.7.2                h8b1b97a_3    conda-forge
sdl2                      2.0.12               h0e60522_1    conda-forge
setuptools                51.0.0           py38haa95532_2  
six                       1.15.0           py38haa95532_0  
sqlite                    3.34.0               h8ffe710_0    conda-forge
tbb                       2020.3               h74a9793_0  
tbb-devel                 2020.3               h74a9793_0  
tiny-process-library      2.0.4                h0e60522_0    conda-forge
tinyxml                   2.6.2                h7ef1ec2_1    conda-forge
tinyxml2                  8.0.0                he025d50_1    conda-forge
urdfdom                   2.3.3                h7ef1ec2_0    conda-forge
urdfdom_headers           1.0.5                h7ef1ec2_2    conda-forge
vc                        14.2                 h21ff451_1  
vs2015_runtime            14.28.29325          h5e1d092_0    conda-forge
wheel                     0.36.2             pyhd3eb1b0_0  
wincertstore              0.2                      py38_0  
xz                        5.2.5                h62dcd97_0  
yaml                      0.2.5                he774522_0  
zeromq                    4.3.2                ha925a31_3  
zlib                      1.2.11               h62dcd97_4  
zstd                      1.4.5                ha9fde0e_0  
zziplib                   0.13.69              ha81a0f5_1    conda-forge

```
</details>

<br/>
Details about  <code>conda</code> and system ( <code>conda info</code> ):
<details>

```
$ conda info

     active environment : gz-simple
    active env location : C:\Users\<username>\miniconda3\envs\gz-simple
            shell level : 2
       user config file : C:\Users\<username>\.condarc
 populated config files : 
          conda version : 4.9.2
    conda-build version : not installed
         python version : 3.8.3.final.0
       virtual packages : __cuda=11.2=0
                          __win=0=0
                          __archspec=1=x86_64
       base environment : C:\Users\<username>\miniconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\<username>\miniconda3\pkgs
                          C:\Users\<username>\.conda\pkgs
                          C:\Users\<username>\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\<username>\miniconda3\envs
                          C:\Users\<username>\.conda\envs
                          C:\Users\<username>\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.9.2 requests/2.25.1 CPython/3.8.3 Windows/10 Windows/10.0.20279
          administrator : False
             netrc file : None
           offline mode : False

```
</details>

danzimmerman, 2020-12-30:
To confirm this is a Windows issue, I also tried this on Linux Bionic (18.04.5 LTS) with `miniconda3`, the same virtual environment, etc, and it works properly.

traversaro, 2021-02-25:
According to https://github.com/osrf/gazebo/issues/2899#issuecomment-785682234 the issue has been confirmed also by @FirefoxMetzger .

FirefoxMetzger, 2021-02-25:
Here's my environment info:

<details>
<summary>conda list</summary>

```
# packages in environment at C:\ProgramData\Miniconda3:
#
# Name                    Version                   Build  Channel
boost-cpp                 1.74.0               h54f0996_2    conda-forge
brotlipy                  0.7.0           py38h2bbff1b_1003
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2020.12.5            h5b45459_0    conda-forge
certifi                   2020.12.5        py38haa244fe_1    conda-forge
cffi                      1.14.3           py38hcd4344a_2
chardet                   3.0.4           py38haa95532_1003
conda                     4.9.2            py38haa244fe_0    conda-forge
conda-package-handling    1.7.2            py38h76e460a_0
console_bridge            1.0.1                h5362a0b_0    conda-forge
console_shortcut          0.1.1                         4
cppzmq                    4.7.1                h50e1b43_1    conda-forge
cryptography              3.2.1            py38hcd4344a_1
curl                      7.71.1               h4b64cdc_8    conda-forge
dlfcn-win32               1.2.0                h39d44d4_0    conda-forge
eigen                     3.3.9                h2d74725_1    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
freeimage                 3.18.0               h33f27b4_0    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
gazebo                    11.3.0               h83ca5bb_7    conda-forge
gettext                   0.19.8.1          h1a89ca6_1005    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
icu                       67.1                 h33f27b4_0    conda-forge
idna                      2.10                       py_0
ilmbase                   2.5.5                h12d4b20_0    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
jsoncpp                   1.9.4                h2d74725_1    conda-forge
krb5                      1.17.2               hbae68bd_0    conda-forge
libclang                  10.0.1          default_hf44288c_1    conda-forge
libcurl                   7.71.1               h4b64cdc_8    conda-forge
libffi                    3.3                  h0e60522_2    conda-forge
libglib                   2.66.7               h8c8e623_0    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libignition-cmake2        2.6.2                h0e60522_1    conda-forge
libignition-common3       3.10.1               he3878d5_0    conda-forge
libignition-fuel-tools4   4.3.0                h4936f55_2    conda-forge
libignition-math6         6.7.0                h0e60522_1    conda-forge
libignition-msgs5         5.3.0                hf5fa1a2_6    conda-forge
libignition-tools1        1.0.0                h57928b3_3    conda-forge
libignition-transport8    8.1.0                h7435cc3_4    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libprotobuf               3.15.2               h7755175_0    conda-forge
libsdformat               9.3.0                h4bf3a07_2    conda-forge
libsodium                 1.0.18               h8d14728_1    conda-forge
libssh2                   1.9.0                hb06d900_5    conda-forge
libzip                    1.7.3                hfed4ece_0    conda-forge
lz4-c                     1.9.3                h8ffe710_0    conda-forge
menuinst                  1.4.16           py38he774522_1
ogre                      1.10.12              hadbb816_7    conda-forge
openexr                   2.5.5                hab3b255_0    conda-forge
openssl                   1.1.1j               h8ffe710_0    conda-forge
pcre                      8.44                 ha925a31_0    conda-forge
pip                       20.2.4           py38haa95532_0
powershell_shortcut       0.0.1                         3
protobuf                  3.15.2           py38h885f38d_0    conda-forge
pugixml                   1.11.4               h0e60522_0    conda-forge
pycosat                   0.6.3            py38h2bbff1b_0
pycparser                 2.20                       py_2
pyopenssl                 19.1.0             pyhd3eb1b0_1
pysocks                   1.7.1            py38haa95532_0
python                    3.8.5                h5fd99cc_1
[... 45 lines cut ...]
    conda-build version : not installed
         python version : 3.8.5.final.0
       virtual packages : __cuda=11.2=0
                          __win=0=0
                          __archspec=1=x86_64
       base environment : C:\ProgramData\Miniconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\ProgramData\Miniconda3\pkgs
                          C:\Users\Sebastian\.conda\pkgs
                          C:\Users\Sebastian\AppData\Local\conda\conda\pkgs
       envs directories : C:\ProgramData\Miniconda3\envs
                          C:\Users\Sebastian\.conda\envs
                          C:\Users\Sebastian\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.9.2 requests/2.24.0 CPython/3.8.5 Windows/10 Windows/10.0.19041
          administrator : True
             netrc file : None
           offline mode : False
```
</details>

traversaro, 2021-04-30:
I did not checked appropriately back in time the @danzimmerman comment https://github.com/osrf/gazebo/issues/2899#issuecomment-752703532 , but indeed the problem seems a duplicate of https://github.com/osrf/gazebo/issues/2611 . I commented there (https://github.com/osrf/gazebo/issues/2611#issuecomment-829941224) on the solution that is quite simple, so if anyone is interested feel free to either add a patch here, or provide a PR upstream in Gazebo, thanks!

Tobias-Fischer, 2021-05-03:
Hi @danzimmerman @FirefoxMetzger - just merged #68 which hopefully should fix this issue. Let us know if not :)

closed 2021-05-10 by traversaro

traversaro, 2021-05-10:
Actually I think that Fuel support was added in Gazebo 11, so my question does not make a lot of sense now that I think about it.

traversaro, 2021-05-10:
I wonder if that problem will be fixed by https://github.com/ignitionrobotics/ign-fuel-tools/pull/178 (fyi @ahcorde).

ahcorde, 2021-05-10:
@traversaro,

I think this problem is fixed in the PR that you mentioned, but it was a regression on [ign-common](https://github.com/ignitionrobotics/ign-fuel-tools/issues/183). I need to re-work this part again.

commit a192205c (2021-05-03): Merge pull request #68 from Tobias-Fischer/2789

Fix #45 (Windows model insert)