conda-forge/gazebo-feedstock #57: Pin tbb-devel to 2020.3

traversaro, 2021-03-03:
I was testing a custom build of Gazebo on conda-forge (see https://github.com/traversaro/gazebo-conda-forge-ci/pull/2) and just I was testing I noticed that tbb 2021.1 was released ( https://github.com/conda-forge/tbb-feedstock/pull/78 ) and Gazebo it is not compatible with it (see https://github.com/osrf/gazebo/issues/2867).

Tobias-Fischer, 2022-02-04:
This is now causing trouble because we can't install gazebo and other packages like vtk side-by-side anymore. Thus reopening and hoping for a solution in gazebo upstream some time :)

traversaro, 2022-02-04:
There is a PR in https://github.com/osrf/gazebo/pull/3146/files . Some problems may need to be solved for it to be merged upstream (mainly for ensure ABI-compatibility with pre-2021 tbb build) but if we do not care for tbb < 2021 compatibility we can probably take it as it is.

Tobias-Fischer, 2022-02-04:
Nice that sounds good. Will the ABI incompatibility cause issues here that we need to take care off?

traversaro, 2022-02-04:
> Nice that sounds good. Will the ABI incompatibility cause issues here that we need to take care off?

I am not 100% sure, but I think that in theory we should add `run_exports` directives for all libraries that are exposed in the public header of Gazebo, as they concur to define the public ABI of Gazebo. We did not get a log of problems related to this, but we could get them in the future. See also https://github.com/conda-forge/gazebo-feedstock/issues/107 . Perhaps we could do a single PR were we add all this run_exports directives, and then migrate to tbb2021. We could try to fix older builds via repodata patches, but that sounds a lot of work with limited gain.

Tobias-Fischer, 2022-02-04:
See my comment in https://github.com/conda-forge/gazebo-feedstock/issues/107 - I think we are okay as tbb (and dartstim etc) correctly specify their run_exports

closed 2022-02-06 by Tobias-Fischer