conda-forge/gazebo-feedstock #62: Gazebo on on Linux fails with Freeimage/jpeg related error

traversaro, 2021-03-12:
- [ ] I read [the conda-forge documentation](https://conda-forge.org/docs/user/introduction.html#how-can-i-install-packages-from-conda-forge) and could not find the solution for my problem there.

Issue:

I installed Gazebo on Linux today, (`conda install gazebo` in a vanila environment) and it fails with error:
~~~
(gazebo-testss) traversaro@IITICUBLAP102:~$ gazebo --verbose
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.30.128.21
[Msg] Publicized address: 172.30.128.21
[Wrn] [GuiIface.cc:120] X server does not support XInput 2
[Wrn] [GuiIface.cc:120] failed to get the current screen resources
[Wrn] [GuiIface.cc:120] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-traversaro'
[Wrn] [GuiIface.cc:120] QXcbConnection: XCB error: 1 (BadRequest), sequence: 165, resource id: 90, major code: 130 (Unknown), minor code: 47
[Wrn] [GuiIface.cc:120] QXcbConnection: XCB error: 170 (Unknown), sequence: 178, resource id: 90, major code: 146 (Unknown), minor code: 20
[Msg] Loading world file [/home/traversaro/miniforge3/envs/gazebo-testss/share/gazebo-11/worlds/empty.world]
[Wrn] [GuiIface.cc:120] JPEG parameter struct mismatch: library thinks size is 632, caller expects 664
[Wrn] [Publisher.cc:136] Queue limit reached for topic /gazebo/default/physics/contacts, deleting message. This warning is printed only once.
~~~

<br/>
Environment (<code>conda list</code>):
<details>

```
$ conda list
(gazebo-testss) traversaro@IITICUBLAP102:~$ conda list
# packages in environment at /home/traversaro/miniforge3/envs/gazebo-testss:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
boost-cpp                 1.74.0               hc6e9bd1_2    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.17.1               h7f98852_1    conda-forge
ca-certificates           2020.12.5            ha878542_0    conda-forge
certifi                   2020.12.5        py39hf3d152e_1    conda-forge
console_bridge            1.0.1                h4bd325d_0    conda-forge
cppzmq                    4.7.1                hf7cf922_2    conda-forge
curl                      7.75.0               h979ede3_0    conda-forge
dbus                      1.13.6               hfdff14a_1    conda-forge
eigen                     3.3.9                h4bd325d_1    conda-forge
expat                     2.2.10               h9c3ff4c_0    conda-forge
ffmpeg                    4.3.1                hca11adc_2    conda-forge
fontconfig                2.13.1            hba837de_1004    conda-forge
freeimage                 3.18.0               he1b5a44_0    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
gazebo                    11.3.0               hecc34a0_9    conda-forge
gdbm                      1.18                 h0a1914f_2    conda-forge
gettext                   0.19.8.1          h0b5b191_1005    conda-forge
glib                      2.66.7               h9c3ff4c_1    conda-forge
glib-tools                2.66.7               h9c3ff4c_1    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
gst-plugins-base          1.18.3               h04508c2_0    conda-forge
gstreamer                 1.18.3               h3560a44_0    conda-forge
gts                       0.7.6                h64030ff_2    conda-forge
[... 112 lines cut ...]
    active env location : /home/traversaro/miniforge3/envs/gazebo-testss
            shell level : 1
       user config file : /home/traversaro/.condarc
 populated config files : /home/traversaro/miniforge3/.condarc
                          /home/traversaro/.condarc
          conda version : 4.9.2
    conda-build version : 3.21.4
         python version : 3.8.8.final.0
       virtual packages : __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/traversaro/miniforge3  (writable)
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/traversaro/miniforge3/pkgs
                          /home/traversaro/.conda/pkgs
       envs directories : /home/traversaro/miniforge3/envs
                          /home/traversaro/.conda/envs
               platform : linux-64
             user-agent : conda/4.9.2 requests/2.25.1 CPython/3.8.8 Linux/4.19.104-microsoft-standard ubuntu/20.04.2 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```
</details>

traversaro, 2021-03-12:
I checked on a few GitHub Actions jobs, and there the correct version freeimage (`3.18.0  h0e1a528_5`) seems to be installed, while in the original post `3.18.0 he1b5a44_0` for some reason was installed.

~~~
2021-03-12T17:09:05.2238728Z   _libgcc_mutex                     0.1  conda_forge         conda-forge/linux-64     Cached
2021-03-12T17:09:05.2239937Z   _openmp_mutex                     4.5  1_gnu               conda-forge/linux-64     Cached
2021-03-12T17:09:05.2242048Z   boost-cpp                      1.74.0  hc6e9bd1_2          conda-forge/linux-64      16 MB
2021-03-12T17:09:05.2243560Z   bzip2                           1.0.8  h7f98852_4          conda-forge/linux-64     Cached
2021-03-12T17:09:05.2244834Z   c-ares                         1.17.1  h7f98852_1          conda-forge/linux-64     Cached
2021-03-12T17:09:05.2246258Z   ca-certificates             2021.1.19  h06a4308_1          pkgs/main/linux-64       118 KB
2021-03-12T17:09:05.2247310Z   certifi                     2020.12.5  py39hf3d152e_1      conda-forge/linux-64     143 KB
2021-03-12T17:09:05.2248180Z   console_bridge                  1.0.1  hc9558a2_0          conda-forge/linux-64      17 KB
2021-03-12T17:09:05.2249197Z   cppzmq                          4.7.1  hf7cf922_2          conda-forge/linux-64      25 KB
2021-03-12T17:09:05.2250149Z   curl                           7.75.0  h979ede3_0          conda-forge/linux-64     147 KB
2021-03-12T17:09:05.2251279Z   dbus                          1.13.18  hb2f20db_0          pkgs/main/linux-64       504 KB
2021-03-12T17:09:05.2252023Z   eigen                           3.3.9  h4bd325d_1          conda-forge/linux-64     970 KB
2021-03-12T17:09:05.2252804Z   expat                          2.2.10  h9c3ff4c_0          conda-forge/linux-64     164 KB
2021-03-12T17:09:05.2253585Z   ffmpeg                          4.3.1  hca11adc_2          conda-forge/linux-64      92 MB
2021-03-12T17:09:05.2254384Z   fontconfig                     2.13.1  hba837de_1004       conda-forge/linux-64     344 KB
2021-03-12T17:09:05.2255202Z   freeimage                      3.18.0  h0e1a528_5          conda-forge/linux-64     476 KB
2021-03-12T17:09:05.2255989Z   freetype                       2.10.4  h0708190_1          conda-forge/linux-64     890 KB
2021-03-12T17:09:05.2256774Z   gazebo                         11.3.0  h5fc56bc_11         conda-forge/linux-64      63 MB
2021-03-12T17:09:05.2257908Z   gdbm                             1.18  h0a1914f_2          conda-forge/linux-64     190 KB
2021-03-12T17:09:05.2258746Z   gettext                      0.19.8.1  h0b5b191_1005       conda-forge/linux-64       4 MB
2021-03-12T17:09:05.2259556Z   glib                           2.66.7  h9c3ff4c_1          conda-forge/linux-64     443 KB
2021-03-12T17:09:05.2260904Z   glib-tools                     2.66.7  h9c3ff4c_1          conda-forge/linux-64      85 KB
2021-03-12T17:09:05.2261735Z   gmp                             6.2.1  h58526e2_0          conda-forge/linux-64     806 KB
2021-03-12T17:09:05.2262493Z   gnutls                         3.6.13  h85f3911_1          conda-forge/linux-64       2 MB
2021-03-12T17:09:05.2263302Z   gst-plugins-base               1.18.3  h04508c2_0          conda-forge/linux-64       2 MB
2021-03-12T17:09:05.2264351Z   gstreamer                      1.18.3  h3560a44_0          conda-forge/linux-64       2 MB
2021-03-12T17:09:05.2265131Z   gts                             0.7.6  h64030ff_2          conda-forge/linux-64     411 KB
2021-03-12T17:09:05.2265914Z   icu                              68.1  h58526e2_0          conda-forge/linux-64     Cached
2021-03-12T17:09:05.2266850Z   ilmbase                         2.5.5  h780b84a_0          conda-forge/linux-64     343 KB
2021-03-12T17:09:05.2267618Z   jpeg                               9d  h516909a_0          conda-forge/linux-64     266 KB
2021-03-12T17:09:05.2268376Z   jsoncpp                         1.9.4  h4bd325d_1          conda-forge/linux-64     310 KB
2021-03-12T17:09:05.2269153Z   jxrlib                            1.1  h516909a_2          conda-forge/linux-64     236 KB
2021-03-12T17:09:05.2269906Z   krb5                           1.17.2  h926e7f8_0          conda-forge/linux-64     Cached
2021-03-12T17:09:05.2270659Z   lame                            3.100  h14c3975_1001       conda-forge/linux-64     498 KB
2021-03-12T17:09:05.2271425Z   lcms2                             2.9  hbd6801e_2          conda-forge/linux-64     409 KB
2021-03-12T17:09:05.2272217Z   ld_impl_linux-64               2.35.1  hea4e1c9_2          conda-forge/linux-64     618 KB
2021-03-12T17:09:05.2273036Z   libccd                            2.1  he1b5a44_1          conda-forge/linux-64      33 KB
2021-03-12T17:09:05.2273830Z   libclang                       11.1.0  default_ha53f305_0  conda-forge/linux-64      19 MB
2021-03-12T17:09:05.2274660Z   libcurl                        7.75.0  hc4aaa36_0          conda-forge/linux-64     Cached
2021-03-12T17:09:05.2275456Z   libedit                  3.1.20191231  he28a2e2_2          conda-forge/linux-64     121 KB
2021-03-12T17:09:05.2276249Z   libev                            4.33  h516909a_1          conda-forge/linux-64     Cached
2021-03-12T17:09:05.2277201Z   libevent                       2.1.10  hcdb4288_3          conda-forge/linux-64       1 MB
2021-03-12T17:09:05.2278247Z   libffi                            3.3  h58526e2_2          conda-forge/linux-64      51 KB
[... cut ...]
2021-03-12T17:09:05.2368377Z   yaml                            0.2.5  h516909a_0          conda-forge/linux-64      82 KB
2021-03-12T17:09:05.2369296Z   zeromq                          4.3.4  h9c3ff4c_0          conda-forge/linux-64     352 KB
2021-03-12T17:09:05.2370150Z   zlib                           1.2.11  h516909a_1010       conda-forge/linux-64     106 KB
2021-03-12T17:09:05.2371445Z   zstd                            1.4.9  ha95c52a_0          conda-forge/linux-64     Cached
2021-03-12T17:09:05.2372241Z   zziplib                       0.13.69  hed695b0_1          conda-forge/linux-64      99 KB
~~~

traversaro, 2021-03-12:
Ok, the main difference here seems that the working version (CI) is also using the `pkgs/main` channel, while locally I am not using it.

traversaro, 2021-03-12:
For some reason `conda install gazebo` on pure conda-forge results in `gazebo 11.3.0 hecc34a0_9` being installed instead of the latest `gazebo 11.3.0 h5fc56bc_11`.

traversaro, 2021-03-12:
Interestingly, if I just use `conda-forge` channel, I can still install the correct `gazebo` and `freeimage` version by using `mamba install gazebo` instead of `conda install gazebo`. 

In particular, this happens with `conda install gazebo`:
~~~
(gazebo-conda-tests3) traversaro@IITICUBLAP102:~$ conda install gazebo
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/traversaro/miniforge3/envs/gazebo-conda-tests3

  added / updated specs:
    - gazebo

The following NEW packages will be INSTALLED:

  _libgcc_mutex      conda-forge/linux-64::_libgcc_mutex-0.1-conda_forge
  _openmp_mutex      conda-forge/linux-64::_openmp_mutex-4.5-1_gnu
  boost-cpp          conda-forge/linux-64::boost-cpp-1.74.0-hc6e9bd1_2
  bzip2              conda-forge/linux-64::bzip2-1.0.8-h7f98852_4
  c-ares             conda-forge/linux-64::c-ares-1.17.1-h7f98852_1
  ca-certificates    conda-forge/linux-64::ca-certificates-2020.12.5-ha878542_0
  certifi            conda-forge/linux-64::certifi-2020.12.5-py39hf3d152e_1
  console_bridge     conda-forge/linux-64::console_bridge-1.0.1-h4bd325d_0
  cppzmq             conda-forge/linux-64::cppzmq-4.7.1-hf7cf922_2
  curl               conda-forge/linux-64::curl-7.75.0-h979ede3_0
  dbus               conda-forge/linux-64::dbus-1.13.6-hfdff14a_1
  eigen              conda-forge/linux-64::eigen-3.3.9-h4bd325d_1
  expat              conda-forge/linux-64::expat-2.2.10-h9c3ff4c_0
  ffmpeg             conda-forge/linux-64::ffmpeg-4.3.1-hca11adc_2
  fontconfig         conda-forge/linux-64::fontconfig-2.13.1-hba837de_1004
  freeimage          conda-forge/linux-64::freeimage-3.18.0-he1b5a44_0
  freetype           conda-forge/linux-64::freetype-2.10.4-h0708190_1
  gazebo             conda-forge/linux-64::gazebo-11.3.0-hecc34a0_9
  gdbm               conda-forge/linux-64::gdbm-1.18-h0a1914f_2
  gettext            conda-forge/linux-64::gettext-0.19.8.1-h0b5b191_1005
  glib               conda-forge/linux-64::glib-2.66.7-h9c3ff4c_1
  glib-tools         conda-forge/linux-64::glib-tools-2.66.7-h9c3ff4c_1
  gmp                conda-forge/linux-64::gmp-6.2.1-h58526e2_0
  gnutls             conda-forge/linux-64::gnutls-3.6.13-h85f3911_1
  gst-plugins-base   conda-forge/linux-64::gst-plugins-base-1.18.3-h04508c2_0
  gstreamer          conda-forge/linux-64::gstreamer-1.18.3-h3560a44_0
  gts                conda-forge/linux-64::gts-0.7.6-h64030ff_2
  icu                conda-forge/linux-64::icu-68.1-h58526e2_0
  ilmbase            conda-forge/linux-64::ilmbase-2.5.5-h780b84a_0
  jpeg               conda-forge/linux-64::jpeg-9d-h36c2ea0_0
  jsoncpp            conda-forge/linux-64::jsoncpp-1.9.4-h4bd325d_1
  krb5               conda-forge/linux-64::krb5-1.17.2-h926e7f8_0
  lame               conda-forge/linux-64::lame-3.100-h7f98852_1001
  ld_impl_linux-64   conda-forge/linux-64::ld_impl_linux-64-2.35.1-hea4e1c9_2
  libccd             conda-forge/linux-64::libccd-2.1-he1b5a44_1
  libclang           conda-forge/linux-64::libclang-11.1.0-default_ha53f305_0
  libcurl            conda-forge/linux-64::libcurl-7.75.0-hc4aaa36_0
  libedit            conda-forge/linux-64::libedit-3.1.20191231-he28a2e2_2
  libev              conda-forge/linux-64::libev-4.33-h516909a_1
  libevent           conda-forge/linux-64::libevent-2.1.10-hcdb4288_3
  libffi             conda-forge/linux-64::libffi-3.3-h58526e2_2
  libgcc-ng          conda-forge/linux-64::libgcc-ng-9.3.0-h2828fa1_18
  libglib            conda-forge/linux-64::libglib-2.66.7-h3e27bee_1
  libglu             conda-forge/linux-64::libglu-9.0.0-he1b5a44_1001
  libgomp            conda-forge/linux-64::libgomp-9.3.0-h2828fa1_18
  libiconv           conda-forge/linux-64::libiconv-1.16-h516909a_0
  libignition-cmake2 conda-forge/linux-64::libignition-cmake2-2.6.2-h9c3ff4c_1
  libignition-commo~ conda-forge/linux-64::libignition-common3-3.11.1-h4b9f6df_0
  libignition-fuel-~ conda-forge/linux-64::libignition-fuel-tools4-4.3.0-hf08c594_2
  libignition-math6  conda-forge/linux-64::libignition-math6-6.7.0-h9c3ff4c_1
  libignition-msgs5  conda-forge/linux-64::libignition-msgs5-5.3.0-hc438e01_6
  libignition-tools1 conda-forge/linux-64::libignition-tools1-1.0.0-h21f569a_3
  libignition-trans~ conda-forge/linux-64::libignition-transport8-8.1.0-h3ef4182_6
[... 377 lines cut ...]
  x264                       1!161.3030  h7f98852_0          conda-forge/linux-64     Cached
  xorg-fixesproto                   5.0  h14c3975_1002       conda-forge/linux-64     Cached
  xorg-kbproto                    1.0.7  h14c3975_1002       conda-forge/linux-64     Cached
  xorg-libice                    1.0.10  h516909a_0          conda-forge/linux-64     Cached
  xorg-libsm                      1.2.3  hd9c2040_1000       conda-forge/linux-64     Cached
  xorg-libx11                    1.6.12  h516909a_0          conda-forge/linux-64     Cached
  xorg-libxau                     1.0.9  h14c3975_0          conda-forge/linux-64     Cached
  xorg-libxaw                    1.0.13  h516909a_1002       conda-forge/linux-64     Cached
  xorg-libxdmcp                   1.1.3  h516909a_0          conda-forge/linux-64     Cached
  xorg-libxext                    1.3.4  h516909a_0          conda-forge/linux-64     Cached
  xorg-libxfixes                  5.0.3  h516909a_1004       conda-forge/linux-64     Cached
  xorg-libxmu                     1.1.3  h516909a_0          conda-forge/linux-64     Cached
[... cut ...]
  xz                              5.2.5  h516909a_1          conda-forge/linux-64     Cached
  yaml                            0.2.5  h516909a_0          conda-forge/linux-64     Cached
  zeromq                          4.3.4  h9c3ff4c_0          conda-forge/linux-64     Cached
  zlib                           1.2.11  h516909a_1010       conda-forge/linux-64     Cached
  zstd                            1.4.9  ha95c52a_0          conda-forge/linux-64     Cached
  zziplib                       0.13.69  hed695b0_1          conda-forge/linux-64     Cached
~~~

traversaro, 2021-03-12:
@wolfv @Tobias-Fischer any idea?

wolfv, 2021-03-12:
There might be an issue with the current repodata "hack" that conda uses to be faster.

You can use something like CONDA_REPODATA_FN=repodata.json conda install ... to force conda to use the full repodata.

Mamba always uses the full repodata.

wolfv, 2021-03-13:
(also we can pull freeimage builds only for linux & osx). I think I have an open PR to freeimage to re-add Windows builds ... maybe we can get that going again. Besides gazebo and libignition, there are not many other projects depending on freeimage on conda-forge (I think there was occt and something else). You can check with:

`mamba repoquery whoneeds freeimage -c conda-forge`

wolfv, 2021-03-13:
we should probably get this one back on: https://github.com/conda-forge/freeimage-feedstock/pull/21 so that the Windows builds also get updated whenever conda-forge's libjpeg / libraw etc. changes. I can try to fix this by next week.

traversaro, 2021-03-13:
> 
> 
> we should probably get this one back on: [conda-forge/freeimage-feedstock#21](https://github.com/conda-forge/freeimage-feedstock/pull/21) so that the Windows builds also get updated whenever conda-forge's libjpeg / libraw etc. changes. I can try to fix this by next week.

Ack, I can check that out myself, I did the original packaging (and unvendoring) of freeimage on vcpkg on Windows (as at the time vcpkg was Windows only): https://github.com/microsoft/vcpkg/pull/366, so I can check there to understand how to get it to compile.

Tobias-Fischer, 2021-03-17:
@traversaro @wolfv - I don't quite understand what needs to be done to resolve this issue? Marking some previous builds as broken?

wolfv, 2021-03-18:
Yes, the freeimage builds that do not link libjpeg dynamically need to be removed (can be checked with ldd). 

I was thinking to wait for the rebuilds by @traversaro to do that but it doesn't matter much for Linux/OS X actually

Tobias-Fischer, 2021-03-18:
Gotcha - agree it's best to wait for the rebuilds then.

Tobias-Fischer, 2021-05-09:
@traversaro @wolfv - should we tackle this issue now? The rebuilds are merged, aren't they?

traversaro, 2021-05-09:
I just tested again `conda install gazebo`  on a vanilla env, and I get:
~~~
(testgaz) C:\Users\STraversaro>conda install gazebo
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.9.2
  latest version: 4.10.1

Please update conda by running

    $ conda update -n base conda

## Package Plan ##

  environment location: C:\Users\STraversaro\Miniforge3\envs\testgaz

  added / updated specs:
    - gazebo

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    boost-cpp-1.74.0           |       h54f0996_2        16.1 MB  conda-forge
    bzip2-1.0.8                |       h8ffe710_4         149 KB  conda-forge
    ca-certificates-2020.12.5  |       h5b45459_0         173 KB  conda-forge
    cairo-1.16.0               |    hb19e0ff_1008         2.3 MB  conda-forge
    certifi-2020.12.5          |   py39hcbf5309_1         144 KB  conda-forge
    cfitsio-3.470              |       h0af3d06_7         596 KB  conda-forge
    console_bridge-1.0.1       |       h5362a0b_0          23 KB  conda-forge
    cppzmq-4.7.1               |       h4324990_2          26 KB  conda-forge
    curl-7.76.1                |       hf1763fc_1         128 KB  conda-forge
    dlfcn-win32-1.3.0          |       h0e60522_0          16 KB  conda-forge
    eigen-3.3.9                |       h2d74725_1         970 KB  conda-forge
    expat-2.3.0                |       h39d44d4_0         271 KB  conda-forge
    ffmpeg-4.3.1               |       ha925a31_0        26.2 MB  conda-forge
    fontconfig-2.13.1          |    h1989441_1005         1.7 MB  conda-forge
    freeimage-3.18.0           |       hfcb8cd7_7         690 KB  conda-forge
    freetype-2.10.4            |       h546665d_1         489 KB  conda-forge
    freexl-1.0.6               |       ha8e266a_0          58 KB  conda-forge
    gazebo-11.5.1              |       hc5976fb_0        61.7 MB  conda-forge
    geos-3.9.1                 |       h39d44d4_2         1.1 MB  conda-forge
    geotiff-1.6.0              |       h8e90983_5         136 KB  conda-forge
    gettext-0.19.8.1           |    h1a89ca6_1005         5.0 MB  conda-forge
    gts-0.7.6                  |       h7c369d9_2         188 KB  conda-forge
    hdf4-4.2.13                |    h0e5069d_1005         1.1 MB  conda-forge
    hdf5-1.10.6                |nompi_h5268f04_1114        19.6 MB  conda-forge
    icu-68.1                   |       h0e60522_0        16.3 MB  conda-forge
    ilmbase-2.5.5              |       h12d4b20_0         294 KB  conda-forge
    jpeg-9d                    |       h8ffe710_0         366 KB  conda-forge
    jsoncpp-1.9.4              |       h2d74725_2         945 KB  conda-forge
    jxrlib-1.1                 |       h8ffe710_2         620 KB  conda-forge
    kealib-1.4.14              |       h96bfa42_2         158 KB  conda-forge
    krb5-1.17.2                |       hbae68bd_0         856 KB  conda-forge
    lcms2-2.12                 |       h2a16943_0         882 KB  conda-forge
    libclang-11.1.0            |default_h5c34c98_0        20.8 MB  conda-forge
    libcurl-7.76.1             |       hf1763fc_1         292 KB  conda-forge
    libffi-3.3                 |       h0e60522_2          37 KB  conda-forge
    libflang-5.0.0             |h6538335_20180525         519 KB  conda-forge
    libgdal-3.2.2              |       hade8cfc_3         9.1 MB  conda-forge
    libglib-2.68.1             |       h1e62bf3_0         3.4 MB  conda-forge
    libiconv-1.16              |       he774522_0         680 KB  conda-forge
    libignition-cmake2-2.8.0   |       h0e60522_0         175 KB  conda-forge
    libignition-common3-3.11.1 |       he3878d5_0         566 KB  conda-forge
    libignition-fuel-tools4-4.3.0|       h4936f55_2         211 KB  conda-forge
[... 169 lines cut ...]
  tbb-devel          conda-forge/win-64::tbb-devel-2020.2-h2d74725_4
  tiledb             conda-forge/win-64::tiledb-2.2.8-hf84e3da_0
  tiny-process-libr~ conda-forge/win-64::tiny-process-library-2.0.4-h0e60522_0
  tinyxml            conda-forge/win-64::tinyxml-2.6.2-h2d74725_2
  tinyxml2           conda-forge/win-64::tinyxml2-8.0.0-h39d44d4_1
  tk                 conda-forge/win-64::tk-8.6.10-h8ffe710_1
  tzdata             conda-forge/noarch::tzdata-2021a-he74cb21_0
  urdfdom            conda-forge/win-64::urdfdom-2.3.3-h2d74725_1
  urdfdom_headers    conda-forge/win-64::urdfdom_headers-1.0.5-h5362a0b_2
  vc                 conda-forge/win-64::vc-14.2-hb210afc_4
  vs2015_runtime     conda-forge/win-64::vs2015_runtime-14.28.29325-h5e1d092_4
  wheel              conda-forge/noarch::wheel-0.36.2-pyhd3deb0d_0
  wincertstore       conda-forge/win-64::wincertstore-0.2-py39hcbf5309_1006
  xerces-c           conda-forge/win-64::xerces-c-3.2.3-h0e60522_2
  xz                 conda-forge/win-64::xz-5.2.5-h62dcd97_1
  yaml               conda-forge/win-64::yaml-0.2.5-he774522_0
  zeromq             conda-forge/win-64::zeromq-4.3.4-h0e60522_0
  zlib               conda-forge/win-64::zlib-1.2.11-h62dcd97_1010
  zstd               conda-forge/win-64::zstd-1.4.9-h6255e5f_0
  zziplib            conda-forge/win-64::zziplib-0.13.69-h1d00b33_1

Proceed ([y]/n)?
~~~

So the latest builds of Freeimage and Gazebo are installed, so I think we can close the issue.

closed 2021-05-09 by traversaro