conda-forge/gazebo-feedstock #63: Gazebo on linux-aarch64: error while loading shared libraries: libtbb.so.2

matteolucchi, 2021-03-17:
- [ ] I read [the conda-forge documentation](https://conda-forge.org/docs/user/introduction.html#how-can-i-install-packages-from-conda-forge) and could not find the solution for my problem there.

Issue:

```
(robostackenv) matteolucchi@ubuntu:~$ gazebo
gzserver: error while loading shared libraries: libtbb.so.2: cannot open shared object file: No such file or directory
```
<br/>
Environment (<code>conda list</code>):
<details>

```
$ conda list
(robostackenv) matteolucchi@ubuntu:~$ conda list
# packages in environment at /home/matteolucchi/miniforge3/envs/robostackenv:
#
# Name                    Version                   Build  Channel
_openmp_mutex             4.5                       1_gnu    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
apr                       1.7.0                hf897c2e_4    conda-forge
assimp                    5.0.1                h66e8f5a_5    conda-forge
atk-1.0                   2.36.0               h8b428f0_4    conda-forge
bcrypt                    3.2.0            py38h9544abe_1    conda-forge
boost                     1.74.0           py38h4031626_3    conda-forge
boost-cpp                 1.74.0               hb17a1af_2    conda-forge
bzip2                     1.0.8                hf897c2e_4    conda-forge
c-ares                    1.17.1               hf897c2e_1    conda-forge
ca-certificates           2020.12.5            h4fd8a4c_0    conda-forge
cairo                     1.16.0            ha675c65_1007    conda-forge
catkin_pkg                0.4.23             pyh9f0ad1d_0    conda-forge
certifi                   2020.12.5        py38h2063c64_1    conda-forge
cffi                      1.14.5           py38hba996e7_0    conda-forge
cmake                     3.19.6               h5e8844f_0    conda-forge
console_bridge            1.0.1                hd62202e_0    conda-forge
cppzmq                    4.7.1                hef188df_2    conda-forge
cryptography              3.4.6            py38h06f97cd_0    conda-forge
curl                      7.75.0               h7f0bf2a_0    conda-forge
cycler                    0.10.0                     py_2    conda-forge
dbus                      1.13.6               h32d69cc_1    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.5.0              pyh9f0ad1d_0    conda-forge
docutils                  0.16             py38h2063c64_3    conda-forge
eigen                     3.3.9                hd62202e_1    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.2.10               h3621d8e_0    conda-forge
ffmpeg                    4.3.1                he2e520b_2    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      2.001                hab24e00_0    conda-forge
font-ttf-source-code-pro  2.030                hab24e00_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            hd44bfdd_1004    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               h2d0d6a1_6    conda-forge
freetype                  2.10.4               hdf53a3c_1    conda-forge
fribidi                   1.0.10               hb9de7d4_0    conda-forge
gazebo                    11.3.0               hbfd4b83_6    robostack
gdk-pixbuf                2.42.2               ha10f721_2    conda-forge
gettext                   0.19.8.1          h4454c44_1005    conda-forge
giflib                    5.2.1                hb9de7d4_2    conda-forge
glib                      2.66.7               h01db608_1    conda-forge
glib-tools                2.66.7               h01db608_1    conda-forge
gmock                     1.10.0               hd62202e_7    conda-forge
gmp                       6.2.1                h7fd3ca4_0    conda-forge
gnutls                    3.6.13               h5fb5790_1    conda-forge
gpgme                     1.15.1               h01db608_0    conda-forge
graphite2                 1.3.13            h7fd3ca4_1001    conda-forge
graphviz                  2.46.1               h4a260bf_4    conda-forge
gst-plugins-base          1.18.3               hf9c081e_0    conda-forge
[... 377 lines cut ...]
         python version : 3.8.8.final.0
       virtual packages : __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=aarch64
       base environment : /home/matteolucchi/miniforge3  (writable)
           channel URLs : https://conda.anaconda.org/robostack/linux-aarch64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/conda-forge/linux-aarch64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/linux-aarch64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-aarch64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/matteolucchi/miniforge3/pkgs
                          /home/matteolucchi/.conda/pkgs
       envs directories : /home/matteolucchi/miniforge3/envs
                          /home/matteolucchi/.conda/envs
               platform : linux-aarch64
             user-agent : conda/4.9.2 requests/2.25.1 CPython/3.8.8 Linux/5.8.0-45-generic ubuntu/20.04.2 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

```
</details>

Tobias-Fischer, 2021-03-17:
I'll rebuild gazebo with the latest commit from master tomorrow and this error should go away. @wolfv how about merging the PR?

traversaro, 2021-03-17:
I think this should be solved by:
~~~
conda install tbb=2020.02
~~~

matteolucchi, 2021-03-17:
> I think this should be solved by:
> 
> ```
> conda install tbb=2020.2
> ```
> 
> The underlying problem is that robostack repodata did not get modified by the repodata patching in [conda-forge/conda-forge-repodata-patches-feedstock#124](https://github.com/conda-forge/conda-forge-repodata-patches-feedstock/pull/124) and [conda-forge/conda-forge-repodata-patches-feedstock#127](https://github.com/conda-forge/conda-forge-repodata-patches-feedstock/pull/127) .

@traversaro Thank you very much! I can confirm that this solved the issue! 

> I'll rebuild gazebo with the latest commit from master tomorrow and this error should go away. @wolfv how about merging the PR?

@Tobias-Fischer thank you! If you let me know when this is done I'll give it a try and reinstall from scratch

Tobias-Fischer, 2021-03-17:
Hi @matteolucchi - the new version is now uploaded: https://anaconda.org/RoboStack/gazebo
It should fix your issue - if not, please feel free to reopen.

closed 2021-03-17 by Tobias-Fischer

matteolucchi, 2021-03-18:
It works! Great! Thanks again :)