conda-forge/gazebo-feedstock #74: Invisible .dae meshes on Windows

danzimmerman, 2021-05-08:
I'm having an issue with missing Collada (`.dae`) link visuals on conda-installed Gazebo 11.5.1 (also affects conda-installed Gazebo 11.3 before I updated everything this afternoon) on Windows 10.

I've observed this so far on certain robots from [the `melodic-devel-staging` branch](https://github.com/ros-industrial/universal_robot/tree/melodic-devel-staging/ur_description) of https://github.com/ros-industrial/universal_robot/

The [universal_robot](https://github.com/ros-industrial/universal_robot/tree/melodic-devel-staging) robots otherwise work without issues with Gazebo 11.3 from `ros-noetic-desktop-full` on my coworker's Focal machine and all of my Melodic installations (Bionic and Chocolatey Windows 10, both `ros-melodic-desktop-full` binary installs), so I thought I'd raise it here first.

**Issue Description**
If I try to visualize the UR5 robot described by:
 * [these Collada visual meshes in `ur_description`](https://github.com/ros-industrial/universal_robot/tree/melodic-devel-staging/ur_description/meshes/ur5/visual)  
 * [this YAML file specifying the visual parameters in `ur_description/config`](https://github.com/ros-industrial/universal_robot/blob/melodic-devel-staging/ur_description/config/ur5/visual_parameters.yaml)
 * [this launch file in `ur_gazebo`](https://github.com/ros-industrial/universal_robot/blob/melodic-devel-staging/ur_gazebo/launch/ur5_bringup.launch) 

I run `roslaunch ur_gazebo ur5_bringup.launch` and I'm missing the forearm and upper-arm visuals. The URDF and STL collision geometries (visualized in orange) are fine, and the robot moves properly. 

[image]

Passing `verbose` to Gazebo through the launch file gives errors like:
```
[Err] [..\gazebo\rendering\Visual.cc:374] Ogre Error:ItemIdentityException: Resource with the name robot::forearm_link::forearm_link_visual_MATERIAL_gazebo_material_2 already exists. in ResourceManager::add at ..\OgreMain\src\OgreResourceManager.cpp (line 158)
[Err] [..\gazebo\rendering\Visual.cc:375] Unable to create a mesh from C:/Code/ros/workspaces/noethp_ws/src\universal_robot\ur_description/meshes/ur5/visual/forearm.dae
[Err] [..\gazebo\rendering\Visual.cc:374] Ogre Error:ItemIdentityException: Resource with the name robot::upper_arm_link::upper_arm_link_visual_MATERIAL_gazebo_material_16 already exists. in ResourceManager::add at ..\OgreMain\src\OgreResourceManager.cpp (line 158)
[Err] [..\gazebo\rendering\Visual.cc:375] Unable to create a mesh from C:/Code/ros/workspaces/noethp_ws/src\universal_robot\ur_description/meshes/ur5/visual/upperarm.dae
```

I can get the link visuals to appear by opening the `.dae` files in Meshlab and re-exporting, but this appears to simply strip all the visual materials out of the file so that there's nothing to trigger the Ogre material error above.

[image]

Anyone else having an issue with this? Any ideas? 

Environment, `conda list` output:
<details>

```
# packages in environment at C:\Users\<username>\miniconda3\envs\noetic:
#
# Name                    Version                   Build  Channel
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
apr                       1.7.0                he38c35c_5    conda-forge
argon2-cffi               20.1.0           py38h294d835_2    conda-forge
assimp                    5.0.1                hc2aa0de_5    conda-forge
async_generator           1.10                       py_0    conda-forge
attrs                     21.2.0             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bcrypt                    3.2.0            py38h294d835_1    conda-forge
bleach                    3.3.0              pyh44b312d_0    conda-forge
blosc                     1.21.0               h0e60522_0    conda-forge
boost                     1.74.0           py38h1266d08_3    conda-forge
boost-cpp                 1.74.0               h54f0996_2    conda-forge
bqplot                    0.12.26            pyhd8ed1ab_0    conda-forge
brotli                    1.0.9                h0e60522_4    conda-forge
brotlipy                  0.7.0           py38hab1e662_1001    conda-forge
bullet                    2.89             py38h4c96930_2    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2020.12.5            h5b45459_0    conda-forge
cairo                     1.16.0            hb19e0ff_1008    conda-forge
catkin_pkg                0.4.23             pyh9f0ad1d_0    conda-forge
certifi                   2020.12.5        py38haa244fe_1    conda-forge
cffi                      1.14.5           py38hd8c33c5_0    conda-forge
cfitsio                   3.470                hbbe6aef_7    conda-forge
chardet                   4.0.0            py38haa244fe_1    conda-forge
charls                    2.2.0                h39d44d4_0    conda-forge
clang_variant             1.0                  cling_v0.8    conda-forge
clangdev                  5.0.0             hc06fbdd_1012    conda-forge
cloudpickle               1.6.0                      py_0    conda-forge
cmake                     3.20.2               h39d44d4_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
console_bridge            1.0.1                h5362a0b_0    conda-forge
[... 568 lines cut ...]
                          __archspec=1=x86_64
       base environment : C:\Users\<username>\miniconda3  (writable)
           channel URLs : https://conda.anaconda.org/robostack/win-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\<username>\miniconda3\pkgs
                          C:\Users\<username>\.conda\pkgs
                          C:\Users\<username>\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\<username>\miniconda3\envs
                          C:\Users\<username>\.conda\envs
                          C:\Users\<username>\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.9.2 requests/2.25.1 CPython/3.8.3 Windows/10 Windows/10.0.21370
          administrator : False
             netrc file : None
           offline mode : False
```
</details>

wolfv, 2021-05-08:
thanks for the very nice bug report! unfortunately I don't know what's going on here yet.

wolfv, 2021-05-08:
just wondering if you also have RViz open (at the same time?) and if you see any issues in RViz, too?

traversaro, 2021-05-08:
I vaguely remember some similar issue on vcpkg-based build on Gazebo (the one from https://github.com/robotology/robotology-superbuild-dependencies-vcpkg) when the same iCub model was removed and then re-inserted in the simulation, but in that case spawning a new Gazebo was solving the problem, so it was not particularly problematic. 

I think the problem is related due to a behavior change from Ogre 1.9 to Ogre 1.10, that changed a lot of aspects related to the resource handling (see https://github.com/ros-visualization/rviz/issues/1251 and https://github.com/ros-visualization/rviz/issues/1049 for a related bugs on the RViz side). The strange thing is that ROROnWindows/Chocolatey Melodic is not affected by this bug, as Ogre 1.10 is used also there (see https://github.com/ms-iot/ros-windows-build/blob/17c45551a500230674bad77c951989e0747c2d6b/ros/melodic/CHANGELOG.md). 

However, I investigated a bit and it seems that ROSOnWindows/Chocolatey Melodic is using a fork of Gazebo ( https://github.com/ms-iot/ros-windows-build/blob/17c45551a500230674bad77c951989e0747c2d6b/ros/melodic/CHANGELOG.md ) and checking in that branch it seems that the `gazebo/rendering/Visual.cc ` fix in contained in this commit https://github.com/ms-iot/gazebo/commit/667a1d3bd23717abf4c6e62e3ef947a659e085b7 is definitely related to that bug and should fix it.

traversaro, 2021-05-08:
The related change is documented in Ogre 1.10 Release Notes ( https://github.com/OGRECave/ogre/blob/master/Docs/1.10-Notes.md ) as:
> using a duplicate resource name is an error now and throws an exception

danzimmerman, 2021-05-08:
@wolfv I don't use RViz yet but it seems to have visual issues as well when I add `RobotModel` or when I `roslaunch ur_description view_urXX.launch` which just brings the robot up in RViz 

[image]

However, the all-white-links issue is uncorrelated from whether or not the same robot appears correctly in Gazebo. The UR10, which works properly in Gazebo, is afflicted the same as the UR5 which doesn't work in Gazebo.

I dumped the UR10 URDF from the parameter server [in this gist](https://gist.github.com/danzimmerman/d946f6a3169fd98641ca1f76d1150ed3)

Gazebo UR10 visualization:
[image]

RViz UR10 visualization by adding `RobotModel`:
[image]

I think this visualization issue is a separate known issue [mentioned here](https://github.com/ros-industrial/universal_robot/issues/537#issuecomment-696840518), and I suppose that affects RViz on Focal equally. I'll check that on Monday and get back to you.

RViz messages:
<details>

```
[ INFO] [1620492036.252901300]: rviz version 1.14.5
[ INFO] [1620492036.253093400]: compiled against Qt version 5.12.6
[ INFO] [1620492036.253268600]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1620492036.268596200]: Forcing OpenGl version 0.
[ INFO] [1620492040.424830800]: Stereo is NOT SUPPORTED
[ INFO] [1620492040.425028300]: OpenGL device: Intel(R) UHD Graphics 630
[ INFO] [1620492040.425183100]: OpenGl version: 4.6 (GLSL 4.6).
```
</details>

Tobias-Fischer, 2021-05-10:
Hi @danzimmerman - please give the new version a go and let us know if that fixed your problem (will take a few hours to build + propagate in the CDN). If not, please re-open the issue :)

closed 2021-05-10 by Tobias-Fischer

danzimmerman, 2021-05-10:
Yes, works here. Thanks so much everyone!

danzimmerman, 2021-05-20:
The RViz issue is not an issue at all, I'm just unfamiliar with RViz. It boots up in the `map` frame, which is invalid in the context of running aside Gazebo. This causes the arm to be collapsed and white. It seems like this is all working correctly with RViz/Gazebo now.

Tobias-Fischer, 2021-05-20:
Great, thanks for letting us know @danzimmerman :)

PR #75 (2021-05-10): Fix invisible meshes (fix #74)

Checklist
* [x] Used a [personal fork of the feedstock to propose changes](https://conda-forge.org/docs/maintainer/updating_pkgs.html#forking-and-pull-requests)
* [x] Bumped the build number (if the version is unchanged)
* [ ] Reset the build number to `0` (if the version changed)
* [x] [Re-rendered]( https://conda-forge.org/docs/maintainer/updating_pkgs.html#rerendering-feedstocks ) with the latest `conda-smithy` (Use the phrase <code>@<space/>conda-forge-admin, please rerender</code> in a comment in this PR for automated rerendering)
* [ ] Ensured the license  [... cut]