conda-forge/gazebo-feedstock #79: "libode.so.0.16.2: cannot open shared object file: No such file or directory" on Linux

johnwason, 2021-05-30:
I am seeing an error with a fresh install of Gazebo on Linux when installing using the following:

`conda create -n gazebo_3_9 gazebo python=3.9`

The package list installed is:

```
  _libgcc_mutex      conda-forge/linux-64::_libgcc_mutex-0.1-conda_forge
  _openmp_mutex      conda-forge/linux-64::_openmp_mutex-4.5-1_gnu
  alsa-lib           conda-forge/linux-64::alsa-lib-1.2.3-h516909a_0
  assimp             conda-forge/linux-64::assimp-5.0.1-hedfc422_5
  atk-1.0            conda-forge/linux-64::atk-1.0-2.36.0-h3371d22_4
  boost-cpp          conda-forge/linux-64::boost-cpp-1.74.0-hc6e9bd1_3
  bullet-cpp         conda-forge/linux-64::bullet-cpp-3.09-hde0f152_2
  bzip2              conda-forge/linux-64::bzip2-1.0.8-h7f98852_4
  c-ares             conda-forge/linux-64::c-ares-1.17.1-h7f98852_1
  ca-certificates    conda-forge/linux-64::ca-certificates-2021.5.30-ha878542_0
  cairo              conda-forge/linux-64::cairo-1.16.0-h6cf1ce9_1008
  certifi            conda-forge/linux-64::certifi-2021.5.30-py39hf3d152e_0
  cfitsio            conda-forge/linux-64::cfitsio-3.470-hb418390_7
  console_bridge     conda-forge/linux-64::console_bridge-1.0.1-h4bd325d_0
  cppzmq             conda-forge/linux-64::cppzmq-4.7.1-hf7cf922_2
  curl               conda-forge/linux-64::curl-7.77.0-hea6ffbf_0
  dartsim            conda-forge/linux-64::dartsim-6.10.1-h53b96f2_0
  dbus               conda-forge/linux-64::dbus-1.13.6-h48d8840_2
  eigen              conda-forge/linux-64::eigen-3.3.9-h4bd325d_1
  expat              conda-forge/linux-64::expat-2.4.1-h9c3ff4c_0
  fcl                conda-forge/linux-64::fcl-0.6.1-h2cbc392_3
  ffmpeg             conda-forge/linux-64::ffmpeg-4.3.1-hca11adc_2
  flann              conda-forge/linux-64::flann-1.9.1-h2e58136_1008
  font-ttf-dejavu-s~ conda-forge/noarch::font-ttf-dejavu-sans-mono-2.37-hab24e00_0
  font-ttf-inconsol~ conda-forge/noarch::font-ttf-inconsolata-3.000-h77eed37_0
  font-ttf-source-c~ conda-forge/noarch::font-ttf-source-code-pro-2.038-h77eed37_0
  font-ttf-ubuntu    conda-forge/noarch::font-ttf-ubuntu-0.83-hab24e00_0
  fontconfig         conda-forge/linux-64::fontconfig-2.13.1-hba837de_1005
  fonts-conda-ecosy~ conda-forge/noarch::fonts-conda-ecosystem-1-0
  fonts-conda-forge  conda-forge/noarch::fonts-conda-forge-1-0
  freeglut           conda-forge/linux-64::freeglut-3.2.1-h9c3ff4c_2
  freeimage          conda-forge/linux-64::freeimage-3.18.0-h88c329d_7
  freetype           conda-forge/linux-64::freetype-2.10.4-h0708190_1
  freexl             conda-forge/linux-64::freexl-1.0.6-h7f98852_0
  fribidi            conda-forge/linux-64::fribidi-1.0.10-h36c2ea0_0
  gazebo             conda-forge/linux-64::gazebo-11.5.1-haa37ec6_2
  gdbm               conda-forge/linux-64::gdbm-1.18-h0a1914f_2
  gdk-pixbuf         conda-forge/linux-64::gdk-pixbuf-2.42.6-h04a7f16_0
  geos               conda-forge/linux-64::geos-3.9.1-h9c3ff4c_2
  geotiff            conda-forge/linux-64::geotiff-1.6.0-hcf90da6_5
  gettext            conda-forge/linux-64::gettext-0.19.8.1-h0b5b191_1005
  giflib             conda-forge/linux-64::giflib-5.2.1-h36c2ea0_2
  glib               conda-forge/linux-64::glib-2.68.2-h9c3ff4c_1
  glib-tools         conda-forge/linux-64::glib-tools-2.68.2-h9c3ff4c_1
  gmp                conda-forge/linux-64::gmp-6.2.1-h58526e2_0
  gnutls             conda-forge/linux-64::gnutls-3.6.13-h85f3911_1
  graphite2          conda-forge/linux-64::graphite2-1.3.13-h58526e2_1001
  graphviz           conda-forge/linux-64::graphviz-2.47.2-h85b4f2f_0
  gst-plugins-base   conda-forge/linux-64::gst-plugins-base-1.18.4-hf529b03_2
  gstreamer          conda-forge/linux-64::gstreamer-1.18.4-h76c114f_2
  gtk2               conda-forge/linux-64::gtk2-2.24.33-h539f30e_1
  gts                conda-forge/linux-64::gts-0.7.6-h64030ff_2
  harfbuzz           conda-forge/linux-64::harfbuzz-2.8.1-h83ec7ef_0
  hdf4               conda-forge/linux-64::hdf4-4.2.15-h10796ff_3
  hdf5               conda-forge/linux-64::hdf5-1.10.6-nompi_h6a2412b_1114
  icu                conda-forge/linux-64::icu-68.1-h58526e2_0
  ilmbase            conda-forge/linux-64::ilmbase-2.5.5-h780b84a_0
  jpeg               conda-forge/linux-64::jpeg-9d-h36c2ea0_0
  json-c             conda-forge/linux-64::json-c-0.15-h98cffda_0
  jsoncpp            conda-forge/linux-64::jsoncpp-1.9.4-h4bd325d_2
  jxrlib             conda-forge/linux-64::jxrlib-1.1-h7f98852_2
  kealib             conda-forge/linux-64::kealib-1.4.14-hcc255d8_2
  krb5               conda-forge/linux-64::krb5-1.19.1-hcc1bbae_0
[... 332 lines cut ...]
  xorg-inputproto    conda-forge/linux-64::xorg-inputproto-2.3.2-h7f98852_1002
  xorg-kbproto       conda-forge/linux-64::xorg-kbproto-1.0.7-h7f98852_1002
  xorg-libice        conda-forge/linux-64::xorg-libice-1.0.10-h7f98852_0
  xorg-libsm         conda-forge/linux-64::xorg-libsm-1.2.3-hd9c2040_1000
  xorg-libx11        conda-forge/linux-64::xorg-libx11-1.6.12-h36c2ea0_0
  xorg-libxau        conda-forge/linux-64::xorg-libxau-1.0.9-h7f98852_0
  xorg-libxaw        conda-forge/linux-64::xorg-libxaw-1.0.14-h7f98852_0
  xorg-libxdmcp      conda-forge/linux-64::xorg-libxdmcp-1.1.3-h7f98852_0
  xorg-libxext       conda-forge/linux-64::xorg-libxext-1.3.4-h516909a_0
  xorg-libxfixes     conda-forge/linux-64::xorg-libxfixes-5.0.3-h516909a_1004
  xorg-libxi         conda-forge/linux-64::xorg-libxi-1.7.10-h516909a_0
  xorg-libxmu        conda-forge/linux-64::xorg-libxmu-1.1.3-h516909a_0
  xorg-libxpm        conda-forge/linux-64::xorg-libxpm-3.5.13-h516909a_0
  xorg-libxrender    conda-forge/linux-64::xorg-libxrender-0.9.10-h516909a_1002
  xorg-libxt         conda-forge/linux-64::xorg-libxt-1.1.5-h516909a_1003
  xorg-renderproto   conda-forge/linux-64::xorg-renderproto-0.11.1-h7f98852_1002
  xorg-xextproto     conda-forge/linux-64::xorg-xextproto-7.3.0-h7f98852_1002
  xorg-xproto        conda-forge/linux-64::xorg-xproto-7.0.31-h7f98852_1007
  xz                 conda-forge/linux-64::xz-5.2.5-h516909a_1
  yaml               conda-forge/linux-64::yaml-0.2.5-h516909a_0
  zeromq             conda-forge/linux-64::zeromq-4.3.4-h9c3ff4c_0
  zlib               conda-forge/linux-64::zlib-1.2.11-h516909a_1010
  zstd               conda-forge/linux-64::zstd-1.4.9-ha95c52a_0
  zziplib            conda-forge/linux-64::zziplib-0.13.69-h27826a3_1
```

Tobias-Fischer, 2021-05-31:
It seems that libode is only built for Python 3.8 and probably needs changing to get built against all supported Python versions @conda-forge/libode

traversaro, 2021-05-31:
The issues seems two:
* `libode` package should not depend on Python, but instead in its index.json it seems to contain:
~~~
{
  "arch": "x86_64",
  "build": "h9c3ff4c_5",
  "build_number": 5,
  "depends": [
    "libgcc-ng >=9.3.0",
    "libstdcxx-ng >=9.3.0",
    "pthread-stubs",
    "python_abi 3.8.* *_cp38"
  ],
  "license": "LGPL-2.1-or-later OR BSD-4-Clause",
  "name": "libode",
  "platform": "linux",
  "subdir": "linux-64",
  "timestamp": 1616591801454,
  "version": "0.16.2"
}
~~~
* pyode (the pytbon bindings of ode) should be available not only for Python 3.8, but also for the other supported versions.

I opened an issue in https://github.com/conda-forge/libode-feedstock/issues/9 .

traversaro, 2021-05-31:
This should be fixed by https://github.com/conda-forge/libode-feedstock/pull/10 .

traversaro, 2021-05-31:
I just tested:
~~~
conda create -n gazebo_3_9 gazebo python=3.9
~~~

and the correct version of libode is installed:
~~~
  libode             conda-forge/linux-64::libode-0.16.2-he80948d_6
~~~

Do you think we can close the issue @johnwason ?

johnwason, 2021-05-31:
Gazebo started up normally. The issue can be closed

closed 2021-05-31 by traversaro