conda-forge/gazebo-feedstock #81: Gazebo Ogre Error:ItemIdentityException: Resource with the name Console already exists error

traversaro, 2021-06-10:
- [x] I read [the conda-forge documentation](https://conda-forge.org/docs/user/introduction.html#how-can-i-install-packages-from-conda-forge) and could not find the solution for my problem there.

Issue:

<br/>
Environment (<code>conda list</code>):
<details>

@HosameldinMohamed  can you provide the `conda info` on the environment in which you are experiencing this? 

```
$ conda list

```
</details>

<br/>
Details about  <code>conda</code> and system ( <code>conda info</code> ):
<details>

@HosameldinMohamed  can you provide the `conda info` on the environment in which you are experiencing this? 

```
$ conda info

```
</details>

traversaro, 2021-06-10:
Hi @HosameldinMohamed, this issue may be a better place where to discuss the problem you are experiencing. Could you provide as additional info: 
* `conda list` output
* `conda info` output
* the output of the `env` command

Thanks!

HosameldinMohamed, 2021-06-10:
<details>
  <summary>  conda list </summary>
  
```
# packages in environment at /home/hosameldin/miniforge3/envs/robenv:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
ace                       7.0.2                h9c3ff4c_0    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
assimp                    5.0.1                hedfc422_5    conda-forge
atk-1.0                   2.36.0               h3371d22_4    conda-forge
boost-cpp                 1.74.0               hc6e9bd1_3    conda-forge
bullet-cpp                3.09                 hde0f152_2    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.17.1               h7f98852_1    conda-forge
ca-certificates           2021.5.30            ha878542_0    conda-forge
cairo                     1.16.0            h6cf1ce9_1008    conda-forge
certifi                   2021.5.30        py39hf3d152e_0    conda-forge
cfitsio                   3.470                hb418390_7    conda-forge
console_bridge            1.0.1                h4bd325d_0    conda-forge
cppzmq                    4.7.1                hf7cf922_2    conda-forge
curl                      7.77.0               hea6ffbf_0    conda-forge
dartsim                   6.10.1               h53b96f2_0    conda-forge
dbus                      1.13.6               h48d8840_2    conda-forge
eigen                     3.3.9                h4bd325d_1    conda-forge
expat                     2.4.1                h9c3ff4c_0    conda-forge
fcl                       0.6.1                h2cbc392_3    conda-forge
ffmpeg                    4.3.1                hca11adc_2    conda-forge
flann                     1.9.1             h2e58136_1008    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            hba837de_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.1                h9c3ff4c_2    conda-forge
freeimage                 3.18.0               h88c329d_7    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
freexl                    1.0.6                h7f98852_0    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gazebo                    11.5.1               haa37ec6_2    conda-forge
gazebo-yarp-plugins       3.6.1               ha18ff89_12    robotology
gdbm                      1.18                 h0a1914f_2    conda-forge
gdk-pixbuf                2.42.6               h04a7f16_0    conda-forge
geos                      3.9.1                h9c3ff4c_2    conda-forge
geotiff                   1.6.0                hcf90da6_5    conda-forge
gettext                   0.19.8.1          h0b5b191_1005    conda-forge
giflib                    5.2.1                h36c2ea0_2    conda-forge
glib                      2.68.2               h9c3ff4c_2    conda-forge
glib-tools                2.68.2               h9c3ff4c_2    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
graphite2                 1.3.13            h58526e2_1001    conda-forge
graphviz                  2.47.2               h85b4f2f_0    conda-forge
gst-plugins-base          1.18.4               hf529b03_2    conda-forge
gstreamer                 1.18.4               h76c114f_2    conda-forge
gtk2                      2.24.33              h539f30e_1    conda-forge
gts                       0.7.6                h64030ff_2    conda-forge
harfbuzz                  2.8.1                h83ec7ef_0    conda-forge
hdf4                      4.2.15               h10796ff_3    conda-forge
hdf5                      1.10.6          nompi_h6a2412b_1114    conda-forge
icu                       68.1                 h58526e2_0    conda-forge
icub-models               1.20.0              h9c3ff4c_12    robotology
ilmbase                   2.5.5                h780b84a_0    conda-forge
jpeg                      9d                   h36c2ea0_0    conda-forge
json-c                    0.15                 h98cffda_0    conda-forge
jsoncpp                   1.9.4                h4bd325d_2    conda-forge
[... 276 lines cut ...]
CONDA_DEFAULT_ENV=robenv
LC_TIME=it_IT.UTF-8
IRONCUB_COMPONENT_INSTALL_PREFIX=/home/hosameldin/Github/robotology-superbuild/build/install
QT_AUTO_SCREEN_SCALE_FACTOR=0
GAZEBO_MODEL_PATH=/home/hosameldin/miniforge3/envs/robenv/share/gazebo-11/models:/home/hosameldin/Github/gazebo/install/share/gazebo-11/models:/home/hosameldin/Github/gazebo/install/share/gazebo-11/models:/home/hosameldin/Github/robotology-superbuild/build/install/share/gazebo/models:/home/hosameldin/Github/robotology-superbuild/build/install/share/iCub/robots:/home/hosameldin/Github/robotology-superbuild/build/install/share:/home/hosameldin/Github/component_ironcub/models/:/home/hosameldin/Github/component_iro [... cut ...] :/home/hosameldin/Github/robotology-superbuild/build/install/bin:/home/hosameldin/Github/robotology-superbuild/build/install/bin:/home/hosameldin/Github/gazebo/install/bin
HISTIGNORE=&:[ ]*:exit:ls:bg:fg:history:clear
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
KDE_APPLICATIONS_AS_SCOPE=1
MAIL=/var/spool/mail/hosameldin
LC_NUMERIC=it_IT.UTF-8
PROJ_NETWORK=ON
OGRE_RESOURCE_PATH=/home/hosameldin/miniforge3/envs/robenv/lib/OGRE
KONSOLE_DBUS_WINDOW=/Windows/1
_=/usr/bin/env
```

</details>

traversaro, 2021-06-10:
> GAZEBO_RESOURCE_PATH=/home/hosameldin/miniforge3/envs/robenv/share/gazebo-11:/home/hosameldin/Github/gazebo/install/share/gazebo-11:/home/hosameldin/Github/gazebo/install/share/gazebo-11:/home/hosamel [...]

This one seems quite messy. If you install Gazebo from source, probably you also need to remove the `setup.sh` of the Gazebo from source install?

HosameldinMohamed, 2021-06-11:
@traversaro Yes I went through the bashrc and removed the setup all the setup files related to the other Gazebo installation

Here's the new `env` command result.

<details>
  <summary>  env </summary>

```
SHELL=/bin/bash
LSCOLORS=Gxfxcxdxdxegedabagacad
SESSION_MANAGER=local/IITICUBLAP203:@/tmp/.ICE-unix/1725,unix/IITICUBLAP203:/tmp/.ICE-unix/1725
WINDOWID=125829127
GAZEBO_MASTER_URI=http://localhost:11345
QT_SCREEN_SCALE_FACTORS=eDP1=1;DP1=1;DP2=1;DP3=1;VIRTUAL1=1;
COLORTERM=truecolor
LESS=-R
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session1
CONDA_EXE=/home/hosameldin/miniforge3/bin/conda
_CE_M=
LANGUAGE=
LC_ADDRESS=it_IT.UTF-8
LC_NAME=it_IT.UTF-8
YARP_ROBOT_NAME=iRonCub-Mk1
SHELL_SESSION_ID=1a457ea73ced4bf6bb60e87d3566bf5f
DESKTOP_SESSION=plasma
LC_MONETARY=it_IT.UTF-8
IRONCUB_INSTALL_PREFIX=/home/hosameldin/Github/robotology-superbuild/build/install
OSH=/home/hosameldin/.oh-my-bash
GTK_RC_FILES=/etc/gtk/gtkrc:/home/hosameldin/.gtkrc:/home/hosameldin/.config/gtkrc
XCURSOR_SIZE=24
ROS_PACKAGE_PATH=/home/hosameldin/miniforge3/envs/robenv/share
EDITOR=/usr/bin/nano
GTK_MODULES=canberra-gtk-module
XDG_SEAT=seat0
PWD=/home/hosameldin
GSETTINGS_SCHEMA_DIR=/home/hosameldin/miniforge3/envs/robenv/share/glib-2.0/schemas
XDG_SESSION_DESKTOP=KDE
LOGNAME=hosameldin
XDG_SESSION_TYPE=x11
CONDA_PREFIX=/home/hosameldin/miniforge3/envs/robenv
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
XAUTHORITY=/home/hosameldin/.Xauthority
MOTD_SHOWN=pam
COMPONENT_IRONCUB_PREFIX=/home/hosameldin/Github
GTK2_RC_FILES=/etc/gtk-2.0/gtkrc:/home/hosameldin/.gtkrc-2.0:/home/hosameldin/.config/gtkrc-2.0
HOME=/home/hosameldin
GAZEBO_PLUGIN_PATH=/home/hosameldin/miniforge3/envs/robenv/lib/gazebo-11/plugins:/home/hosameldin/miniforge3/envs/robenv/lib
LC_PAPER=it_IT.UTF-8
LANG=en_US.UTF-8
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=KDE
KONSOLE_DBUS_SERVICE=:1.406
KONSOLE_DBUS_SESSION=/Sessions/1
PROFILEHOME=
YARP_COLORED_OUTPUT=1
CONDA_PROMPT_MODIFIER=(robenv) 
AMENT_PREFIX_PATH=/home/hosameldin/miniforge3/envs/robenv/share
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
COMP_WORDBREAKS= 
"'><=;|&(
CPL_ZIP_ENCODING=UTF-8
GAZEBO_RESOURCE_PATH=/home/hosameldin/miniforge3/envs/robenv/share/gazebo-11::/home/hosameldin/Github/component_ironcub/models/worlds/iRonCub-Mk0_playground:/home/hosameldin/Github/component_ironcub/models/worlds/iRonCub-Mk1_playground:/home/hosameldin/Github/component_ironcub/models/worlds/iRonCub-Mk0_playground:/home/hosameldin/Github/component_ironcub/models/worlds/iRonCub-Mk1_playground:/home/hosameldin/miniforge3/envs/robenv/share/gazebo/worls
KONSOLE_VERSION=210401
PREFIX_IRONCUB_MISCELLANEOUS=/home/hosameldin/Github
KDE_SESSION_UID=1000
PREFIX_JETCAT_TURBINES=/home/hosameldin/Github
GEM_HOME=/home/hosameldin/miniforge3/envs/robenv/share/rubygems/
XDG_SESSION_CLASS=user
TERM=xterm-256color
[... 51 lines cut ...]

```
libGL error: MESA-LOADER: failed to open iris: /home/hosameldin/miniforge3/envs/robenv/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /home/hosameldin/miniforge3/envs/robenv/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /home/hosameldin/miniforge3/envs/robenv/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: swrast
libGL error: MESA-LOADER: failed to open iris: /home/hosameldin/miniforge3/envs/robenv/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
[... cut ...]
libGL error: MESA-LOADER: failed to open iris: /home/hosameldin/miniforge3/envs/robenv/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /home/hosameldin/miniforge3/envs/robenv/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri)
libGL error: failed to load driver: swrast
```

</details>

Maybe I should make a fresh installation of Gazebo again.

traversaro, 2021-06-11:
> Maybe I should make a fresh installation of Gazebo again.

You could try that. In the meanwhile, can you provide the exact info on the distro used? I remember it was something Arch-based, but if you could report it it may help.

HosameldinMohamed, 2021-06-11:
> can you provide the exact info on the distro used? I remember it was something Arch-based, but if you could report it it may help.

Yes,

```
Operating System: Manjaro Linux
Release:          21.0.6
Codename:         Ornara
Kernel: Linux 5.9.16-1-MANJARO
Architecture: x86-64
```

traversaro, 2021-06-25:
@HosameldinMohamed I renamed the issue as now the blocking error is another.

traversaro, 2021-06-25:
@wolfv reported a similar problem in Fedora 34, see:
https://gitter.im/conda-forge/conda-forge.github.io?at=60d1f2108c12474d8cc7e4a1 

The workaround proposed by @nehaljwani was to run the application as:
~~~
LD_PRELOAD=/usr/lib64/libstdc++.so.6 /path/to/qtapp
~~~

The rationale here is that the Intel iris driver is requiring a version of the  libstdc++ that is more recent that the one provided in conda-forge. The workaround proposed is to (temporary) used the system libstdc++ instead of the one provided by conda-forge, at least until conda-forge is updated to use the new GCC/libstdc++ . Why this happens in Fedora 34 or in Manjaro Linux 21.0.6 ? Probably because both are fairly recent distribution with a fairly recent Intel iris driver.

In the case of Manjaro, assuming that the system `libstdc++` is placed in `/usr/lib/libstdc++.so.6` (if it is placed somewhere else, change the argument of `LD_PRELOAD`), it should be possible to run Gazebo (or any GL-using executable) as:
~~~
LD_PRELOAD=/usr/lib/libstdc++.so.6 gazebo
~~~

wolfv, 2021-06-25:
Once conda-forge ships GCC 11.1 this issue should be resolved.

traversaro, 2021-08-05:
From what I understand, GCC 11 was packaged in https://github.com/conda-forge/ctng-compilers-feedstock/pull/53 and several recent PRs. I did not fully understand the relation between how this packages and the @HosameldinMohamed error are related, but if @HosameldinMohamed wants to try again with a clean environment and check if the problem is solved it may be a good thing to test.

HosameldinMohamed, 2021-08-10:
> From what I understand, GCC 11 was packaged in [conda-forge/ctng-compilers-feedstock#53](https://github.com/conda-forge/ctng-compilers-feedstock/pull/53) and several recent PRs. I did not fully understand the relation between how this packages and the @HosameldinMohamed error are related, but if @HosameldinMohamed wants to try again with a clean environment and check if the problem is solved it may be a good thing to test.

@traversaro I've tried again with a clean environment and `gazebo` can be launched with no issues.

wolfv, 2021-08-10:
awesome!

traversaro, 2021-08-10:
Great, I guess we can close the issue the n!

closed 2021-08-10 by traversaro