conda-forge/gazebo-feedstock #88: Gazebo ogre 1.12 problems

traversaro, 2021-07-20:
The build added in https://github.com/conda-forge/gazebo-feedstock/pull/85 have some problems. They have been historically tracked in https://github.com/osrf/gazebo/issues/2700, I just opened this issue for giving publicity of that debugging activity.

traversaro, 2022-03-02:
Relevant comment: https://github.com/osrf/gazebo/issues/2700#issuecomment-1041417601 .

Tobias-Fischer, 2022-12-13:
Unfortunately, these issues still persist and block us from using Gazebo with ros-humble (see https://github.com/RoboStack/ros-humble/issues/24). As far as I know, ros2 requires ogre 1.12.

Even with the latest builds, when running `gazebo` with ogre 1.12, the splash screen just appears briefly and then gazebo crashes. With ogre 1.10, gazebo runs fine.

Tobias-Fischer, 2022-12-13:
On linux, the error is
```
(gazebo_only_2) tobias@QUTLAB:~$ gazebo --verbose
Gazebo multi-robot simulator, version 11.12.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.12.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 131.181.33.140
[Msg] Loading world file [/home/tobias/mambaforge/envs/gazebo_only_2/share/gazebo-11/worlds/empty.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 131.181.33.140
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[Wrn] [GuiIface.cc:120] Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

[Err] [main.cc:37] Ogre Error:RuntimeAssertionException: Ogre/ShadowExtrudePointLight not found. Verify that you referenced the 'ShadowVolume' folder in your resources.cfg in initialise at /home/conda/feedstock_root/build_artifacts/ogre_1666481265720/work/OgreMain/src/OgreShadowVolumeExtrudeProgram.cpp (line 70)
```

/cc @traversaro @nicogene

traversaro, 2022-12-13:
In the next days I will try look into this again, also considering https://github.com/gazebosim/gazebo-classic/issues/2700#issuecomment-1041417601 that seems useful.

Tobias-Fischer, 2023-02-02:
Hi @traversaro, did you have a chance to look into this?

traversaro, 2023-02-02:
Hi @Tobias-Fischer, I tried to set up the dev environment for this some time ago but compiling Gazebo locally is quite slow. :D 

The only modifications I was able to do is to try to do what was suggested in https://github.com/gazebosim/gazebo-classic/issues/2700#issuecomment-1041417601 at the CMake level, here a quick patch:

~~~
diff --git a/media/CMakeLists.txt b/media/CMakeLists.txt
index 8c651154c0..033252a8c7 100644
--- a/media/CMakeLists.txt
+++ b/media/CMakeLists.txt
@@ -5,11 +5,29 @@ add_subdirectory(gui)
 add_subdirectory(materials)
 add_subdirectory(models)

-# use glsl 150 shaders if ogre version >= 1.9
-if ((NOT ${OGRE_VERSION} VERSION_LESS 1.9.0) AND NOT APPLE)
-  add_subdirectory(rtshaderlib150)
-else ()
-  add_subdirectory(rtshaderlib)
+option(GAZEBO_INSTALL_SHADERS_FROM_OGRE "If ON, do not install custom shader and only copy the one available in OGRE" OFF)
+if(NOT GAZEBO_INSTALL_SHADERS_FROM_OGRE)
+  # use glsl 150 shaders if ogre version >= 1.9
+  if ((NOT ${OGRE_VERSION} VERSION_LESS 1.9.0) AND NOT APPLE)
+    add_subdirectory(rtshaderlib150)
+  else ()
+    add_subdirectory(rtshaderlib)
+  endif()
+else()
+  if(NOT PKG_CONFIG_EXECUTABLE)
+    message(FATAL_ERROR "GAZEBO_INSTALL_SHADERS_FROM_OGRE option requires that pkg-config is available")
+  endif()
+  execute_process(COMMAND ${PKG_CONFIG_EXECUTABLE} --variable=prefix OGRE
+                  OUTPUT_VARIABLE ogre_install_prefix
+                  RESULT_VARIABLE ogre_install_prefix_result)
+  message(STATUS "=====> ogre_install_prefix: ${ogre_install_prefix}")
+  # See https://github.com/gazebosim/gazebo-classic/issues/2700#issuecomment-1041417601
+  file(GLOB rtshader_glsl_from_ogre "${ogre_install_prefix}/share/OGRE/Media/RTShaderLib/GLSL/*")
+  install(FILES ${rtshader_glsl_from_ogre}
+          DESTINATION ${CMAKE_INSTALL_PREFIX}/share/gazebo-${GAZEBO_MAJOR_VERSION}/media/rtshaderlib)
+  file(GLOB shadowvolume_files_from_ogre "${ogre_install_prefix}/share/OGRE/Media/ShadowVolume/*")
+  install(FILES ${shadowvolume_files_from_ogre}
+          DESTINATION ${CMAKE_INSTALL_PREFIX}/share/gazebo-${GAZEBO_MAJOR_VERSION}/media)
 endif()

 add_subdirectory(skyx)
~~~

However, I still need to test this. Furthermore, even if everything works fine, this will result in no shading at all (basically shapes with flat color) so to actually make it usuable something else needs to be fixed.

Tobias-Fischer, 2023-02-03:
Ah, I see. This is an annoying issue for RoboStack - because of rviz2 we are tied to ogre 1.12. Someone just dropped by my office and told me they moved away from RoboStack because of they could not use Gazebo :(.

I wonder what’s easier - debugging this issue here, or switching back to rviz_ogre_vendor and build the rest of RoboStack with ogre 1.10?!?

Sorry to abuse this issue, happy to discuss elsewhere.

Cc @wolfv

traversaro, 2023-02-03:
> Ah, I see. This is an annoying issue for RoboStack - because of rviz2 we are tied to ogre 1.12. Someone just dropped by my office and told me they moved away from RoboStack because of they could not use Gazebo :(.

:(

> I wonder what’s easier - debugging this issue here, or switching back to rviz_ogre_vendor and build the rest of RoboStack with ogre 1.10?!?

Despite I all in for de-vendoring, make stuff work with later version of the libraries, etc etc I am afraid switching  back to rviz_ogre_vendor may be the easier option, at least in the short term. We have been struggling with this for a long time, and it is quite difficult to have an ETA on when this could be, for several reasons.

traversaro, 2024-04-07:
We removed ogre 1.12 builds in https://github.com/conda-forge/gazebo-feedstock/pull/164 .