conda-forge/gazebo-feedstock #90: Gazebo not starting on Windows

johnwason, 2021-08-19:
- [x ] I read [the conda-forge documentation](https://conda-forge.org/docs/user/introduction.html#how-can-i-install-packages-from-conda-forge) and could not find the solution for my problem there.

Issue:

Gazebo won't start on Windows with the latest package 11.8.0. This is a new problem, Gazebo has been working fine in the past. The error message is:

```
Gazebo multi-robot simulator, version 11.8.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 11.8.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 169.254.99.251
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 169.254.99.251
[Err] [..\gazebo\gui\main.cc:37] Ogre Error:ItemIdentityException: Resource with the name shadow_caster_vp_glsl already exists. in ResourceManager::add at ..\OgreMain\src\OgreResourceManager.cpp (line 158)
[Err] [..\gazebo\server_main.cc:57] Ogre Error:ItemIdentityException: Resource with the name shadow_caster_vp_glsl already exists. in ResourceManager::add at ..\OgreMain\src\OgreResourceManager.cpp (line 158)
[Err] [..\gazebo\transport\Connection.cc:273] The I/O operation has been aborted because of either a thread exit or an application requestValue[995]
```

<br/>
Environment (<code>conda list</code>):
<details>

```
$ conda list

# Name                    Version                   Build  Channel
boost-cpp                 1.74.0               h5b4e17d_4    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.5.30            h5b45459_0    conda-forge
cairo                     1.16.0            hb19e0ff_1008    conda-forge
cfitsio                   3.470                h0af3d06_7    conda-forge
console_bridge            1.0.1                h5362a0b_0    conda-forge
cppzmq                    4.7.1                h4324990_2    conda-forge
curl                      7.78.0               h789b8ee_0    conda-forge
cvxopt                    1.2.6            py39h73f7814_1    conda-forge
cython                    0.29.24          py39h415ef7b_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
dotnet-runtime            5.0.9                h2d74725_0    conda-forge
dotnet-sdk                5.0.400              h2d74725_0    conda-forge
dsdp                      5.8               h6e01ec9_1203    conda-forge
ecos                      2.0.8            py39h5d4886f_1    conda-forge
eigen                     3.3.9                h2d74725_1    conda-forge
expat                     2.4.1                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
fftw                      3.3.9           nompi_hd3ad3c4_101    conda-forge
fontconfig                2.13.1            h1989441_1005    conda-forge
freeglut                  3.2.1                h0e60522_2    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
gazebo                    11.8.0               h2e5bf29_0    conda-forge
gazebo_model_robotraconteur_driver 0.1.3.1                       0    robotraconteur
gazebo_robotraconteur_server_plugin 0.1.3.3              h3ad6029_5    robotraconteur
geos                      3.9.1                h39d44d4_2    conda-forge
geotiff                   1.6.0                ha8a8a2d_6    conda-forge
gettext                   0.19.8.1          h1a89ca6_1005    conda-forge
glpk                      4.65              h8ffe710_1004    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
[... 143 lines cut ...]
      conda av data dir : C:\tools\miniconda3\etc\conda
  conda av metadata url : https://repo.anaconda.com/pkgs/main
           channel URLs : https://conda.anaconda.org/robostack/win-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\tools\miniconda3\pkgs
                          C:\Users\user\.conda\pkgs
                          C:\Users\user\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\user\.conda\envs
                          C:\tools\miniconda3\envs
                          C:\Users\user\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.10.1 requests/2.25.1 CPython/3.8.8 Windows/10 Windows/10.0.19041
          administrator : False
             netrc file : None
           offline mode : False
```
</details>

Tobias-Fischer, 2021-08-21:
Should be fixed after upgrading to the new build :) (will take a few hours)

closed 2021-08-21 by Tobias-Fischer

johnwason, 2021-08-22:
I am not seeing new packages on anaconda.org

Tobias-Fischer, 2021-08-22:
Not sure why the build wasn't triggered. Any idea what to do @traversaro @wolfv?

traversaro, 2021-08-23:
The CI job from the last commit seems to be failing with a patch-related problem, see : https://github.com/conda-forge/gazebo-feedstock/runs/3402506511, and a similar failure is in the related PR (see https://github.com/conda-forge/gazebo-feedstock/pull/91)?

Tobias-Fischer, 2021-08-24:
Ah gosh, no idea how I missed that. Thanks @traversaro!

johnwason, 2021-08-25:
Gazebo seems to be working normally again. Thanks for the fix!

The project that uses this package: https://github.com/robotraconteur-contrib/robotraconteur_training_sim

PR #91 (2021-08-21): Fix shader program

Checklist
* [x] Used a [personal fork of the feedstock to propose changes](https://conda-forge.org/docs/maintainer/updating_pkgs.html#forking-and-pull-requests)
* [x] Bumped the build number (if the version is unchanged)
* [ ] Reset the build number to `0` (if the version changed)
* [x] [Re-rendered]( https://conda-forge.org/docs/maintainer/updating_pkgs.html#rerendering-feedstocks ) with the latest `conda-smithy` (Use the phrase <code>@<space/>conda-forge-admin, please rerender</code> in a comment in this PR for automated rerendering)
* [ ] Ensured the license file is bein [... cut]