conda-forge/gazebo-feedstock #94: Gazebo on Windows (via conda-forge)

PrasadNR, 2021-08-31:
- [x] I read [the conda-forge documentation](https://conda-forge.org/docs/user/introduction.html#how-can-i-install-packages-from-conda-forge) and could not find the solution for my problem there. This is related to https://github.com/osrf/gazebo/issues/2899 .

Issue:

<br/>
Environment (<code>conda list</code>):
<details>

```
$ conda list
# packages in environment at C:\Users\Prasad\anaconda3\envs\ros2gazebo:
#
# Name                    Version                   Build  Channel
boost-cpp                 1.74.0               h5b4e17d_4    conda-forge
brotlipy                  0.7.0           py39hb82d6ee_1001    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.5.30            h5b45459_0    conda-forge
cairo                     1.16.0            hb19e0ff_1008    conda-forge
certifi                   2021.5.30        py39hcbf5309_0    conda-forge
cffi                      1.14.6           py39h0878f49_0    conda-forge
cfitsio                   3.470                h0af3d06_7    conda-forge
chardet                   4.0.0            py39hcbf5309_1    conda-forge
charset-normalizer        2.0.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
conda                     4.10.3           py39hcbf5309_0    conda-forge
conda-package-handling    1.7.3            py39hb3671d1_0    conda-forge
console_bridge            1.0.1                h5362a0b_0    conda-forge
cppzmq                    4.7.1                h4324990_2    conda-forge
cryptography              3.4.7            py39hd8d06c1_0    conda-forge
curl                      7.78.0               h789b8ee_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.1                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
fontconfig                2.13.1            h1989441_1005    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
gazebo                    11.8.1               h2e5bf29_0    conda-forge
geos                      3.9.1                h39d44d4_2    conda-forge
geotiff                   1.6.0                ha8a8a2d_6    conda-forge
gettext                   0.19.8.1          h1a89ca6_1005    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.10.6          nompi_h5268f04_1114    conda-forge
icu                       68.1                 h0e60522_0    conda-forge
idna                      3.1                pyhd3deb0d_0    conda-forge
ilmbase                   2.5.5                h12d4b20_0    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
jsoncpp                   1.9.4                h2d74725_3    conda-forge
jxrlib                    1.1                  h8ffe710_2    conda-forge
kealib                    1.4.14               h96bfa42_2    conda-forge
krb5                      1.19.2               hbae68bd_0    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
lerc                      2.2.1                h0e60522_0    conda-forge
libarchive                3.5.2                hb45042f_0    conda-forge
libclang                  11.1.0          default_h5c34c98_1    conda-forge
libcurl                   7.78.0               h789b8ee_0    conda-forge
libdeflate                1.7                  h8ffe710_5    conda-forge
libffi                    3.3                  h0e60522_2    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libgdal                   3.3.1                ha6cada3_1    conda-forge
libglib                   2.68.4               h1e62bf3_0    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libignition-cmake2        2.8.0                h0e60522_0    conda-forge
libignition-common3       3.13.2               h9bd54ec_1    conda-forge
libignition-fuel-tools4   4.4.0                hf705ca2_1    conda-forge
libignition-math6         6.8.0                h0e60522_0    conda-forge
[... 105 lines cut ...]
                          __archspec=1=x86_64
       base environment : C:\Users\Prasad\anaconda3\envs\ros2gazebo  (writable)
      conda av data dir : C:\Users\Prasad\anaconda3\envs\ros2gazebo\etc\conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\Prasad\anaconda3\envs\ros2gazebo\pkgs
                          C:\Users\Prasad\.conda\pkgs
                          C:\Users\Prasad\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\Prasad\anaconda3\envs\ros2gazebo\envs
                          C:\Users\Prasad\.conda\envs
                          C:\Users\Prasad\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.10.3 requests/2.26.0 CPython/3.9.6 Windows/8.1
 Windows/6.3.9600
          administrator : False
             netrc file : None
           offline mode : False

```
</details>

traversaro, 2021-08-31:
As mentioned in https://github.com/osrf/gazebo/issues/2899#issuecomment-908098859, what happens if you:
* Run `gazebo --verbose` 
* Try to first run `gzserver --verbose` in a terminal and `gzclient --verbose` in another 

Thanks!

traversaro, 2021-08-31:
Furthermore, could you please provide a more informative issue name, for example "Gazebo does not start on Windows 8"? Thanks!

PrasadNR, 2021-09-05:
```
(ros2gazebo) C:\Users\Prasad>gazebo --verbose
Gazebo multi-robot simulator, version 11.8.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.0.0.9
Gazebo multi-robot simulator, version 11.8.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.0.0.9
[Msg] Loading world file [C:/Users/Prasad/anaconda3/envs/ros2gazebo\Library\shar
e\gazebo-11\worlds/empty.world]
``````

(ros2gazebo) C:\Users\Prasad>gzclient --verbose
Gazebo multi-robot simulator, version 11.8.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.0.0.9
```