conda-forge/gazebo-feedstock #96: Unable to run gazebo in a docker enviroment

GiulioRomualdi, 2021-09-22:
Hi all, 
I tried to install gazebo via conda in a docker environment. 
In details pulled the mambforge image
```
docker pull condaforge/mambaforge:4.9.2-5
```
Then I ran it with
```
docker run -it --env="DISPLAY" --net=host  condaforge/mambaforge:4.9.2-5
```

I finally installed gazebo 
```
conda install -c conda-forge gazebo
```

when I tried to run gazebo, I got the following error
```
(base) root@iiticublap261:/# gazebo
gzserver: error while loading shared libraries: libGL.so.1: cannot open shared object file: No such file or directory
gzclient: error while loading shared libraries: libGL.so.1: cannot open shared object file: No such file or directory
```

<br/>
Environment (<code>conda list</code>):
<details>

```
$ conda list
# packages in environment at /opt/conda:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
boost-cpp                 1.74.0               h9d3c048_1    conda-forge
brotlipy                  0.7.0           py38h8df0ef7_1001    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.17.1               h36c2ea0_0    conda-forge
ca-certificates           2021.5.30            ha878542_0    conda-forge
certifi                   2021.5.30        py38h578d9bd_0    conda-forge
cffi                      1.14.4           py38ha65f79e_1    conda-forge
chardet                   4.0.0            py38h578d9bd_0    conda-forge
conda                     4.10.3           py38h578d9bd_1    conda-forge
conda-package-handling    1.7.2            py38h8df0ef7_0    conda-forge
console_bridge            1.0.1                h4bd325d_0    conda-forge
cppzmq                    4.7.1                hf7cf922_2    conda-forge
cryptography              3.3.1            py38h2b97feb_0    conda-forge
curl                      7.71.1               he644dc0_8    conda-forge
dbus                      1.13.6               h48d8840_2    conda-forge
eigen                     3.3.9                h4bd325d_1    conda-forge
expat                     2.4.1                h9c3ff4c_0    conda-forge
ffmpeg                    4.3.2                hca11adc_0    conda-forge
fontconfig                2.13.1            hba837de_1005    conda-forge
freeimage                 3.18.0               he1b5a44_0    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
gazebo                    11.3.0               hecc34a0_9    conda-forge
gdbm                      1.18                 h0a1914f_2    conda-forge
gettext                   0.19.8.1          h0b5b191_1005    conda-forge
glib                      2.68.3               h9c3ff4c_0    conda-forge
glib-tools                2.68.3               h9c3ff4c_0    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
gst-plugins-base          1.14.5               h0935bb2_2    conda-forge
gstreamer                 1.18.4               h76c114f_2    conda-forge
gts                       0.7.6                h64030ff_2    conda-forge
icu                       68.1                 h58526e2_0    conda-forge
idna                      2.10               pyh9f0ad1d_0    conda-forge
ilmbase                   2.5.5                h780b84a_0    conda-forge
jpeg                      9d                   h36c2ea0_0    conda-forge
jsoncpp                   1.9.4                h4bd325d_3    conda-forge
[... 124 lines cut ...]
 populated config files : /opt/conda/.condarc
          conda version : 4.10.3
    conda-build version : not installed
         python version : 3.8.6.final.0
       virtual packages : __linux=5.11.0=0
                          __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /opt/conda  (writable)
      conda av data dir : /opt/conda/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/conda/pkgs
                          /root/.conda/pkgs
       envs directories : /opt/conda/envs
                          /root/.conda/envs
               platform : linux-64
             user-agent : conda/4.10.3 requests/2.25.1 CPython/3.8.6 Linux/5.11.0-34-generic ubuntu/20.04.1 glibc/2.31
                UID:GID : 0:0
             netrc file : None
           offline mode : False

```
</details>

wolfv, 2021-09-22:
Hi @GiulioRomualdi you need to install the driver stuff in the container. those usually come from the system. E.g. with ubuntu it might be something like this answer:

https://stackoverflow.com/questions/63889102/unable-to-run-docker-image-due-to-libgl-error

Also here for Centos / Fedora:

https://conda-forge.org/docs/maintainer/knowledge_base.html#libgl

Note the extra yum dependencies that are probably necessary

GiulioRomualdi, 2021-09-22:
Hi @wolfv, thank you for your answer. I followed what written [here](http://wiki.ros.org/docker/Tutorials/Hardware%20Acceleration#Intel) and I installed 
```
libgl1-mesa-glx libgl1-mesa-dri  mesa-utils
```
I ran docker with the following command
```
xhost +
docker run \
  --volume=/tmp/.X11-unix:/tmp/.X11-unix \
  --device=/dev/dri:/dev/dri \
  --env="DISPLAY=$DISPLAY" \
  condaforge/mambaforge:4.9.2-5
```

But still I got the following error
```
gazebo --verbose
Gazebo multi-robot simulator, version 11.8.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.8.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.185
[Err] [RenderEngine.cc:460] Unable to load Ogre Plugin[/opt/conda/lib/OGRE/RenderSystem_GL]. Rendering will not be possible.Make sure you have installed OGRE and Gazebo properly.
[Err] [RenderEngine.cc:685] EXCEPTION: unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.

[Err] [RenderingIface.cc:41] Failed to load the Rendering engine subsystem
unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.
[Err] [gazebo.cc:81] Unable to load sensors
[Err] [Server.cc:328] Unable to load gazebo
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.185
[Err] [RenderEngine.cc:460] Unable to load Ogre Plugin[/opt/conda/lib/OGRE/RenderSystem_GL]. Rendering will not be possible.Make sure you have installed OGRE and Gazebo properly.
[Err] [RenderEngine.cc:685] EXCEPTION: unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.

[Err] [RenderingIface.cc:41] Failed to load the Rendering engine subsystem
unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.
[Wrn] [RenderEngine.cc:292] Cannot initialize render engine since render path type is NONE. Ignore this warning ifrendering has been turned off on purpose.
[Wrn] [GuiIface.cc:120] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
```
 
There seems that OpenGL is not installed

GiulioRomualdi, 2021-09-22:
I installed `libqt5opengl5-dev` and all the dependencies and now everything is working

closed 2021-09-22 by wolfv