conda-forge/gazebo-feedstock #107: Is run_exports/pin_subpackage with dart necessary?

traversaro, 2021-12-20:
Based on the problem described in https://github.com/robotology/yarp.js/issues/28#issuecomment-997459819, it seems that  run_exports/pin_subpackage with dart is necessary as dart is exposed in the public headers of Gazebo. I am just opening this issue to remember this.

Tobias-Fischer, 2022-02-04:
So as far as I understand https://github.com/robotology/yarp.js/issues/28 this is not an issue with gazebo which properly specifies the dartsim (and tbb etc) dependencies; see 
[image]

Instead in this particular case it's gazebo-yarp-plugins that does not have any dependencies on dartsim although it should have it (I'm not sure why that happens).

So I think we can close the issue here?!? Or maybe I'm misunderstanding :)

traversaro, 2022-02-05:
> So I think we can close the issue here?!? Or maybe I'm misunderstanding :)

It is a bit hairy (and fortunatly a corner case), so mutual misunderstanding is quite possible. :D 

> Instead in this particular case it's gazebo-yarp-plugins that does not have any dependencies on dartsim although it should have it (I'm not sure why that happens).

The problem is that `gazebo-yarp-plugins` does not depend on dart directly, it just links dart as dart is contained in `GAZEBO_LIBRARIES` variable (see https://github.com/robotology/gazebo-yarp-plugins/blob/40ef333672b9cc85cdf50255d229b6bb462237cb/plugins/imu/CMakeLists.txt#L14).
Why dart is listed in `GAZEBO_LIBRARIES`? Because in theory all the DART-related data structures are installed in public headers (see https://github.com/osrf/gazebo/blob/cc06796b135f03066a96c68dfa4e2f89d5f76b3a/gazebo/physics/dart/CMakeLists.txt#L58), so a downstream project could include this files that itself include DART headers. A similar issue exists for all other libraries that are used in public headers, even more in the APIs that are actually used by plugins (such as boost).

Tobias-Fischer, 2022-02-05:
Hm. You are probably right, but one more thing: why does the gazebo-yarp-plugins not depend on gazebo, which to my understanding would pull in all the other dependencies? It looks as if it just depends on yarp and libopencv.

Tobias-Fischer, 2022-09-22:
@traversaro - have you checked this further?

traversaro, 2023-01-30:
> @traversaro - have you checked this further?

I finally have an example of recipe that depends on Gazebo on conda-forge, see . 

This is failing with:
~~~
["  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_linkattacher.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_showmodelcom.so): .. but ['libprotobuf'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_showmodelcom.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_showmodelcom.so): .. but ['libignition-math6'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_camera.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_worldinterface.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_MultiCameraPlugin.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_externalwrench.so): .. but ['libprotobuf'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_externalwrench.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_externalwrench.so): .. but ['libignition-math6'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_multicamera.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_multicamera.so): .. but ['libignition-math6'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_lasersensor.so): .. but ['libignition-math6'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_fakecontrolboard.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_depthCamera.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_depthCamera.so): .. but ['libignition-math6'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_videotexture.so): .. but ['ogre'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_videotexture.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_maissensor.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)", "  ERROR (libgazebo-yarp-plugins,lib/libgazebo_yarp_controlboard.so): .. but ['tbb'] not in reqs/run, (i.e. it is overlinking) (likely) or a missing dependency (less likely)"]
~~~

traversaro, 2024-05-23:
Related proposal: https://github.com/conda/ceps/issues/77 .

traversaro, 2025-08-29:
Related proposal: https://github.com/conda/ceps/pull/129 .