conda-forge/gazebo-feedstock #111: Compile issue on osx-arm64 (native)

Tobias-Fischer, 2022-01-23:
Hi @conda-forge/gazebo,

I am currently looking into getting gazebo (and it's dependencies) to run on osx-arm64. I have local builds for all dependencies (uploaded to the robostack-experimental channel as well).

As a first step before cross-compilation, I am trying a native compile. It currently fails when creating the msgs with protoc:
```
[91/1183] Running C++ protocol buffer compiler on pose_stamped.proto
FAILED: gazebo/msgs/pose_stamped.pb.cc gazebo/msgs/pose_stamped.pb.h $SRC_DIR/build/gazebo/msgs/pose_stamped.pb.cc $SRC_DIR/build/gazebo/msgs/pose_stamped.pb.h 
cd $SRC_DIR/build/gazebo/msgs && $PREFIX/bin/protoc --plugin=protoc-gen-gazebomsgs=$SRC_DIR/build/gazebo/msgs/gazebomsgs_out --cpp_out=dllexport_decl=GZ_MSGS_VISIBLE:$SRC_DIR/build/gazebo/msgs --gazebomsgs_out=$SRC_DIR/build/gazebo/msgs --proto_path=$SRC_DIR/gazebo/msgs $SRC_DIR/gazebo/msgs/pose_stamped.proto
--gazebomsgs_out: protoc-gen-gazebomsgs: Plugin killed by signal 9.
```

Just putting this here in case anyone has ideas.

Tobias-Fischer, 2022-01-23:
The diff so far:

```diff
(base) fischert@VXTGFM46Y6 recipe % git diff
diff --git a/conda-forge.yml b/conda-forge.yml
index 1c09752..c1f8f44 100644
--- a/conda-forge.yml
+++ b/conda-forge.yml
@@ -1 +1,5 @@
 conda_forge_output_validation: true
+build_platform:
+  osx_arm64: osx_64
+test_on_native_only: true
+
diff --git a/recipe/build.sh b/recipe/build.sh
index 11a6753..f2963c2 100644
--- a/recipe/build.sh
+++ b/recipe/build.sh
@@ -11,8 +11,8 @@ cmake .. \
       -DBoost_DEBUG=OFF \
       -DHAVE_OPENAL:BOOL=OFF 
 
-cmake --build . --config $BUILD_TYPE -- -j$CPU_COUNT
-cmake --build . --config $BUILD_TYPE --target install
+cmake --build . -- -j$CPU_COUNT
+cmake --build . --target install
 
 # Copy the [de]activate scripts to $PREFIX/etc/conda/[de]activate.d.
 # This will allow them to be run on environment activation.
diff --git a/recipe/meta.yaml b/recipe/meta.yaml
index 97aacc3..bdb7c4a 100644
--- a/recipe/meta.yaml
+++ b/recipe/meta.yaml
@@ -28,13 +28,13 @@ requirements:
   build:
     - {{ compiler('cxx') }}
     - {{ compiler('c') }}
-    - cmake 3.16.*
+    - cmake
     - pkg-config
     - ninja
 
     # x11 stuff
-    - {{ cdt('mesa-libgl-devel') }}  # [unix]
-    - {{ cdt('mesa-dri-drivers') }}  # [unix]
+    - {{ cdt('mesa-libgl-devel') }}  # [unix and not (osx and arm64)]
+    - {{ cdt('mesa-dri-drivers') }}  # [unix and not (osx and arm64)]
     - {{ cdt('libselinux') }}        # [linux]
     - {{ cdt('libxcb') }}            # [linux]
     - {{ cdt('libxdamage') }}        # [linux]
@@ -62,7 +62,8 @@ requirements:
     - libignition-transport8
     - libignition-common3
     - libignition-fuel-tools4
-    - qt
+    - qt  # [not (osx and arm64)]
+    - qt-main  # [osx and arm64]
     {% if OGRE_VERSION == "1.10" %}
     - ogre 1.10.*
     {% else %}
@@ -116,7 +117,8 @@ requirements:
     - libglu            # [linux]
     # deps
     - libprotobuf
-    - qt
+    - qt  # [not (osx and arm64)]
+    - qt-main  # [osx and arm64]
     - ogre
     - freeimage
     - curl
```

traversaro, 2022-01-24:
The protobuf msgs generation machinery is indeed a bit complex. Just to have a simpler but similar case, if it easy can you try to build natively ign-msgs?

Tobias-Fischer, 2022-01-24:
Hmm when I do it directly all seems fine @wolfv:
```
(/Users/fischert/mambaforge/conda-bld/gazebo_1642975576516/_build_env) fischert@VXTGFM46Y6 work % cd $SRC_DIR/build/gazebo/msgs && $PREFIX/bin/protoc --plugin=protoc-gen-gazebomsgs=$SRC_DIR/build/gazebo/msgs/gazebomsgs_out --cpp_out=dllexport_decl=GZ_MSGS_VISIBLE:$SRC_DIR/build/gazebo/msgs --gazebomsgs_out=$SRC_DIR/build/gazebo/msgs --proto_path=$SRC_DIR/gazebo/msgs $SRC_DIR/gazebo/msgs/pose_stamped.proto
(/Users/fischert/mambaforge/conda-bld/gazebo_1642975576516/_build_env) fischert@VXTGFM46Y6 msgs % echo $?
0
```

Tobias-Fischer, 2022-02-08:
So it seems like it's yet another instance of a code signing issue. After code signing `gazebomsgs_out` manually (https://github.com/osrf/gazebo/blob/gazebo11/gazebo/msgs/CMakeLists.txt), the build works fine. Any ideas how to codesign immediately, before installing, so it can be used to generate the messages?

Tobias-Fischer, 2022-02-08:
Got it ..
```patch
diff --git a/gazebo/msgs/CMakeLists.txt b/gazebo/msgs/CMakeLists.txt
index c12584da96..c1ec30cb04 100644
--- a/gazebo/msgs/CMakeLists.txt
+++ b/gazebo/msgs/CMakeLists.txt
@@ -156,6 +156,11 @@ if (NOT MSVC)
   endif()
 endif()
 
+if(APPLE)
+    add_custom_command(TARGET gazebomsgs_out POST_BUILD COMMAND codesign -s - -f $<TARGET_FILE:gazebomsgs_out>)
+
+endif()
+
 set (sources msgs.cc MsgFactory.cc)
 set (headers msgs.hh MsgFactory.hh)
 
```

Tobias-Fischer, 2022-05-09:
We have cross-compiled builds so all good.

closed 2022-05-09 by Tobias-Fischer