conda-forge/gazebo-feedstock #119: Recent builds of Gazebo segfaults on WSL2

traversaro, 2022-02-08:
I am experiencing non-deterministic (but frequenct, like 1/2 or 2/3 of the times) in recent builds of Gazebo:
* gazebo=11.10.1=h9f65e32_1
* gazebo=11.10.1=h7ab597c_2

List (in testing different versions only tbb and gazebo change):
~~~
(gazebolatest) traversaro@IITICUBLAP257:/mnt/c/Users/STraversaro$ mamba list
# packages in environment at /home/traversaro/mambaforge/envs/gazebolatest:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
aom                       3.2.0                h9c3ff4c_2    conda-forge
assimp                    5.2.0                hedfc422_1    conda-forge
atk-1.0                   2.36.0               h3371d22_4    conda-forge
blosc                     1.21.0               h9c3ff4c_0    conda-forge
boost-cpp                 1.74.0               h6cacc03_7    conda-forge
bullet-cpp                3.21                 hb5077e9_1    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
cairo                     1.16.0            ha00ac49_1009    conda-forge
cfitsio                   4.0.0                h9a35b8e_0    conda-forge
console_bridge            1.0.1                h4bd325d_0    conda-forge
cppzmq                    4.8.1                hf7cf922_0    conda-forge
curl                      7.81.0               h2574ce0_0    conda-forge
dartsim                   6.12.1               h7529616_2    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
expat                     2.4.4                h9c3ff4c_0    conda-forge
fcl                       0.7.0                h2cbc392_0    conda-forge
ffmpeg                    4.4.1                h6987444_0    conda-forge
flann                     1.9.1             hfe772e8_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              ha180cfb_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.2                h9c3ff4c_0    conda-forge
freeimage                 3.18.0               h88c329d_7    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
freexl                    1.0.6                h7f98852_0    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gazebo                    11.10.1              h7ab597c_2    conda-forge
gdbm                      1.18                 h0a1914f_2    conda-forge
gdk-pixbuf                2.42.6               h04a7f16_0    conda-forge
geos                      3.10.2               h9c3ff4c_0    conda-forge
geotiff                   1.7.0                h6593c0a_6    conda-forge
gettext                   0.19.8.1          h73d1719_1008    conda-forge
giflib                    5.2.1                h36c2ea0_2    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
graphite2                 1.3.13            h58526e2_1001    conda-forge
graphviz                  2.50.0               h8e749b2_2    conda-forge
gst-plugins-base          1.18.5               hf529b03_3    conda-forge
gstreamer                 1.18.5               h9f60fe5_3    conda-forge
gtk2                      2.24.33              h90689f9_2    conda-forge
gts                       0.7.6                h64030ff_2    conda-forge
harfbuzz                  3.3.1                hb4a5f5f_0    conda-forge
hdf4                      4.2.15               h10796ff_3    conda-forge
hdf5                      1.12.1          nompi_h2750804_103    conda-forge
icu                       69.1                 h9c3ff4c_0    conda-forge
ilmbase                   2.5.5                h780b84a_0    conda-forge
jbig                      2.1               h7f98852_2003    conda-forge
jpeg                      9e                   h7f98852_0    conda-forge
json-c                    0.15                 h98cffda_0    conda-forge
jsoncpp                   1.9.5                h4bd325d_1    conda-forge
[... 124 lines cut ...]
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxaw               1.0.14               h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxmu               1.1.3                h7f98852_0    conda-forge
xorg-libxpm               3.5.13               h7f98852_0    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zlib                      1.2.11            h36c2ea0_1013    conda-forge
zstd                      1.5.2                ha95c52a_0    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
~~~

I initially tought this was due to https://github.com/conda-forge/gazebo-feedstock/pull/115, but it seems that it happens also in earlier build. It also may be something in my setup.

traversaro, 2022-02-11:
The great debugging of @HosameldinMohamed and @kouroshD suggested that the problem is related to Qt. Indeed, a new qt build was uploaded ~25 days ago: https://anaconda.org/conda-forge/qt/files .

An older version can be installed with:
~~~
mamba install qt=5.12.9=hda022c4_4
~~~

The latest gazebo build is not compatible with `qt=5.12.9=hda022c4_4`, so the solver installs gazebo 11.10.0, that unfortunatly does not start with:
~~~
(gazebo) traversaro@IITICUBLAP257:/mnt/c/Users/STraversaro$ gazebo --verbose
Gazebo multi-robot simulator, version 11.10.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.10.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.20.249.14
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.20.249.14
[Err] [server_main.cc:57] Ogre Error:ItemIdentityException: Resource with the name shadow_caster_vp_glsl already exists. in ResourceManager::add at /home/conda/feedstock_root/build_artifacts/ogre_1612037790705/work/OgreMain/src/OgreResourceManager.cpp (line 157)
~~~

(This was an upstream problem fixed in 11.10.1, see https://github.com/osrf/gazebo/pull/3161).

However, if one ask for an older gazebo such as 11.9, gazebo works again:
~~~
mamba install gazebo=11.9 qt=5.12.9=hda022c4_4
~~~

Note that this is not related to the gazebo version, as if one installs:
~~~
mamba install gazebo=11.9 qt=5.12.9=ha98a1a1_5
~~~
gazebo segfaults as well.

traversaro, 2022-02-11:
<details>
    <summary>Details on working and not working environments</summary>

Environment that is working (`mamba install gazebo=11.9 qt=5.12.9=hda022c4_4`):
~~~sh
(gazebo_working) traversaro@IITICUBLAP257:/mnt/c/Users/STraversaro$ mamba list
# packages in environment at /home/traversaro/mambaforge/envs/gazebo_working:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
aom                       3.2.0                h9c3ff4c_2    conda-forge
assimp                    5.0.1                hedfc422_7    conda-forge
atk-1.0                   2.36.0               h3371d22_4    conda-forge
blosc                     1.21.0               h9c3ff4c_0    conda-forge
boost-cpp                 1.74.0               h312852a_4    conda-forge
bullet-cpp                3.17                 hb5077e9_1    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
cairo                     1.16.0            h6cf1ce9_1008    conda-forge
cfitsio                   4.0.0                h9a35b8e_0    conda-forge
console_bridge            1.0.1                h4bd325d_0    conda-forge
cppzmq                    4.8.1                hf7cf922_0    conda-forge
curl                      7.81.0               h2574ce0_0    conda-forge
dartsim                   6.12.1               he39ca3a_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
expat                     2.4.4                h9c3ff4c_0    conda-forge
fcl                       0.7.0                h2cbc392_0    conda-forge
ffmpeg                    4.4.1                h6987444_0    conda-forge
flann                     1.9.1             hfe772e8_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              ha180cfb_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.2                h9c3ff4c_0    conda-forge
freeimage                 3.18.0               h88c329d_7    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
freexl                    1.0.6                h7f98852_0    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gazebo                    11.9.1               h9f65e32_2    conda-forge
gdbm                      1.18                 h0a1914f_2    conda-forge
gdk-pixbuf                2.42.6               h04a7f16_0    conda-forge
geos                      3.10.1               h9c3ff4c_1    conda-forge
geotiff                   1.7.0                h90a4e78_5    conda-forge
gettext                   0.19.8.1          h73d1719_1008    conda-forge
giflib                    5.2.1                h36c2ea0_2    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
graphite2                 1.3.13            h58526e2_1001    conda-forge
graphviz                  2.50.0               h85b4f2f_1    conda-forge
gst-plugins-base          1.18.5               hf529b03_3    conda-forge
gstreamer                 1.18.5               h9f60fe5_3    conda-forge
gtk2                      2.24.33              h539f30e_1    conda-forge
gts                       0.7.6                h64030ff_2    conda-forge
harfbuzz                  3.1.1                h83ec7ef_0    conda-forge
hdf4                      4.2.15               h10796ff_3    conda-forge
hdf5                      1.12.1          nompi_h2750804_103    conda-forge
icu                       68.2                 h9c3ff4c_0    conda-forge
ilmbase                   2.5.5                h780b84a_0    conda-forge
jbig                      2.1               h7f98852_2003    conda-forge
jpeg                      9e                   h7f98852_0    conda-forge
json-c                    0.15                 h98cffda_0    conda-forge
jsoncpp                   1.9.5                h4bd325d_1    conda-forge
jxrlib                    1.1                  h7f98852_2    conda-forge
[... 502 lines cut ...]
 tbb                       2020.2               h4bd325d_4    conda-forge
 tbb-devel                 2020.2               h4bd325d_4    conda-forge
-tiledb                    2.5.3                h2038895_0    conda-forge
+tiledb                    2.6.2                h2038895_1    conda-forge
 tinyxml                   2.6.2                h4bd325d_2    conda-forge
 tinyxml2                  9.0.0                h9c3ff4c_2    conda-forge
 tk                        8.6.11               h27826a3_1    conda-forge
@@ -183,7 +183,7 @@
 wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
[... cut ...]
+xerces-c                  3.2.3                h8ce2273_4    conda-forge
 xorg-fixesproto           5.0               h7f98852_1002    conda-forge
 xorg-inputproto           2.3.2             h7f98852_1002    conda-forge
 xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
@@ -209,5 +209,3 @@
 zlib                      1.2.11            h36c2ea0_1013    conda-forge
 zstd                      1.5.2                ha95c52a_0    conda-forge
 zziplib                   0.13.69              h27826a3_1    conda-forge
-
-
~~~

</details>

traversaro, 2022-02-11:
(Renamed the issue after the recent investigations)

traversaro, 2022-02-11:
I just tested on Windows, and there the `qt-5.12.9-h556501e_5` build of qt is not creating any problem.

Tobias-Fischer, 2022-02-11:
Wow great debugging! I can't wait for the new qt 5.15 ..

traversaro, 2022-02-13:
> Wow great debugging! I can't wait for the new qt 5.15 ..

Yes, I am afraid that looking more deep is going to consume a lot of time with no benefit. To test things before the official qt-main migration, we could do dual builds of gazebo with both qt 5.12 and qt-main 5.15, but we would need to do the same with qwt before.

traversaro, 2022-02-13:
Thanks to https://github.com/conda-forge/gazebo-feedstock/pull/120, the workaround now is simply:
~~~
mamba install gazebo icu=68
~~~
that also have the advantage of being cross-platform.

traversaro, 2022-02-13:
> I've got qwt builds on robostack-experinental channel ..

It seems to be only `osx` for now: https://anaconda.org/robostack-experimental/qwt/, right? Anyhow, I can easily build both qwt and gazebo locally such that they depend on qt-main and see if they work fine.

Tobias-Fischer, 2022-02-13:
Yes correct, only `osx` for now. Thanks for https://github.com/conda-forge/qwt-feedstock/pull/30!

traversaro, 2022-03-29:
On Windows the problem does not seem to be present.

hmaarrfk, 2022-03-29:
I'm not sure what the issue can be. The change from `_4` to `_5` in qt was the introduction of `icu` version 69.

https://github.com/conda-forge/qt-feedstock/pull/214

traversaro, 2022-06-05:
After the `qt515` migration (https://github.com/conda-forge/gazebo-feedstock/pull/131) the default version of gazebo installed will be the one linked against qt 5.15 that is not affected by this issue. I think we can probably wait for a rebuild of ros-noetic and ros galactic (or humble) to definitely close this issue, as until then I guess that many user may install gazebo with qt 5.12 by installing some ros package that use qt 5.12 .

hmaarrfk, 2022-06-05:
I think this is related to #125

traversaro, 2022-06-05:
Even without including ROS in the picture, unfortunatly there are still many cases for which a gazebo that links qt 5.12 is installed. If one tries to install qt-main, libopencv and gazebo, the installation fails with:
~~~
C:\Users\STraversaro>mamba create -n test libopencv gazebo qt-main

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (0.22.1) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['libopencv', 'gazebo', 'qt-main']

conda-forge/win-64                                          Using cache
conda-forge/noarch                                          Using cache
Encountered problems while solving:
  - package libopencv-4.5.1-py38hf7032e7_0 requires qt 5.12.*, but none of the providers can be installed
~~~

After a bit of investigation, I understood that the problem is that gazebo is linking libgdal, that still needs to complete the icu=70 migration, and instead still links icu=69 . Instead libopencv requires qt-main 5.15.3, that requires icu=70. Probably we need to migrate libgdal to icu=70 to improve the situation.

traversaro, 2022-06-05:
> I think this is related to #125

Yes, they seem all related.

traversaro, 2022-06-05:
> After a bit of investigation, I understood that the problem is that gazebo is linking libgdal, that still needs to complete the icu=70 migration, and instead still links icu=69 . Instead libopencv [...]

Indirectly related to https://github.com/conda-forge/xerces-c-feedstock/pull/32#issuecomment-1146603839 .

ngam, 2022-06-06:
gdal to icu 70 is really needed!

Could you help me track this?

gdal (1) Error: not solvable ([bot CI job](https://github.com/regro/autotick-bot/actions/runs/2444598933)): main: ['osx_64_openssl1.1.1: Encountered problems while solving:\n - package xerces-c-3.2.0-0 requires icu 58.*, but none of the providers can be installed\n', 'osx_64_openssl3: Encountered problems while solving:\n - package xerces-c-3.2.0-0 requires icu 58.*, but none of the providers can be installed\n', 'osx_arm64_openssl1.1.1: Encountered problems while solving:\n - package xerces-c-3.2.3-h4fb8152_2 requires icu >=68.1,<69.0a0, but none of the providers can be installed\n', 'osx_arm64_openssl3: Encountered problems while solving:\n - package xerces-c-3.2.3-h4fb8152_2 requires icu >=68.1,<69.0a0, but none of the providers can be installed\n'] 
Immediate Children: qgis
octave (0) Error: not solvable ([bot CI job](https://github.com/regro/autotick-bot/actions/runs/2444215461)): main: ['linux_64_: Encountered problems while solving:\n - package qt-5.12.1-hbfa2a6e_0 requires icu >=64.2,<65.0a0, but none of the providers can be installed\n', 'osx_64_: Encountered problems while solving:\n - package qt-5.12.1-h1b46049_0 requires icu >=64.2,<65.0a0, but none of the providers can be installed\n']

We need to manually open issues

ngam, 2022-06-06:
update, submitted PRs to fix gdal and octave

hmaarrfk, 2022-06-06:
one thing you could do, is to rebuild with a recent version of qt 5.12.

There may be a library that is reexporting symbols accidentally too.. now that one is really hard to find :/

Tobias-Fischer, 2022-09-22:
Shall we close here seeing that qt 5.15 build have landed?

traversaro, 2023-01-20:
> Shall we close here seeing that qt 5.15 build have landed?

Sorry, I never realized that. Probably we can close once we have rebuilds of ROS such that the qt 5.15 builds are the one installed once one runs: `mamba create -n ros-noetic -c conda-forge -c robostack ros-noetic-desktop-full` ?

Until then, I think that keeping the issue open helps its visibility (see https://github.com/RoboStack/ros-noetic/issues/323).

Tobias-Fischer, 2023-05-22:
We now have qt 5.15 builds in robostack-staging, so closing here

closed 2023-05-22 by Tobias-Fischer