conda-forge/gazebo-feedstock #125: Gazebo 11.10.2 fails to load: MergeFrom@Clock@msgs@ignition symbol not found in libignition-transport8

j-rivero, 2022-03-24:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

(disclaimer: excuse me if this is not the right repo to report the problem, I was unable to locate the ignition-transport8 feedstock, is [this one](https://github.com/conda-forge/libignition-transport4-feedstock/)?)

I was able to successfully install Gazebo 11.10.2 using miniconda on Windows10. However it is not able to boot since there are problems with ignition-msgs symbols while loading libignition-transport8. The error message is:

```
MergeFrom@Clock@msgs@ignition@UEAAXEBVAMessage@protobuf@google@Z could not be located in the dynamic library C:\...miniconda3\Library\bin\ignition-transport8.dll
```
[image]

Running the dependency walker shows that there are problems with some of the symbols:

[image]

Symbols seems to be related to protobuf so I wonder if libignition-transport8 needs a rebuild against a new version of Protobuf.

Thanks!

### Installed packages

```shell
# packages in environment at C:\Users\jrivero\miniconda3:
#
# Name                    Version                   Build  Channel
blosc                     1.21.0               h0e60522_0    conda-forge
boost-cpp                 1.74.0               h9f4b32c_7    conda-forge
brotlipy                  0.7.0           py39hb82d6ee_1003    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cairo                     1.16.0            h15b3021_1010    conda-forge
certifi                   2021.10.8        py39hcbf5309_1    conda-forge
cffi                      1.15.0           py39h0878f49_0    conda-forge
cfitsio                   4.1.0                h5a969a9_0    conda-forge
charset-normalizer        2.0.12             pyhd8ed1ab_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
conda                     4.12.0           py39hcbf5309_0    conda-forge
conda-content-trust       0.1.3              pyhd8ed1ab_0    conda-forge
conda-package-handling    1.8.0            py39hb3671d1_0    conda-forge
console_bridge            1.0.2                h5362a0b_0    conda-forge
console_shortcut          0.1.1                         4
cppzmq                    4.7.1                h50e1b43_1    conda-forge
cryptography              36.0.2           py39h7bc7c5c_0    conda-forge
curl                      7.82.0               h789b8ee_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.7                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              hce3cb01_2    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
gazebo                    11.10.2              h6d7d6d0_0    conda-forge
geos                      3.10.2               h39d44d4_0    conda-forge
geotiff                   1.7.0                h38b14a8_7    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.12.1          nompi_h2a0e4a3_104    conda-forge
[... 119 lines cut ...]
                          __win=0=0
                          __archspec=1=x86_64
       base environment : C:\Users\jrivero\miniconda3  (writable)
      conda av data dir : C:\Users\jrivero\miniconda3\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\jrivero\miniconda3\pkgs
                          C:\Users\jrivero\.conda\pkgs
                          C:\Users\jrivero\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\jrivero\miniconda3\envs
                          C:\Users\jrivero\.conda\envs
                          C:\Users\jrivero\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.12.0 requests/2.27.1 CPython/3.9.10 Windows/10 Windows/10.0.19044
          administrator : False
             netrc file : None
           offline mode : False
```

traversaro, 2022-03-29:
> (disclaimer: excuse me if this is not the right repo to report the problem, I was unable to locate the ignition-transport8 feedstock, is [this one](https://github.com/conda-forge/libignition-transport4-feedstock/)?)

Yes, it is that one.

traversaro, 2022-03-29:
I tried to reproduce the problem by creating a new environment, i.e. : 
~~~
mamba create -n gazebo -c conda-forge gazebo
~~~
and also with Python 3.9:
~~~
mamba create -n gazebo -c conda-forge gazebo python=3.9
~~~

I could not reproduce the problem, i.e. Gazebo works fine. @j-rivero it is possible that the problem is due to some strange interaction with the other packages that are already installed in your base environment, see the difference between your environment and mine created by `mamba create -n gazebo -c conda-forge gazebo`: 
~~~diff
--- <unnamed>
+++ <unnamed>
@@ -1,24 +1,15 @@
-# packages in environment at C:\Users\jrivero\miniconda3:
+(gazebo) C:\Users\STraversaro>mamba list
+# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\gazebo:
 #
 # Name                    Version                   Build  Channel
 blosc                     1.21.0               h0e60522_0    conda-forge
 boost-cpp                 1.74.0               h9f4b32c_7    conda-forge
-brotlipy                  0.7.0           py39hb82d6ee_1003    conda-forge
 bzip2                     1.0.8                h8ffe710_4    conda-forge
 ca-certificates           2021.10.8            h5b45459_0    conda-forge
 cairo                     1.16.0            h15b3021_1010    conda-forge
-certifi                   2021.10.8        py39hcbf5309_1    conda-forge
-cffi                      1.15.0           py39h0878f49_0    conda-forge
 cfitsio                   4.1.0                h5a969a9_0    conda-forge
-charset-normalizer        2.0.12             pyhd8ed1ab_0    conda-forge
-colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
-conda                     4.12.0           py39hcbf5309_0    conda-forge
-conda-content-trust       0.1.3              pyhd8ed1ab_0    conda-forge
-conda-package-handling    1.8.0            py39hb3671d1_0    conda-forge
 console_bridge            1.0.2                h5362a0b_0    conda-forge
-console_shortcut          0.1.1                         4
-cppzmq                    4.7.1                h50e1b43_1    conda-forge
-cryptography              36.0.2           py39h7bc7c5c_0    conda-forge
+cppzmq                    4.8.1                h4324990_0    conda-forge
 curl                      7.82.0               h789b8ee_0    conda-forge
 dirent                    1.21                          0    conda-forge
 dlfcn-win32               1.3.0                h0e60522_0    conda-forge
@@ -43,7 +34,6 @@
 hdf4                      4.2.15               h0e5069d_3    conda-forge
 hdf5                      1.12.1          nompi_h2a0e4a3_104    conda-forge
 icu                       69.1                 h0e60522_0    conda-forge
-idna                      3.3                pyhd8ed1ab_0    conda-forge
 ilmbase                   2.5.5                h12d4b20_0    conda-forge
 jbig                      2.1               h8d14728_2003    conda-forge
 jpeg                      9e                   h8ffe710_0    conda-forge
@@ -58,19 +48,19 @@
 libdeflate                1.10                 h8ffe710_0    conda-forge
 libffi                    3.4.2                h8ffe710_5    conda-forge
 libflang                  5.0.0           h6538335_20180525    conda-forge
-libgdal                   3.4.2                ha5e465b_1    conda-forge
+libgdal                   3.4.2                hc8ef25b_3    conda-forge
 libglib                   2.70.2               h3be07f2_4    conda-forge
 libiconv                  1.16                 he774522_0    conda-forge
 libignition-cmake2        2.11.0               h0e60522_0    conda-forge
 libignition-common3       3.13.2               h9bd54ec_1    conda-forge
-libignition-fuel-tools4   4.4.0                h98d6f7e_5    conda-forge
+libignition-fuel-tools4   4.4.0                h98d6f7e_6    conda-forge
 libignition-math6         6.10.0               h0e60522_0    conda-forge
 libignition-msgs5         5.7.0                h5897182_2    conda-forge
-libignition-tools1        1.0.0                h21ff451_0    conda-forge
-libignition-transport8    8.1.0                h7435cc3_4    conda-forge
+libignition-tools1        1.4.1                h0e60522_2    conda-forge
+libignition-transport8    8.1.0               h845afca_10    conda-forge
 libkml                    1.3.0             h9859afa_1014    conda-forge
 libnetcdf                 4.8.1           nompi_h1cc8e9d_101    conda-forge
-libode                    0.16.2               h415ef7b_9    conda-forge
[... 55 lines cut ...]
 tinyxml2                  9.0.0                h0e60522_2    conda-forge
 tk                        8.6.12               h8ffe710_0    conda-forge
-tqdm                      4.63.0             pyhd8ed1ab_0    conda-forge
 tzdata                    2022a                h191b570_0    conda-forge
 ucrt                      10.0.20348.0         h57928b3_0    conda-forge
 urdfdom                   2.3.5                h2d74725_0    conda-forge
 urdfdom_headers           1.0.5                h5362a0b_2    conda-forge
-urllib3                   1.26.9             pyhd8ed1ab_0    conda-forge
 vc                        14.2                 hb210afc_6    conda-forge
 vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
 wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
-win_inet_pton             1.1.0            py39hcbf5309_3    conda-forge
[... cut ...]
+zlib                      1.2.11            h8ffe710_1014    conda-forge
 zstd                      1.5.2                h6255e5f_0    conda-forge
 zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

While in theory conda packages should be installable with arbitrary constraints and everything should work fine, in practice it may be difficult to debug all possible corner case, and it is easier to never install anything in the `base` environment, and always created new environments (different from `base`) for actual use.

johnwason, 2022-03-31:
I am seeing this problem as well in a clean environment.

johnwason, 2022-03-31:
I have tried older versions 11.10.1, 11.10.0, and 11.9.1 of the gazebo package and am seeing the same error. The problem may not be in the gazebo package.

traversaro, 2022-03-31:
> I am seeing this problem as well in a clean environment.

Can you report the environent that you get by installing gazebo in a clean environment that result in this error? Thanks.

johnwason, 2022-03-31:
Here is my environment after `conda create -n gazebo gazebo`

```
# packages in environment at C:\tools\miniconda3\envs\gazebo:
#
# Name                    Version                   Build  Channel
blosc                     1.21.0               h0e60522_0    conda-forge
boost-cpp                 1.74.0               h9f4b32c_7    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cairo                     1.16.0            h15b3021_1010    conda-forge
cfitsio                   4.1.0                h5a969a9_0    conda-forge
console_bridge            1.0.2                h5362a0b_0    conda-forge
cppzmq                    4.7.1                h50e1b43_1    conda-forge
curl                      7.82.0               h789b8ee_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.8                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              hce3cb01_2    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
gazebo                    11.10.2              h6d7d6d0_0    conda-forge
geos                      3.10.2               h39d44d4_0    conda-forge
geotiff                   1.7.1                h38b14a8_1    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.12.1          nompi_h2a0e4a3_104    conda-forge
icu                       69.1                 h0e60522_0    conda-forge
ilmbase                   2.5.5                h12d4b20_0    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jpeg                      9e                   h8ffe710_0    conda-forge
jsoncpp                   1.9.5                h2d74725_1    conda-forge
jxrlib                    1.1                  h8ffe710_2    conda-forge
kealib                    1.4.14               h8995ca9_3    conda-forge
krb5                      1.19.3               h1176d77_0    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
lerc                      3.0                  h0e60522_0    conda-forge
libclang                  13.0.1          default_h81446c8_0    conda-forge
libcurl                   7.82.0               h789b8ee_0    conda-forge
libdeflate                1.10                 h8ffe710_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libgdal                   3.4.2                h0bdba65_4    conda-forge
libglib                   2.70.2               h3be07f2_4    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libignition-cmake2        2.11.0               h0e60522_0    conda-forge
libignition-common3       3.13.2               h9bd54ec_1    conda-forge
libignition-fuel-tools4   4.4.0                h98d6f7e_5    conda-forge
libignition-math6         6.10.0               h0e60522_0    conda-forge
libignition-msgs5         5.7.0                h5897182_2    conda-forge
libignition-tools1        1.0.0                h21ff451_0    conda-forge
libignition-transport8    8.1.0                h7435cc3_4    conda-forge
libkml                    1.3.0             h9859afa_1014    conda-forge
libnetcdf                 4.8.1           nompi_h1cc8e9d_101    conda-forge
libode                    0.16.2               h8a704f9_9    conda-forge
libopenblas               0.3.18          pthreads_hc469a61_0    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libpq                     14.2                 hfcc5ef8_0    conda-forge
libprotobuf               3.19.4               h7755175_0    conda-forge
libraw                    0.20.2               hee1bdec_1    conda-forge
[... 33 lines cut ...]
setuptools                61.2.0          py310h5588dad_3    conda-forge
simbody                   3.7                  hade3207_3    conda-forge
sqlite                    3.37.1               h8ffe710_0    conda-forge
tbb                       2021.5.0             h2d74725_0    conda-forge
tbb-devel                 2021.5.0             h2d74725_0    conda-forge
tiledb                    2.7.1                h95dad36_0    conda-forge
tiny-process-library      2.0.4                h0e60522_0    conda-forge
tinyxml                   2.6.2                h2d74725_2    conda-forge
[... cut ...]
xerces-c                  3.2.3                h0e60522_4    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.3                h0e60522_3    conda-forge
zlib                      1.2.11            h8ffe710_1014    conda-forge
zstd                      1.5.2                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

johnwason, 2022-03-31:
I copied the environment from my laptop that I haven't updated for a while, and gazebo runs normally. There are a few other packages installed. Here is the environment for reference:

```
# packages in environment at C:\tools\miniconda3\envs\rr_training_sim_old:
#
# Name                    Version                   Build  Channel
boost-cpp                 1.74.0               h5b4e17d_5    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cairo                     1.16.0            hb19e0ff_1008    conda-forge
cfitsio                   3.470                h0af3d06_7    conda-forge
console_bridge            1.0.1                h5362a0b_0    conda-forge
cppzmq                    4.8.1                h4324990_0    conda-forge
curl                      7.80.0               h789b8ee_0    conda-forge
cvxopt                    1.2.7            py39h73f7814_1    conda-forge
cython                    0.29.24          py39h415ef7b_1    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
dotnet-runtime            6.0.0                h2d74725_1    conda-forge
dotnet-sdk                6.0.100              h2d74725_1    conda-forge
dsdp                      5.8               h6e01ec9_1203    conda-forge
ecos                      2.0.8            py39h5d4886f_2    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.1                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
fftw                      3.3.10          nompi_hb424310_101    conda-forge
fontconfig                2.13.1            h1989441_1005    conda-forge
freeglut                  3.2.1                h0e60522_2    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
gazebo                    11.9.0               hbd30de6_0    conda-forge
gazebo_model_robotraconteur_driver 0.1.3.1                       0    robotraconteur
gazebo_robotraconteur_server_plugin 0.1.3.3              h3ad6029_6    robotraconteur
geos                      3.9.1                h39d44d4_2    conda-forge
geotiff                   1.6.0                ha8a8a2d_6    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
glpk                      4.65              h8ffe710_1004    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.10.6          nompi_h5268f04_1114    conda-forge
icu                       68.2                 h0e60522_0    conda-forge
ilmbase                   2.5.5                h12d4b20_0    conda-forge
importlib_resources       5.4.0              pyhd8ed1ab_0    conda-forge
intel-openmp              2021.4.0          h57928b3_3556    conda-forge
jasper                    2.0.33               h77af90b_0    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
jsoncpp                   1.9.4                h2d74725_3    conda-forge
jxrlib                    1.1                  h8ffe710_2    conda-forge
kealib                    1.4.14               h96bfa42_2    conda-forge
krb5                      1.19.2               h20d022d_3    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
lerc                      3.0                  h0e60522_0    conda-forge
libblas                   3.9.0                     8_mkl    conda-forge
libcblas                  3.9.0                     8_mkl    conda-forge
libclang                  11.1.0          default_h5c34c98_1    conda-forge
libcurl                   7.80.0               h789b8ee_0    conda-forge
libdeflate                1.8                  h8ffe710_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libgdal                   3.3.1                ha6cada3_1    conda-forge
libglib                   2.70.1               h3be07f2_0    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libignition-cmake2        2.9.0                h0e60522_1    conda-forge
libignition-common3       3.13.2               h9bd54ec_1    conda-forge
libignition-fuel-tools4   4.4.0                he2780a4_2    conda-forge
libignition-math6         6.9.2                h0e60522_0    conda-forge
libignition-msgs5         5.7.0                hb9694b1_1    conda-forge
[... 68 lines cut ...]
sqlite                    3.36.0               h8ffe710_2    conda-forge
tbb                       2020.2               h2d74725_4    conda-forge
tbb-devel                 2020.2               h2d74725_4    conda-forge
tiledb                    2.3.4                h78dabda_0    conda-forge
tiny-process-library      2.0.4                h0e60522_0    conda-forge
[... cut ...]
yaml                      0.2.5                he774522_0    conda-forge
yaml-cpp                  0.6.3                ha925a31_4    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zipp                      3.6.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11            h8ffe710_1013    conda-forge
zstd                      1.5.0                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

traversaro, 2022-04-02:
I was able to reproduce the problem. Basically `conda create -n gazeboconda gazebo` and `mamba create -n gazeboconda gazebo` resolve to different environments, and the mamba version is working while the conda version is not working.

traversaro, 2022-04-02:
The difference between the two environements is:
~~~diff
--- C:\src\gzws\gazeboconda
+++ C:\src\gzws\gazebomamba
@@ -1,4 +1,4 @@
-# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\gazeboconda:
+# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\gazebomamba:
 #
 # Name                    Version                   Build  Channel
 blosc                     1.21.0               h0e60522_0    conda-forge
@@ -8,7 +8,7 @@
 cairo                     1.16.0            h15b3021_1010    conda-forge
 cfitsio                   4.1.0                h5a969a9_0    conda-forge
 console_bridge            1.0.2                h5362a0b_0    conda-forge
-cppzmq                    4.7.1                h50e1b43_1    conda-forge
+cppzmq                    4.8.1                h4324990_0    conda-forge
 curl                      7.82.0               h789b8ee_0    conda-forge
 dirent                    1.21                          0    conda-forge
 dlfcn-win32               1.3.0                h0e60522_0    conda-forge
@@ -52,11 +52,11 @@
 libiconv                  1.16                 he774522_0    conda-forge
 libignition-cmake2        2.11.0               h0e60522_0    conda-forge
 libignition-common3       3.13.2               h9bd54ec_1    conda-forge
-libignition-fuel-tools4   4.4.0                h98d6f7e_5    conda-forge
+libignition-fuel-tools4   4.4.0                h98d6f7e_6    conda-forge
 libignition-math6         6.10.0               h0e60522_0    conda-forge
 libignition-msgs5         5.7.0                h5897182_2    conda-forge
-libignition-tools1        1.0.0                h21ff451_0    conda-forge
-libignition-transport8    8.1.0                h7435cc3_4    conda-forge
+libignition-tools1        1.4.1                h0e60522_2    conda-forge
+libignition-transport8    8.1.0               h845afca_10    conda-forge
 libkml                    1.3.0             h9859afa_1014    conda-forge
 libnetcdf                 4.8.1           nompi_h1cc8e9d_101    conda-forge
 libode                    0.16.2               h8a704f9_9    conda-forge
@@ -97,6 +97,7 @@
 python_abi                3.10                    2_cp310    conda-forge
 qt                        5.12.9               h556501e_6    conda-forge
 qwt                       6.2.0                hc28b1ea_2    conda-forge
+ruby                      2.7.2                h1d9c0f5_8    conda-forge
 sdl2                      2.0.18               h0e60522_0    conda-forge
 setuptools                61.3.0          py310h5588dad_0    conda-forge
 simbody                   3.7                  hade3207_3    conda-forge
@@ -118,7 +119,7 @@
 xerces-c                  3.2.3                h0e60522_4    conda-forge
 xz                        5.2.5                h62dcd97_1    conda-forge
 yaml                      0.2.5                h8ffe710_2    conda-forge
-zeromq                    4.3.3                h0e60522_3    conda-forge
+zeromq                    4.3.4                h0e60522_1    conda-forge
 zlib                      1.2.11            h8ffe710_1014    conda-forge
 zstd                      1.5.2                h6255e5f_0    conda-forge
 zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

As soon as one explicitly ask the correct version of `libignition-fuel-tools` or `libignition-transport`, the correct dependency are resolved:
~~~
conda create -n gazebocondapinfueltools gazebo libignition-fuel-tools4=4.4.0=h98d6f7e_6
~~~
or: 
~~~
conda create -n gazebocondapintransport gazebo libignition-transport8=8.1.0=h845afca_10
~~~

traversaro, 2022-04-02:
Out of those versions, for sure `libignition-fuel-tools4=4.4.0=h98d6f7e_5` is strange, as it was built against a broken libyaml: https://github.com/conda-forge/libignition-fuel-tools-feedstock/pull/61 . Probably a first attempt could be to mark that build as broken.

traversaro, 2022-04-02:
> Probably a first attempt could be to mark that build as broken.

Done in https://github.com/conda-forge/admin-requests/pull/421 .

johnwason, 2022-04-09:
I am still seeing this error.

Tobias-Fischer, 2022-04-09:
Probably because https://github.com/conda-forge/admin-requests/pull/421 hasn't been merged yet

traversaro, 2022-04-11:
Even after https://github.com/conda-forge/admin-requests/pull/421, the problem remains. Now the difference between conda (not working) and mamba (working) is:
~~~diff
--- <unnamed>
+++ <unnamed>
@@ -1,5 +1,5 @@
-(gazeboapril) C:\Users\STraversaro>mamba list
-# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\gazeboapril:
+(gazeboaprilmamba) C:\Users\STraversaro>mamba list
+# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\gazeboaprilmamba:
 #
 # Name                    Version                   Build  Channel
 blosc                     1.21.0               h0e60522_0    conda-forge
@@ -9,7 +9,7 @@
 cairo                     1.16.0            h15b3021_1010    conda-forge
 cfitsio                   4.1.0                h5a969a9_0    conda-forge
 console_bridge            1.0.2                h5362a0b_1    conda-forge
-cppzmq                    4.7.1                h50e1b43_1    conda-forge
+cppzmq                    4.8.1                h4324990_0    conda-forge
 curl                      7.82.0               h789b8ee_0    conda-forge
 dirent                    1.21                          0    conda-forge
 dlfcn-win32               1.3.0                h0e60522_0    conda-forge
@@ -53,11 +53,11 @@
 libiconv                  1.16                 he774522_0    conda-forge
 libignition-cmake2        2.11.0               h0e60522_0    conda-forge
 libignition-common3       3.13.2               h9bd54ec_1    conda-forge
-libignition-fuel-tools4   4.4.0                h98d6f7e_4    conda-forge
+libignition-fuel-tools4   4.4.0                h98d6f7e_6    conda-forge
 libignition-math6         6.10.0               h0e60522_0    conda-forge
 libignition-msgs5         5.7.0                h5897182_2    conda-forge
-libignition-tools1        1.0.0                h21ff451_0    conda-forge
-libignition-transport8    8.1.0                h7435cc3_4    conda-forge
+libignition-tools1        1.4.1                h0e60522_2    conda-forge
+libignition-transport8    8.1.0               h845afca_10    conda-forge
 libkml                    1.3.0             h9859afa_1014    conda-forge
 libnetcdf                 4.8.1           nompi_h1cc8e9d_101    conda-forge
 libode                    0.16.2               h8a704f9_9    conda-forge
@@ -98,6 +98,7 @@
 python_abi                3.10                    2_cp310    conda-forge
 qt                        5.12.9               h556501e_6    conda-forge
 qwt                       6.2.0                hc28b1ea_2    conda-forge
+ruby                      2.7.2                h1d9c0f5_8    conda-forge
 sdl2                      2.0.18               h0e60522_0    conda-forge
 setuptools                62.0.0          py310h5588dad_0    conda-forge
 simbody                   3.7                  hade3207_3    conda-forge
@@ -119,7 +120,7 @@
 xerces-c                  3.2.3                h0e60522_4    conda-forge
 xz                        5.2.5                h62dcd97_1    conda-forge
 yaml                      0.2.5                h8ffe710_2    conda-forge
-zeromq                    4.3.3                h0e60522_3    conda-forge
+zeromq                    4.3.4                h0e60522_1    conda-forge
 zlib                      1.2.11            h8ffe710_1014    conda-forge
 zstd                      1.5.2                h6255e5f_0    conda-forge
 zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

I still have no clue of the root issue, but a possible way of changing things to see if the issue goes away is to release an update of gazebo11 dependencies. We need to do that anyhow for the protobuf 3.20 migration, but that is currently (indirectly) blocked by https://github.com/ignitionrobotics/ign-msgs/issues/240 .

traversaro, 2022-04-21:
Even after protobuf 3.20 migration there is a difference between conda and mamba, actually conda does not even install libprotobuf 3.20 or the latest build of Gazebo:
~~~diff
--- C:\Users\STraversaro\gazeboconda
+++ C:\Users\STraversaro\gazebomamba
@@ -1,4 +1,4 @@
-# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\gazebo:
+# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\gazebomamba:
 #
 # Name                    Version                   Build  Channel
 blosc                     1.21.0               h0e60522_0    conda-forge
@@ -8,7 +8,7 @@
 cairo                     1.16.0            h15b3021_1010    conda-forge
 cfitsio                   4.1.0                h5a969a9_0    conda-forge
 console_bridge            1.0.2                h5362a0b_1    conda-forge
-cppzmq                    4.7.1                h50e1b43_1    conda-forge
+cppzmq                    4.8.1                h4324990_0    conda-forge
 curl                      7.82.0               h789b8ee_0    conda-forge
 dirent                    1.21                          0    conda-forge
 dlfcn-win32               1.3.0                h0e60522_0    conda-forge
@@ -25,7 +25,7 @@
 freeimage                 3.18.0               h6676e37_8    conda-forge
 freetype                  2.10.4               h546665d_1    conda-forge
 freexl                    1.0.6                ha8e266a_0    conda-forge
-gazebo                    11.10.2              h6d7d6d0_1    conda-forge
+gazebo                    11.10.2              hc81694e_2    conda-forge
 geos                      3.10.2               h39d44d4_0    conda-forge
 geotiff                   1.7.1                h38b14a8_1    conda-forge
 gettext                   0.19.8.1          ha2e2712_1008    conda-forge
@@ -52,18 +52,18 @@
 libiconv                  1.16                 he774522_0    conda-forge
 libignition-cmake2        2.12.1               h0e60522_0    conda-forge
 libignition-common3       3.13.2               h9bd54ec_2    conda-forge
-libignition-fuel-tools4   4.4.0                h98d6f7e_4    conda-forge
+libignition-fuel-tools4   4.4.0                h3fac903_7    conda-forge
 libignition-math6         6.10.0               h0e60522_0    conda-forge
-libignition-msgs5         5.7.0                h5897182_2    conda-forge
-libignition-tools1        1.0.0                h21ff451_0    conda-forge
-libignition-transport8    8.1.0                h7435cc3_4    conda-forge
+libignition-msgs5         5.9.0                hc835698_0    conda-forge
+libignition-tools1        1.4.1                h0e60522_2    conda-forge
+libignition-transport8    8.1.0               h819de12_11    conda-forge
 libkml                    1.3.0             h9859afa_1014    conda-forge
 libnetcdf                 4.8.1           nompi_h1cc8e9d_101    conda-forge
 libode                    0.16.2              h8a704f9_10    conda-forge
 libopenblas               0.3.20          pthreads_hc469a61_0    conda-forge
 libpng                    1.6.37               h1d00b33_2    conda-forge
 libpq                     14.2                 hfcc5ef8_0    conda-forge
-libprotobuf               3.19.4               h7755175_0    conda-forge
+libprotobuf               3.20.0               h7755175_0    conda-forge
 libraw                    0.20.2               hee1bdec_1    conda-forge
 librttopo                 1.1.0                hb1df466_9    conda-forge
 libsdformat               9.3.0                h0e60522_5    conda-forge
@@ -97,6 +97,7 @@
 python_abi                3.10                    2_cp310    conda-forge
 qt                        5.12.9               h556501e_6    conda-forge
 qwt                       6.2.0                hc28b1ea_2    conda-forge
+ruby                      2.7.2                h1d9c0f5_8    conda-forge
 sdl2                      2.0.18               h0e60522_0    conda-forge
 setuptools                62.1.0          py310h5588dad_0    conda-forge
 simbody                   3.7                  hade3207_3    conda-forge
@@ -118,7 +119,7 @@
 xerces-c                  3.2.3                h0e60522_4    conda-forge
 xz                        5.2.5                h62dcd97_1    conda-forge
 yaml                      0.2.5                h8ffe710_2    conda-forge
-zeromq                    4.3.3                h0e60522_3    conda-forge
+zeromq                    4.3.4                h0e60522_1    conda-forge
 zlib                      1.2.11            h8ffe710_1014    conda-forge
 zstd                      1.5.2                h6255e5f_0    conda-forge
 zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

traversaro, 2022-04-21:
By the way, using the new libmamba backend indeed installs the correct latest version of Gazebo:
~~~
C:\Users\STraversaro>conda create -n gazebocondalibmamba gazebo --experimental-solver=libmamba

***

NOTE: You are using the EXPERIMENTAL libmamba solver integration.

If something is not working as expected, please:

1. Go to https://github.com/conda/conda/issues/new/choose
2. Choose the "Libmamba Solver Feedback (Experimental Feature)" option
3. Attach the log file found in the following path:

C:\Users\STraversaro\AppData\Local\mambaforge\pkgs\.logs\20220421-094122-520121.log

Thank you for your help!

***

conda-forge/noarch                                   7.9MB @   3.1MB/s  3.5s
conda-forge/win-64                                  14.5MB @   2.8MB/s  7.2s
Collect all metadata: done
Solving environment: done

## Package Plan ##

  environment location: C:\Users\STraversaro\AppData\Local\mambaforge\envs\gazebocondalibmamba

  added / updated specs:
    - gazebo

The following NEW packages will be INSTALLED:

  blosc              conda-forge/win-64::blosc-1.21.0-h0e60522_0
  boost-cpp          conda-forge/win-64::boost-cpp-1.74.0-h9f4b32c_8
  bzip2              conda-forge/win-64::bzip2-1.0.8-h8ffe710_4
  ca-certificates    conda-forge/win-64::ca-certificates-2021.10.8-h5b45459_0
  cairo              conda-forge/win-64::cairo-1.16.0-h15b3021_1010
  cfitsio            conda-forge/win-64::cfitsio-4.1.0-h5a969a9_0
  console_bridge     conda-forge/win-64::console_bridge-1.0.2-h5362a0b_1
  cppzmq             conda-forge/win-64::cppzmq-4.8.1-h4324990_0
  curl               conda-forge/win-64::curl-7.82.0-h789b8ee_0
  dirent             conda-forge/win-64::dirent-1.21-0
  dlfcn-win32        conda-forge/win-64::dlfcn-win32-1.3.0-h0e60522_0
  eigen              conda-forge/win-64::eigen-3.4.0-h2d74725_0
  expat              conda-forge/win-64::expat-2.4.8-h39d44d4_0
  ffmpeg             conda-forge/win-64::ffmpeg-4.3.1-ha925a31_0
  font-ttf-dejavu-s~ conda-forge/noarch::font-ttf-dejavu-sans-mono-2.37-hab24e00_0
  font-ttf-inconsol~ conda-forge/noarch::font-ttf-inconsolata-3.000-h77eed37_0
  font-ttf-source-c~ conda-forge/noarch::font-ttf-source-code-pro-2.038-h77eed37_0
  font-ttf-ubuntu    conda-forge/noarch::font-ttf-ubuntu-0.83-hab24e00_0
  fontconfig         conda-forge/win-64::fontconfig-2.14.0-hce3cb01_0
  fonts-conda-ecosy~ conda-forge/noarch::fonts-conda-ecosystem-1-0
  fonts-conda-forge  conda-forge/noarch::fonts-conda-forge-1-0
  freeimage          conda-forge/win-64::freeimage-3.18.0-h6676e37_8
  freetype           conda-forge/win-64::freetype-2.10.4-h546665d_1
  freexl             conda-forge/win-64::freexl-1.0.6-ha8e266a_0
  gazebo             conda-forge/win-64::gazebo-11.10.2-hc81694e_2
  geos               conda-forge/win-64::geos-3.10.2-h39d44d4_0
  geotiff            conda-forge/win-64::geotiff-1.7.1-h38b14a8_1
  gettext            conda-forge/win-64::gettext-0.19.8.1-ha2e2712_1008
  gts                conda-forge/win-64::gts-0.7.6-h7c369d9_2
  hdf4               conda-forge/win-64::hdf4-4.2.15-h0e5069d_3
  hdf5               conda-forge/win-64::hdf5-1.12.1-nompi_h2a0e4a3_104
  icu                conda-forge/win-64::icu-69.1-h0e60522_0
  imath              conda-forge/win-64::imath-3.1.5-h12d4b20_0
  jbig               conda-forge/win-64::jbig-2.1-h8d14728_2003
  jpeg               conda-forge/win-64::jpeg-9e-h8ffe710_1
  jsoncpp            conda-forge/win-64::jsoncpp-1.9.5-h2d74725_1
[... 62 lines cut ...]
  setuptools         conda-forge/win-64::setuptools-62.1.0-py310h5588dad_0
  simbody            conda-forge/win-64::simbody-3.7-hade3207_3
  sqlite             conda-forge/win-64::sqlite-3.38.2-h8ffe710_0
  tbb                conda-forge/win-64::tbb-2021.5.0-h2d74725_1
  tbb-devel          conda-forge/win-64::tbb-devel-2021.5.0-h2d74725_1
  tiledb             conda-forge/win-64::tiledb-2.8.0-h5689973_1
  tiny-process-libr~ conda-forge/win-64::tiny-process-library-2.0.4-h0e60522_0
  tinyxml            conda-forge/win-64::tinyxml-2.6.2-h2d74725_2
  tinyxml2           conda-forge/win-64::tinyxml2-9.0.0-h0e60522_2
  tk                 conda-forge/win-64::tk-8.6.12-h8ffe710_0
  tzdata             conda-forge/noarch::tzdata-2022a-h191b570_0
  ucrt               conda-forge/win-64::ucrt-10.0.20348.0-h57928b3_0
  urdfdom            conda-forge/win-64::urdfdom-2.3.5-h2d74725_0
  urdfdom_headers    conda-forge/win-64::urdfdom_headers-1.0.6-h5362a0b_2
  vc                 conda-forge/win-64::vc-14.2-hb210afc_6
  vs2015_runtime     conda-forge/win-64::vs2015_runtime-14.29.30037-h902a5da_6
  wheel              conda-forge/noarch::wheel-0.37.1-pyhd8ed1ab_0
  xerces-c           conda-forge/win-64::xerces-c-3.2.3-h0e60522_4
  xz                 conda-forge/win-64::xz-5.2.5-h62dcd97_1
  yaml               conda-forge/win-64::yaml-0.2.5-h8ffe710_2
  zeromq             conda-forge/win-64::zeromq-4.3.4-h0e60522_1
  zlib               conda-forge/win-64::zlib-1.2.11-h8ffe710_1014
  zstd               conda-forge/win-64::zstd-1.5.2-h6255e5f_0
  zziplib            conda-forge/win-64::zziplib-0.13.69-h1d00b33_1
~~~

PR #139 (2022-06-13): Fix #136

Checklist
* [x] Used a [personal fork of the feedstock to propose changes](https://conda-forge.org/docs/maintainer/updating_pkgs.html#forking-and-pull-requests)
* [x] Bumped the build number (if the version is unchanged)
* [ ] Reset the build number to `0` (if the version changed)
* [x] [Re-rendered]( https://conda-forge.org/docs/maintainer/updating_pkgs.html#rerendering-feedstocks ) with the latest `conda-smithy` (Use the phrase <code>@<space/>conda-forge-admin, please rerender</code> in a comment in this PR for automated rerendering)
* [ ] Ensured the license file is being packaged [... cut]
closed 2022-06-13 by Tobias-Fischer