conda-forge/gazebo-feedstock #148: MacOS - Gazebo: Cannot Add Model to World

traversaro, 2022-09-07:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Even if a user starts `gazebo` from the terminal, the GUI gets launched, but apparently no simulation is running. This has been already reported in https://github.com/RoboStack/ros-noetic/issues/279, but with @nunoguedelha we reproduced it without any ROS involved.

### Installed packages

```shell
.
```

### Environment info

```shell
.
```

traversaro, 2022-09-07:
@nunoguedelha can you report the installed packages (`mamba list`) in the working env and in the not working one?

traversaro, 2022-09-07:
I reproduce the issue on macOS on an M1 machine, i.e. `arm64`. This are the packages installed:

nunoguedelha, 2022-09-07:
> 

My setup information...

Conda info and setup from an environment where Gazebo (11.9.1) is working:

<details>
<summary>$ conda info</summary>

```bash
     active environment : robotologydeps
    active env location : /Users/nunoguedelha/miniforge3/envs/robotologydeps
            shell level : 1
       user config file : /Users/nunoguedelha/.condarc
 populated config files : /Users/nunoguedelha/miniforge3/.condarc
                          /Users/nunoguedelha/.condarc
          conda version : 4.14.0
    conda-build version : not installed
         python version : 3.8.12.final.0
       virtual packages : __osx=10.15.7=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /Users/nunoguedelha/miniforge3  (writable)
      conda av data dir : /Users/nunoguedelha/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/nunoguedelha/miniforge3/pkgs
                          /Users/nunoguedelha/.conda/pkgs
       envs directories : /Users/nunoguedelha/miniforge3/envs
                          /Users/nunoguedelha/.conda/envs
               platform : osx-64
             user-agent : conda/4.14.0 requests/2.28.1 CPython/3.8.12 Darwin/19.6.0 OSX/10.15.7
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

</details>

<details>
<summary>$ conda list</summary>

```bash
# packages in environment at /Users/nunoguedelha/miniforge3/envs/robotologydeps:
#
# Name                    Version                   Build  Channel
ace                       7.0.6                he49afe7_0    conda-forge
ampl-mp                   3.1.0             h2beb688_1005    conda-forge
aom                       3.2.0                he49afe7_2    conda-forge
asio                      1.14.1               he49afe7_1    conda-forge
assimp                    5.0.1                h1224e73_7    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
boost                     1.74.0           py39ha1f3e3e_4    conda-forge
boost-cpp                 1.74.0               hff03dee_4    conda-forge
bullet-cpp                3.17                 h428e3ee_1    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
c-compiler                1.3.0                h0d85af4_0    conda-forge
ca-certificates           2021.10.8            h033912b_0    conda-forge
cairo                     1.16.0            he43a7df_1008    conda-forge
cctools                   973.0.1              hd9211c8_5    conda-forge
cctools_osx-64            973.0.1              hb41342b_5    conda-forge
cfitsio                   4.0.0                hb20e66c_0    conda-forge
clang                     11.1.0               h694c41f_1    conda-forge
clang-11                  11.1.0          default_he082bbe_1    conda-forge
[... 536 lines cut ...]
xorg-libxaw               1.0.14               h0d85af4_1    conda-forge
xorg-libxdmcp             1.1.3                h35c211d_0    conda-forge
xorg-libxext              1.3.4                h0d85af4_1    conda-forge
xorg-libxfixes            5.0.3             h0d85af4_1004    conda-forge
xorg-libxmu               1.1.3                h0d85af4_0    conda-forge
xorg-libxpm               3.5.13               h0d85af4_0    conda-forge
xorg-libxrender           0.9.10            h0d85af4_1003    conda-forge
xorg-libxt                1.2.1                h0d85af4_2    conda-forge
xorg-renderproto          0.11.1            h0d85af4_1002    conda-forge
xorg-xextproto            7.3.0             h35c211d_1002    conda-forge
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.6                h775f41a_0    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
yaml-cpp                  0.7.0                hb486fe8_1    conda-forge
yarp                      3.7.2                h694c41f_0    conda-forge
yarp-cxx                  3.7.2                h247d3f9_0    conda-forge
yarp-python               3.7.2           py310hd4537e4_0    conda-forge
ycm-cmake-modules         0.14.2               hb486fe8_0    conda-forge
zeromq                    4.3.4                he49afe7_1    conda-forge
zlib                      1.2.12               hfe4f2af_2    conda-forge
zstd                      1.5.2                hfa58983_4    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
```

</details>

nunoguedelha, 2022-09-07:
I've tried to fall back to Gazebo 11.10.0 , running:
```sh
mamba clean --all
mamba install gazebo=11.10.0
```

but I then stumbled on the problem:
```sh
$ gzserver --verbose
Gazebo multi-robot simulator, version 11.10.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.70
[Err] [server_main.cc:57] Ogre Error:ItemIdentityException: Resource with the name shadow_caster_vp_glsl already exists. in ResourceManager::add at /Users/runner/miniforge3/conda-bld/ogre_1655955436130/work/OgreMain/src/OgreResourceManager.cpp (line 158)
```
Already tracked in https://github.com/conda-forge/gazebo-feedstock/issues/119.

Apparently the issue was due to QT 5.12, so after installing `qt-main` (which upgrades qt to ver 5.15), I've installed **Gazebo 11.10.2**.

nunoguedelha, 2022-09-07:
Same problem with Gazebo 11.10.2, the simulator doesn't run after launching the server.

nunoguedelha, 2022-09-07:
Actually same problem for Gazebo ≥ 11.10.1.

traversaro, 2022-09-08:
I reproduced the same problem compiling gazebo (commit https://github.com/osrf/gazebo/commit/92f1b42efcbcf53f63e9904da7bb090a926eecf9) from source.

nunoguedelha, 2022-09-08:
> I reproduced the same problem compiling gazebo (commit [osrf/gazebo@92f1b42](https://github.com/osrf/gazebo/commit/92f1b42efcbcf53f63e9904da7bb090a926eecf9)).
> 
> To reproduce (dependencies from https://github.com/osrf/gazebo/blob/bf205dd8cf67c0ab0f61560d875f5eb5118fe9e1/.github/workflows/conda-forge.yml):

I've tried the same but I'm having lots of compile errors when running `ninja` due to Protobuf headers. Those are actually being found in the system (`/usr/local/include/google/protobuf`) instead of Conda install prefix, which can be seen in the Cmake cache:
```
 Protobuf_DIR                     Protobuf_DIR-NOTFOUND
 Protobuf_INCLUDE_DIR             /usr/local/include
 Protobuf_LIBRARY_DEBUG           /usr/local/lib/libprotobuf.dylib
 Protobuf_LIBRARY_RELEASE         /usr/local/lib/libprotobuf.dylib
 Protobuf_LITE_LIBRARY_DEBUG      /usr/local/lib/libprotobuf-lite.dylib
 Protobuf_LITE_LIBRARY_RELEASE    /usr/local/lib/libprotobuf-lite.dylib
 Protobuf_PROTOC_EXECUTABLE       /Users/nunoguedelha/miniforge3/envs/robotologybin3/bin/protoc
 Protobuf_PROTOC_LIBRARY_DEBUG    /usr/local/lib/libprotoc.dylib
 Protobuf_PROTOC_LIBRARY_RELEAS   /usr/local/lib/libprotoc.dylib
```

I could manually make these variables point to `/Users/nunoguedelha/miniforge3/envs/robotologybin3/include` and `/Users/nunoguedelha/miniforge3/envs/robotologybin3/lib/libprotobufxxx.dylib` but I guess it's better to do it in the `CMakeLists.txt`...

traversaro, 2022-09-08:
I also had this problem with tags `gazebo11_11.9.0` and  `gazebo11_11.9.1` with this env:
~~~
(gazebodev119) icub@iCubs-Mac-mini ~ % mamba list
# packages in environment at /Users/icub/mambaforge/envs/gazebodev119:
#
# Name                    Version                   Build  Channel
aom                       3.4.0                hbe31e94_1    conda-forge
assimp                    5.2.4                h3287542_0    conda-forge
atk-1.0                   2.36.0               heb41896_4    conda-forge
blosc                     1.21.1               hd414afc_3    conda-forge
boost-cpp                 1.74.0               h1cb353e_8    conda-forge
bullet-cpp                3.21            py310ha6a5cd6_3    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.18.1               h3422bc3_0    conda-forge
c-compiler                1.5.0                h5008568_0    conda-forge
ca-certificates           2022.6.15            h4653dfc_0    conda-forge
cairo                     1.16.0            h73a0509_1013    conda-forge
cctools                   973.0.1             hcbb26d4_10    conda-forge
cctools_osx-arm64         973.0.1             he930820_10    conda-forge
cfitsio                   4.1.0                hd4f5c17_0    conda-forge
clang                     14.0.4               hce30654_0    conda-forge
clang-14                  14.0.4          default_h81a5282_0    conda-forge
clang_osx-arm64           14.0.4               hc634a12_2    conda-forge
clangxx                   14.0.4          default_hb7ecf47_0    conda-forge
clangxx_osx-arm64         14.0.4               hb7a33d8_2    conda-forge
cmake                     3.24.1               h967666c_0    conda-forge
compiler-rt               14.0.4               h4418234_0    conda-forge
compiler-rt_osx-arm64     14.0.4               hb07f8d1_0    conda-forge
compilers                 1.5.0                hce30654_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
cppzmq                    4.8.1                h3212551_0    conda-forge
curl                      7.83.1               h2fcd78c_0    conda-forge
cxx-compiler              1.5.0                hffc8910_0    conda-forge
dartsim                   6.12.1               h217873d_5    conda-forge
eigen                     3.4.0                hc021e02_0    conda-forge
expat                     2.4.8                h6b3803e_0    conda-forge
fcl                       0.6.1                hda9dc21_3    conda-forge
ffmpeg                    4.4.2           gpl_hfdc7bce_107    conda-forge
flann                     1.9.1             hd3e9afc_1011    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hfb34624_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.5.0                h2ccabda_0    conda-forge
freeimage                 3.18.0              hb513136_10    conda-forge
freetype                  2.12.1               hd633e50_0    conda-forge
freexl                    1.0.6                h27ca646_0    conda-forge
fribidi                   1.0.10               h27ca646_0    conda-forge
gdk-pixbuf                2.42.8               hdded42e_0    conda-forge
geos                      3.11.0               h9a09cb3_0    conda-forge
geotiff                   1.7.1                hc898e3f_3    conda-forge
gettext                   0.19.8.1          h049c9fb_1008    conda-forge
gfortran                  11.3.0               h1ca8e4b_0    conda-forge
gfortran_impl_osx-arm64   11.3.0              h95416cd_25    conda-forge
gfortran_osx-arm64        11.3.0               h57527a5_0    conda-forge
giflib                    5.2.1                h27ca646_2    conda-forge
glib                      2.72.1               h332123e_0    conda-forge
glib-tools                2.72.1               h332123e_0    conda-forge
gmp                       6.2.1                h9f76cd9_0    conda-forge
gnutls                    3.7.7                h27a1928_0    conda-forge
graphite2                 1.3.13            h9f76cd9_1001    conda-forge
graphviz                  5.0.1                h37ddf74_0    conda-forge
gst-plugins-base          1.20.3               h8b7775e_1    conda-forge
gstreamer                 1.20.3               hcb7b3dd_1    conda-forge
gtk2                      2.24.33              h57013de_2    conda-forge
gts                       0.7.6                h4b6d4d6_2    conda-forge
harfbuzz                  5.1.0                hd36a07e_0    conda-forge
[... 140 lines cut ...]
x264                      1!164.3095           h57fd34a_2    conda-forge
x265                      3.5                  hbc6ce65_3    conda-forge
xerces-c                  3.2.3                h1b0b642_5    conda-forge
xorg-kbproto              1.0.7             h27ca646_1002    conda-forge
xorg-libice               1.0.10               h27ca646_0    conda-forge
xorg-libsm                1.2.3             h27ca646_1000    conda-forge
xorg-libx11               1.7.2                h3422bc3_0    conda-forge
[... cut ...]
xz                        5.2.6                h57fd34a_0    conda-forge
yaml                      0.2.5                h3422bc3_2    conda-forge
zeromq                    4.3.4                hbdafb3b_1    conda-forge
zlib                      1.2.12               ha287fd2_2    conda-forge
zstd                      1.5.2                h8128057_4    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
~~~

So probably the issue is not related to 11.9 vs 11.11, but rather to something else (compilation, dependencies).

traversaro, 2022-09-08:
> I could manually make these variables point to `/Users/nunoguedelha/miniforge3/envs/robotologybin3/include` and `/Users/nunoguedelha/miniforge3/envs/robotologybin3/lib/libprotobufxxx.dylib` but I guess it's better to do it in the `CMakeLists.txt`...

As long as they are correct, probably that is ok, even if it is a good idea to avoiding polluting the `CMakeLists.txt` .

traversaro, 2022-09-08:
> Maybe installing the google protobuf (libprotobuf-dev) would help?

All the development libraries should be provided by `libprotobuf` , I do not think there is a `libprotobuf-dev` package in conda-forge .

nunoguedelha, 2022-09-08:
I added to the CMake command line `DCMAKE_PREFIX_PATH=$CONDA_PREFIX` to force CMake to find protobuf in conda environment. With this, the `ninja` compilation step seems to be working.

traversaro, 2022-09-16:
I wondered if the problem was related to the clang version, but I tried with both clang 12 and clang 13 and the result was the same.

traversaro, 2022-09-16:
I am debugging more in depth the issue, and apparently on macOS `gzserver` is running normally (the physics is effectively stepped), but for some reason not all topics are opened:
~~~
(gazebo) icub@iCubs-Mac-mini ~ % gz topic -l
/gazebo/default/atmosphere
/gazebo/default/factory
/gazebo/default/factory/light
/gazebo/default/joint
/gazebo/default/light/modify
/gazebo/default/log/control
/gazebo/default/model/modify
/gazebo/default/physics
/gazebo/default/playback_control
/gazebo/default/pose/modify
/gazebo/default/request
/gazebo/default/undo_redo
/gazebo/default/user_cmd
/gazebo/default/wind
/gazebo/default/world_control
/gazebo/server/control
~~~
vs a normal empty gazebo instance:
~~~
~~~

traversaro, 2022-09-16:
I debugged quite in depth, and the issue seems that `ConnectionManager::Instance` is not behaving as Singleton, but instead it is returning different instances over time. I am not sure what is the reason behind that, but may be it is just a strange side effect of some ABI incompatibility in the dependency chain?

traversaro, 2022-09-16:
I added the following code to further debug:
~~~diff
diff --git a/gazebo/common/SingletonT.hh b/gazebo/common/SingletonT.hh
index e97b7c792e..3861046895 100644
--- a/gazebo/common/SingletonT.hh
+++ b/gazebo/common/SingletonT.hh
@@ -24,6 +24,10 @@

 #include "gazebo/util/system.hh"

+#include <iostream>
+#include <thread>
+#include <typeinfo>
+
 /// \addtogroup gazebo_common Common
 /// \{

@@ -48,8 +52,14 @@ class SingletonT
   private: static T &GetInstance()
            {
              static T t;
+             std::cerr << "\nSingletonT::GetInstance with type " <<  typeid(t).name() << " and instance " << &t << " thread_id " << std::this_thread::get_id() << std::endl;
              return static_cast<T &>(t);
            }
+
+  public: SingletonT(SingletonT const&) = delete;
+  public: void operator=(SingletonT const&)  = delete;
+  public: SingletonT(SingletonT &&) = delete;
+  public: void operator=(SingletonT &&)  = delete;
 };

 /// \brief Helper to declare typed SingletonT
~~~

and this is the output: https://gist.github.com/traversaro/9783ef649048eab0a22ce82efed94978 .

It is possible to see how in sometimes the singleton type is the same, but the returned instance is different, for example:
~~~
SingletonT::GetInstance with type N6gazebo9transport17ConnectionManagerE and instance 0x1019888a8 thread_id 0x100ca8580
SingletonT::GetInstance with type N6gazebo9transport17ConnectionManagerE and instance 0x10284ef90 thread_id 0x100ca8580
SingletonT::GetInstance with type N6gazebo9transport17ConnectionManagerE and instance 0x100ba4db8 thread_id 0x100ca8580
SingletonT::GetInstance with type N6gazebo9transport17ConnectionManagerE and instance 0x100e78f50 thread_id 0x100ca8580
SingletonT::GetInstance with type N6gazebo9transport17ConnectionManagerE and instance 0x101871248 thread_id 0x100ca8580
~~~

However, after the initial confusion, the instance return always the first (correctly initialized) instance:
~~~
SingletonT::GetInstance with type N6gazebo9transport17ConnectionManagerE and instance 0x1019888a8 thread_id 0x17202f000
~~~

I have still no idea of what is happening. My best hypothesis is of a magical ABI problem in the dependency that would explain that, but to be honest at the moment I do not know how to debug this further.

traversaro, 2022-09-21:
I also tried to re-compiled all Gazebo and ignition dependencies, both in Release and in Debug, and the problem is still there.

traversaro, 2022-09-22:
> I wondered if the problem was related to the clang version, but I tried with both clang 12 and clang 13 and the result was the same.

This was quite unlucky. After a week of trying different combination of the dependencies, I reached a point in which I reproduced almost exactly the environment that built the latest working binary (gazebo 11.9.1), and even in that case it was not working. At that point, I checked the `info/hash_input.json` file in the package, and I noticed that the compiler used in that case was clang 11 . After trying with clang 11 in several combination, gazebo was always working. So it seems there is some kind of problem in clang 12/13/13, or some problem in Gazebo that becames apparent only with clang 12/13/14 .

Tobias-Fischer, 2022-09-22:
Oh no, that is unlucky! And not very satisfying either as those compiler differences are near impossible to debug. Thanks a lot for your efforts!

closed 2022-10-13 by github-actions[bot]