conda-forge/gazebo-feedstock #159: cgraph.dll and gvc.dll missing on Windows

johnwason, 2022-12-21:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I am getting errors for cgraph.dll and gvc.dll missing on Windows

### Installed packages

```shell
# Name                    Version                   Build  Channel
aom                       3.5.0                h63175ca_0    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
blosc                     1.21.3               hdccc3a2_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2022.12.7            h5b45459_0    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
cfitsio                   4.2.0                h9ebe7e4_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppzmq                    4.9.0                h449d27f_0    conda-forge
curl                      7.86.0               h68f0423_2    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.5.0                h1537add_0    conda-forge
ffmpeg                    5.1.2           gpl_h5b1d025_105    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.1               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0              h064cf08_10    conda-forge
freetype                  2.12.1               h546665d_1    conda-forge
freexl                    1.0.6                h67ca5e6_1    conda-forge
gazebo                    11.12.0              ha5ac59a_4    conda-forge
geos                      3.11.1               h1537add_0    conda-forge
geotiff                   1.7.1                h4ffd875_4    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glib                      2.74.1               h12be248_1    conda-forge
glib-tools                2.74.1               h12be248_1    conda-forge
gst-plugins-base          1.21.2               h001b923_0    conda-forge
gstreamer                 1.21.2               h6b5321d_0    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
hdf4                      4.2.15               h1b1b6ef_5    conda-forge
hdf5                      1.12.2          nompi_h2a0e4a3_100    conda-forge
icu                       70.1                 h0e60522_0    conda-forge
imath                     3.1.6                h12be248_1    conda-forge
intel-openmp              2023.0.0         h57928b3_25922    conda-forge
jpeg                      9e                   h8ffe710_2    conda-forge
jsoncpp                   1.9.5                h2d74725_1    conda-forge
jxrlib                    1.1                  h8ffe710_2    conda-forge
kealib                    1.5.0                h61be68b_0    conda-forge
krb5                      1.20.1               h6609f42_0    conda-forge
lcms2                     2.14                 h90d422f_0    conda-forge
lerc                      4.0.0                h63175ca_0    conda-forge
libblas                   3.9.0              16_win64_mkl    conda-forge
libcblas                  3.9.0              16_win64_mkl    conda-forge
libccd-double             2.1                  h0e60522_3    conda-forge
libclang                  15.0.6          default_h77d9078_0    conda-forge
libclang13                15.0.6          default_h77d9078_0    conda-forge
libcurl                   7.86.0               h68f0423_2    conda-forge
libdeflate                1.14                 hcfcfb64_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libgdal                   3.6.1                h1cb26dd_1    conda-forge
libglib                   2.74.1               he8f3873_1    conda-forge
[... 100 lines cut ...]
                          __win=0=0
                          __archspec=1=x86_64
       base environment : C:\tools\miniconda3  (read only)
      conda av data dir : C:\tools\miniconda3\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\tools\miniconda3\pkgs
                          C:\Users\u\.conda\pkgs
                          C:\Users\u\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\u\.conda\envs
                          C:\tools\miniconda3\envs
                          C:\Users\u\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/22.9.0 requests/2.26.0 CPython/3.9.10 Windows/10 Windows/10.0.19044
          administrator : False
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2022-12-21:
Looks like graphviz is missing run exports :(

Tobias-Fischer, 2022-12-21:
Can you try whether the error goes away if you manually install graphviz?

johnwason, 2022-12-21:
Yep, it launched with graphviz

Tobias-Fischer, 2022-12-21:
Ok - are you happy to open an issue in the graphviz feedstock?

traversaro, 2022-12-21:
This is indeed a regression on our side, see https://github.com/conda-forge/gazebo-feedstock/pull/154 .

traversaro, 2022-12-21:
I opened https://github.com/conda-forge/graphviz-feedstock/pull/111 . If it is merged quickly, I will then trigger a re-build of gazebo to take into account this new constraint, otherwise if it takes some time I will manually added the constraint a re-build the package.

closed 2022-12-23 by traversaro