conda-forge/gazebo-feedstock #161: "Cannot Add Model to World because simulation is not running" strikes back

traversaro, 2023-02-15:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

According to user reports, https://github.com/conda-forge/gazebo-feedstock/issues/148 is happening again, see https://github.com/RoboStack/ros-noetic/issues/323#issuecomment-1431914379 . I checked and I think I understood what happen: in https://github.com/conda-forge/gazebo-feedstock/pull/154 we accidentally started to apply the patch that fixed  https://github.com/conda-forge/gazebo-feedstock/issues/148  only on Win, so the builds: `11.12.0=*_3`, `11.12.0=*_4` and `11.12.0=*_5` are broken on macOS.

### Installed packages

```shell
.
```

### Environment info

```shell
.
```

traversaro, 2023-02-15:
Hopefully https://github.com/conda-forge/gazebo-feedstock/pull/162 should fix it for 11.12.0=*_6 builds. If necessary, we may need to mark 11.12.0=*_3, 11.12.0=*_4 and 11.12.0=*_5 builds as broken on macOS.

traversaro, 2023-02-15:
> Hopefully #162 should fix it for `11.12.0=*_6` builds. If necessary, we may need to mark `11.12.0=*_3`, `11.12.0=*_4` and `11.12.0=*_5` builds as broken on macOS.

In the end I propose to mark the build as broken as it is possible that a broke build was going to be selected in the future due to the migration that occured in the meanwhile: https://github.com/conda-forge/admin-requests/pull/677 .

traversaro, 2023-02-19:
> In the end I propose to mark the build as broken as it is possible that a broke build was going to be selected in the future due to the migration that occured in the meanwhile: [conda-forge/admin-requests#677](https://github.com/conda-forge/admin-requests/pull/677) .

Until this PR is fixed, a workaround if ones uses gazebo with ros-noetic is to manually install:
~~~
mamba install gazebo=11.12.0=*_2
~~~
see https://github.com/RoboStack/ros-noetic/issues/323#issuecomment-1432861574 .

Tobias-Fischer, 2023-02-19:
PR was merged :).

Thanks again @traversaro!

closed 2023-02-19 by Tobias-Fischer