conda-forge/gazebo-feedstock #183: Installing py-opencv and gazebo on Linux causes gazebo failure

johnwason, 2023-08-28:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Installing `gazebo` and `py-opencv` from `conda-forge` causes the following error on `gazebo --verbose`:

```
Err] [RenderEngine.cc:460] Unable to load Ogre Plugin[/home/user/mambaforge/envs/rr_training_sim_dev3/lib/OGRE/RenderSystem_GL]. Rendering will not be possible. Make sure you have installed OGRE and Gazebo properly.

[Err] [RenderEngine.cc:685] EXCEPTION: unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.

[Err] [RenderingIface.cc:41] Failed to load the Rendering engine subsystem

unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.

[Err] [gazebo.cc:80] Unable to load sensors

[Err] [Server.cc:330] Unable to load gazebo
```

I can confirm that the `RenderSystem_GL.so` file is missing. This is odd, since simply installing gazebo does not cause a problem. If I install `py-opencv` after `gazebo`, I get this upgrade message:

```
Looking for: ['py-opencv']

conda-forge/linux-64                                        Using cache

conda-forge/noarch                                          Using cache

pkgs/main/noarch                                              No change

pkgs/r/linux-64                                               No change

pkgs/main/linux-64                                            No change

pkgs/r/noarch                                                 No change

Pinned packages:

  - python 3.10.*

Transaction

  Prefix: /home/wasonj/mambaforge/envs/rr_training_sim_dev3

  Updating specs:

   - py-opencv

   - ca-certificates

   - openssl

  Package                                  Version  Build                Channel                    Size

──────────────────────────────────────────────────────────────────────────────────────────────────────────

  Install:

──────────────────────────────────────────────────────────────────────────────────────────────────────────

  + jasper                                   4.0.0  h32699f2_1           conda-forge/linux-64     Cached

  + liblapacke                               3.9.0  17_linux64_openblas  conda-forge/linux-64     Cached

  + libopencv                                4.8.0  py310hd58e3cd_0      conda-forge/linux-64     Cached

  + libopenvino                           2023.0.1  h59595ed_2           conda-forge/linux-64     Cached
[... 677 lines cut ...]

                          https://repo.anaconda.com/pkgs/main/noarch

                          https://repo.anaconda.com/pkgs/r/linux-64

                          https://repo.anaconda.com/pkgs/r/noarch

          package cache : /home/user/mambaforge/pkgs

                          /home/user/.conda/pkgs

       envs directories : /home/user/mambaforge/envs

                          /home/user/.conda/envs

               platform : linux-64

             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Linux/6.2.0-26-generic ubuntu/22.04.2 glibc/2.35

                UID:GID : 1000:1000

             netrc file : None

           offline mode : False
```

Tobias-Fischer, 2023-08-28:
Can you try `mamba install 'ogre=1.10.12=*_10'`? It downgrades ogre for some reason, and that old build is broken.

johnwason, 2023-08-28:
It just downgrades py-opencv

```
  Package                                  Version  Build            Channel                    Size

──────────────────────────────────────────────────────────────────────────────────────────────────────

  Change:

──────────────────────────────────────────────────────────────────────────────────────────────────────

  - gazebo                                 11.13.0  hebecd67_4       conda-forge                    

  + gazebo                                 11.13.0  h84fa542_4       conda-forge/linux-64     Cached

  - libgoogle-cloud                         2.12.0  h840a212_1       conda-forge                    

  + libgoogle-cloud                         2.12.0  hac9eb74_1       conda-forge/linux-64     Cached

  - libignition-fuel-tools4                  4.6.0  hf5721fa_2       conda-forge                    

  + libignition-fuel-tools4                  4.6.0  h2fc7885_2       conda-forge/linux-64     Cached

  - libignition-msgs5                       5.11.0  h3189f74_1       conda-forge                    

  + libignition-msgs5                       5.11.0  haf0fa91_1       conda-forge/linux-64     Cached

  - libignition-transport8                   8.4.0  hb139ee8_3       conda-forge                    

  + libignition-transport8                   8.4.0  h87fd66a_3       conda-forge/linux-64     Cached

  - ogre                                   1.10.12  hc56bac0_2       conda-forge                    

  + ogre                                   1.10.12  h7cc4a1d_10      conda-forge/linux-64     Cached

  Downgrade:

──────────────────────────────────────────────────────────────────────────────────────────────────────

  - libgrpc                                 1.56.2  h3905398_1       conda-forge                    

  + libgrpc                                 1.54.3  hb20ce57_0       conda-forge/linux-64     Cached

  - libopencv                                4.8.0  py310hd58e3cd_0  conda-forge                    

  + libopencv                                4.7.0  py310h3e876cf_6  conda-forge/linux-64     Cached

  - libopenvino                           2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino                           2023.0.0  h59595ed_0       conda-forge/linux-64        6MB

  - libopenvino-auto-batch-plugin         2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-auto-batch-plugin         2023.0.0  h59595ed_0       conda-forge/linux-64      120kB

  - libopenvino-auto-plugin               2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-auto-plugin               2023.0.0  h59595ed_0       conda-forge/linux-64      245kB

  - libopenvino-hetero-plugin             2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-hetero-plugin             2023.0.0  h59595ed_0       conda-forge/linux-64      229kB

  - libopenvino-intel-cpu-plugin          2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-intel-cpu-plugin          2023.0.0  h59595ed_0       conda-forge/linux-64       10MB

  - libopenvino-intel-gpu-plugin          2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-intel-gpu-plugin          2023.0.0  h59595ed_0       conda-forge/linux-64        6MB

  - libopenvino-ir-frontend               2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-ir-frontend               2023.0.0  h59595ed_0       conda-forge/linux-64      248kB

  - libopenvino-onnx-frontend             2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-onnx-frontend             2023.0.0  h59595ed_0       conda-forge/linux-64        1MB

  - libopenvino-paddle-frontend           2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-paddle-frontend           2023.0.0  h59595ed_0       conda-forge/linux-64      637kB

  - libopenvino-pytorch-frontend          2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-pytorch-frontend          2023.0.0  h59595ed_0       conda-forge/linux-64      602kB

  - libopenvino-tensorflow-frontend       2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-tensorflow-frontend       2023.0.0  h59595ed_0       conda-forge/linux-64        1MB

  - libopenvino-tensorflow-lite-frontend  2023.0.1  h59595ed_2       conda-forge                    

  + libopenvino-tensorflow-lite-frontend  2023.0.0  h59595ed_0       conda-forge/linux-64      425kB

  - libprotobuf                             4.23.3  hd1fb520_0       conda-forge                    

  + libprotobuf                            3.21.12  hfc55251_1       conda-forge/linux-64     Cached

  - pugixml                                   1.13  h59595ed_1       conda-forge                    

  + pugixml                                 1.11.4  h59595ed_1       conda-forge/linux-64     Cached

  - py-opencv                                4.8.0  py310hfdc917e_0  conda-forge                    

  + py-opencv                                4.7.0  py310hfdc917e_6  conda-forge/linux-64     Cached

  Summary:

  Change: 6 packages

  Downgrade: 17 packages

  Total download: 27MB

```

Tobias-Fischer, 2023-08-28:
For some reason @conda-forge/ogre has not been migrated to pugixml 1.13, which causes this issue. Not sure why. Any clue @traversaro?

Tobias-Fischer, 2023-08-28:
Ah, I know why. There is no migrator for it. Seems like we just need to rebuild ogre 1.10 and it should pick it up.

Tobias-Fischer, 2023-08-28:
@johnwason - are you happy to open a PR in the ogre-feedstock, simply bumping the build number, for the 1.10 branch?

johnwason, 2023-08-28:
> are you happy to open a PR in the ogre-feedstock, simply bumping the build number, for the 1.10 branch?

Done

traversaro, 2023-08-28:
Related PR: https://github.com/conda-forge/conda-forge-pinning-feedstock/pull/4824 . As that PR is not merged, rebuilding ogre is for sure the fastest solution.

Tobias-Fischer, 2023-08-28:
Should be ok now? Can you please try?

traversaro, 2023-08-28:
> Should be ok now? Can you please try?

I just restarted a failing CI (https://github.com/robotology/gazebo-yarp-plugins/actions/runs/5994777125) and it is indeed working fine, I think we can close the issue, @johnwason if you still have problems feel free to comment.

johnwason, 2023-08-28:
Gazebo does start, but installing `py-opencv` after `gazebo` is causing downgrades of packages:

```
Looking for: ['py-opencv']

conda-forge/linux-64                                        Using cache

conda-forge/noarch                                          Using cache

pkgs/r/noarch                                                 No change

pkgs/main/noarch                                              No change

pkgs/r/linux-64                                               No change

pkgs/main/linux-64                                            No change

Pinned packages:

  - python 3.11.*

Transaction

  Prefix: /home/wasonj/mambaforge/envs/gazebo_test

  Updating specs:

   - py-opencv

   - ca-certificates

   - openssl

  Package                                  Version  Build                Channel                    Size

──────────────────────────────────────────────────────────────────────────────────────────────────────────

  Install:

──────────────────────────────────────────────────────────────────────────────────────────────────────────

  + jasper                                   4.0.0  h32699f2_1           conda-forge/linux-64     Cached

  + liblapacke                               3.9.0  17_linux64_openblas  conda-forge/linux-64     Cached

  + libopencv                                4.8.0  py311h8aafb54_0      conda-forge/linux-64       30MB

  + libopenvino                           2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-auto-batch-plugin         2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-auto-plugin               2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-hetero-plugin             2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-intel-cpu-plugin          2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-intel-gpu-plugin          2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-ir-frontend               2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-onnx-frontend             2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-paddle-frontend           2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-pytorch-frontend          2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-tensorflow-frontend       2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + libopenvino-tensorflow-lite-frontend  2023.0.1  h59595ed_2           conda-forge/linux-64     Cached

  + ocl-icd                                  2.3.1  h7f98852_0           conda-forge/linux-64     Cached

  + ocl-icd-system                           1.0.0  1                    conda-forge/linux-64     Cached

  + py-opencv                                4.8.0  py311h781c19f_0      conda-forge/linux-64        1MB

  Change:

──────────────────────────────────────────────────────────────────────────────────────────────────────────

  - gazebo                                 11.13.0  hebecd67_4           conda-forge                    

  + gazebo                                 11.13.0  h84fa542_4           conda-forge/linux-64     Cached

  - libgoogle-cloud                         2.12.0  h840a212_1           conda-forge                    

  + libgoogle-cloud                         2.12.0  hac9eb74_1           conda-forge/linux-64     Cached

  - libignition-fuel-tools4                  4.6.0  hf5721fa_2           conda-forge                    

  + libignition-fuel-tools4                  4.6.0  h2fc7885_2           conda-forge/linux-64     Cached

  - libignition-msgs5                       5.11.0  h3189f74_1           conda-forge                    

  + libignition-msgs5                       5.11.0  haf0fa91_1           conda-forge/linux-64     Cached

  - libignition-transport8                   8.4.0  hb139ee8_3           conda-forge                    

  + libignition-transport8                   8.4.0  h87fd66a_3           conda-forge/linux-64     Cached

  Downgrade:

──────────────────────────────────────────────────────────────────────────────────────────────────────────

  - libgrpc                                 1.56.2  h3905398_1           conda-forge                    

  + libgrpc                                 1.54.3  hb20ce57_0           conda-forge/linux-64     Cached

  - libprotobuf                             4.23.3  hd1fb520_0           conda-forge                    

  + libprotobuf                            3.21.12  hfc55251_1           conda-forge/linux-64     Cached

  Summary:

  Install: 18 packages

  Change: 5 packages

  Downgrade: 2 packages

  Total download: 32MB

```

Installing py-opencv should be unrelated to these packages. My other packages are failing because of the weird state the workspace is in after this install.

johnwason, 2023-08-28:
I am seeing similar package downgrade behavior on windows when installing py-opencv after gazebo.

traversaro, 2023-08-29:
I reproduced the problem with the following command:
~~~
mamba create -n test183together gazebo=11.13.0 py-opencv=4.8.0 python=3.11 libprotobuf=4.23.3 hdf5=1.14.2
~~~

that gives as error:
~~~
Looking for: ['gazebo=11.13.0', 'py-opencv=4.8.0', 'python=3.11', 'libprotobuf=4.23.3']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
Could not solve for environment specs
The following packages are incompatible
├─ gazebo 11.13.0**  is installable with the potential options
│  ├─ gazebo 11.13.0 would require
│  │  └─ hdf5 >=1.14.1,<1.14.2.0a0 , which can be installed;
│  ├─ gazebo 11.13.0 would require
│  │  └─ libprotobuf >=4.23.2,<4.23.3.0a0 , which can be installed;
│  └─ gazebo 11.13.0 would require
│     └─ libprotobuf >=3.21.12,<3.22.0a0 , which can be installed;
├─ libprotobuf 4.23.3**  is uninstallable because it conflicts with any installable versions previously reported;
├─ py-opencv 4.8.0**  is installable with the potential options
│  ├─ py-opencv 4.8.0 would require
│  │  └─ python_abi 3.8 *_pypy38_pp73, which can be installed;
│  ├─ py-opencv 4.8.0 would require
│  │  └─ python_abi 3.10.* *_cp310, which can be installed;
│  ├─ py-opencv 4.8.0 would require
│  │  └─ libopencv [4.8.0 py311h8aafb54_0|4.8.0 py39h76ed197_0], which requires
│  │     └─ libprotobuf >=3.21.12,<3.22.0a0 , which can be installed;
│  ├─ py-opencv 4.8.0 would require
│  │  └─ libopencv 4.8.0 py311h3563e6c_1, which requires
│  │     └─ hdf5 >=1.14.2,<1.14.3.0a0 , which conflicts with any installable versions previously reported;
│  ├─ py-opencv 4.8.0 would require
│  │  └─ python_abi 3.8.* *_cp38, which can be installed;
│  ├─ py-opencv 4.8.0 would require
│  │  └─ python_abi 3.9 *_pypy39_pp73, which can be installed;
│  └─ py-opencv 4.8.0 would require
│     └─ python_abi 3.9.* *_cp39, which can be installed;
└─ python 3.11**  is requested and can be installed.
~~~

that can be simplified further to:
~~~
traversaro@IITICUBLAP257:~$ mamba create -n test183together gazebo=11.13.0 hdf5=1.14.2

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.4.2) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['gazebo=11.13.0', 'hdf5=1.14.2']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
Could not solve for environment specs
The following packages are incompatible
├─ gazebo 11.13.0**  is installable with the potential options
│  ├─ gazebo 11.13.0 would require
│  │  └─ hdf5 >=1.14.0,<1.14.1.0a0 , which can be installed;
│  └─ gazebo 11.13.0 would require
│     └─ hdf5 >=1.14.1,<1.14.2.0a0 , which can be installed;
└─ hdf5 1.14.2**  is uninstallable because it conflicts with any installable versions previously reported.
~~~

traversaro, 2023-08-29:
The problem boils down to https://github.com/conda-forge/flann-feedstock/pull/37 not being merged, that is blocking the hdf5 1.14.2 migration in Gazebo. Anyhow, this should not be a big problem as the `libprotobuf=3.21.12` builds should be working fine, what problems are you seeing with the @johnwason ?

johnwason, 2023-08-29:
The problem I am having is with a plugin I am building for gazebo. When the libprotobuf is downgraded, the plugin is complaining it can't find the libprotobuf version it was built against.

traversaro, 2023-08-29:
Is the plugin in a conda package? If not, this is expected, there is nothing enforcing that the ABI of the packages remain compatible after a `mamba install`. If yes, probably the issue is connected to https://github.com/conda-forge/gazebo-feedstock/issues/107 . My current workaround in gazebo-yarp-plugins is to manually list the public dependency of gazebo as host dependencies, see https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/blob/438c8b6442204188301d1fc44806ac88d75e9424/recipe/meta.yaml#L40 .

johnwason, 2023-08-29:
The plugin is currently in the "robotraconteur" organization until I can get everything merged in. I am currently waiting on https://github.com/conda-forge/staged-recipes/pull/23781 . The plugin builds and works fine, until I install py-opencv in the same environment. At that point it stops working due to the libprotobuf version error.

johnwason, 2023-08-29:
I was able to make everything work by creating an environment with gazebo 11.12 and installing my packages into the environment directly. I will distribute the binaries using conda-pack for now. I will see if the problem is resolved when all the packages are in conda-forge.

closed 2023-08-29 by johnwason