conda-forge/gazebo-feedstock #184: Gazebo client hangs on open on Windows

johnwason, 2023-08-29:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

gazebo client will hang on open intermittently on Windows with the most recent 11.13 builds. It will start up normally for a few times, and then will randomly refuse to start several times, until randomly starting normally again. I tried installing version 11.12 and it seems to work normally. I tested in a clean environment with no other packages installed beyond gazebo and its dependencies.

It looks similar to this problem: https://github.com/ms-iot/ROSOnWindows/issues/47

This problem is intermittent so it is very difficult to determine what conditions cause it.

### Installed packages

```shell
aom                       3.5.0                h63175ca_0    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
blosc                     1.21.4               hdccc3a2_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_4    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-ares                    1.19.1               hcfcfb64_0    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cairo                     1.16.0            hdecc03f_1016    conda-forge
cfitsio                   4.3.0                h9b0cee5_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppzmq                    4.10.0               h449d27f_0    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
ffmpeg                    6.0.0           gpl_h1f67d4f_104    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0              hcff1195_15    conda-forge
freetype                  2.12.1               h546665d_1    conda-forge
freexl                    1.0.6                h67ca5e6_1    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gazebo                    11.13.0              h1fde3ec_4    conda-forge
geos                      3.12.0               h1537add_0    conda-forge
geotiff                   1.7.1               h4e61e90_11    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glib                      2.76.4               h12be248_0    conda-forge
glib-tools                2.76.4               h12be248_0    conda-forge
graphite2                 1.3.13                     1000    conda-forge
graphviz                  8.1.0                h51cb2cd_0    conda-forge
gst-plugins-base          1.22.5               h001b923_0    conda-forge
gstreamer                 1.22.5               hb4038d2_0    conda-forge
gts                       0.7.6                h6b5321d_4    conda-forge
harfbuzz                  7.3.0                h196d34a_0    conda-forge
hdf4                      4.2.15               h1334946_6    conda-forge
hdf5                      1.14.1          nompi_h73e8ff5_100    conda-forge
icu                       72.1                 h63175ca_0    conda-forge
imath                     3.1.8                h12be248_0    conda-forge
intel-openmp              2023.2.0         h57928b3_49496    conda-forge
jsoncpp                   1.9.5                h2d74725_1    conda-forge
jxrlib                    1.1                  h8ffe710_2    conda-forge
kealib                    1.5.1                h96def13_4    conda-forge
krb5                      1.21.2               heb0366b_0    conda-forge
lcms2                     2.15                 h3e3b177_1    conda-forge
lerc                      4.0.0                h63175ca_0    conda-forge
libabseil                 20230125.3      cxx17_h63175ca_0    conda-forge
libaec                    1.0.6                h63175ca_1    conda-forge
libarchive                3.6.2                h6f8411a_1    conda-forge
[... 142 lines cut ...]
       virtual packages : __archspec=1=x86_64
                          __cuda=12.2=0
                          __win=0=0
       base environment : C:\Users\user\mambaforge  (writable)
      conda av data dir : C:\Users\user\mambaforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : D:\ws\conda\pkgs
       envs directories : D:\ws\conda\envs
                          C:\Users\user\mambaforge\envs
                          C:\Users\user\.conda\envs
                          C:\Users\usor\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Windows/10 Windows/10.0.19045
          administrator : False
             netrc file : None
           offline mode : False
```

traversaro, 2023-08-30:
Can you describe your test environment? In a sense, are you starting `gzserver` and then trying to start `gzclient` several times, or for every test you are launching a new `gzserver` and `gzclient`?

johnwason, 2023-08-30:
I was starting one instance each of the client and server at a time, using either the `gazebo` command or `gzserver` followed by `gzclient`. I was trying to figure out if there was an issue with `gzclient` connecting too soon. I also tinkered with the environmental variables for `GAZEBO_IP` and `IGN_IP` along with other variables to see if it was a networking issue. The problem may not happen with an empty world. I have been using version 11.12 without seeing this problem. I was not able to reliably reproduce it either.

johnwason, 2023-09-20:
After a lot of trial and error, it appears the problem is that if the gzclient window spawns on a secondary monitor it will hang on startup. If it spawns on the main monitor, it seems to work. I have a three monitor setup so it will spawn on different screens depending on how Windows is feeling at that moment. Once the window has spawned on the main monitor it can be moved to a different monitor. The easiest solution is to disable extra monitors when using gazebo. I don't know if this is a problem with the Nvidia drivers for my RTX 3090 or if it is a problem with Ogre/Gazebo. If someone else has a multi-monitor setup it would be interesting to experiment.

johnwason, 2024-03-05:
I finally found the problem!

It has something to do with `NahimicOSD.dll`, which is a filter driver that is inserting itself into programs to provide some kind of useless on screen display. Apparently it comes with Alienware systems, along with a lot of other computers device drivers. It is loading an old version of DirectX which is causing the problem. DirectX is not being loaded by Gazebo.

This appears to be a similar problem: https://github.com/prusa3d/PrusaSlicer/issues/5573