conda-forge/gazebo-feedstock #204: Error starting on Mac OS

johnwason, 2024-03-05:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Gazebo does not start on my Mac Studio M1 Extreme when running in x86_64 mode. I get the following error:

```
(gazebo) wasonj@JohnsMacStudio ~ % gazebo
dyld[4183]: missing symbol called
zsh: abort      gazebo
```

Here is the full install and run:

```
(base) wasonj@JohnsMacStudio ~ % conda create -n gazebo gazebo
Channels:
 - conda-forge
Platform: osx-64
Collecting package metadata (repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
    current version: 23.11.0
    latest version: 24.1.2

Please update conda by running

    $ conda update -n base -c conda-forge conda

## Package Plan ##

  environment location: /Users/wasonj/miniforge3/envs/gazebo

  added / updated specs:
    - gazebo

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    bullet-cpp-3.25            |       h28a66ae_2        38.0 MB  conda-forge
    libignition-math6-6.15.1   |  py312hbebec0a_1         1.0 MB  conda-forge
    libode-0.16.2              |      heafc425_12         442 KB  conda-forge
    numpy-1.26.4               |  py312he3a82b2_0         6.7 MB  conda-forge
    python-3.12.2              |h9f0c242_0_cpython        13.9 MB  conda-forge
    python_abi-3.12            |          4_cp312           6 KB  conda-forge
    ------------------------------------------------------------
                                           Total:        60.0 MB

The following NEW packages will be INSTALLED:

  aom                conda-forge/osx-64::aom-3.8.1-h73e2aa4_0 
  assimp             conda-forge/osx-64::assimp-5.3.1-h460e769_2 
  atk-1.0            conda-forge/osx-64::atk-1.0-2.38.0-h1d18e73_1 
  aws-c-auth         conda-forge/osx-64::aws-c-auth-0.7.16-h79cb451_6 
  aws-c-cal          conda-forge/osx-64::aws-c-cal-0.6.10-h7beb4c2_1 
  aws-c-common       conda-forge/osx-64::aws-c-common-0.9.13-h10d778d_0 
  aws-c-compression  conda-forge/osx-64::aws-c-compression-0.2.18-hf5538d0_1 
  aws-c-event-stream conda-forge/osx-64::aws-c-event-stream-0.4.2-hb970d5a_4 
  aws-c-http         conda-forge/osx-64::aws-c-http-0.8.1-h45f72bc_5 
  aws-c-io           conda-forge/osx-64::aws-c-io-0.14.5-h59ded20_1 
  aws-c-mqtt         conda-forge/osx-64::aws-c-mqtt-0.10.2-ha7d5480_4 
  aws-c-s3           conda-forge/osx-64::aws-c-s3-0.5.2-hb71b441_0 
  aws-c-sdkutils     conda-forge/osx-64::aws-c-sdkutils-0.1.15-hf5538d0_1 
  aws-checksums      conda-forge/osx-64::aws-checksums-0.1.18-hf5538d0_1 
  aws-crt-cpp        conda-forge/osx-64::aws-crt-cpp-0.26.2-h56a5195_7 
  aws-sdk-cpp        conda-forge/osx-64::aws-sdk-cpp-1.11.267-h4da54b2_1 
[... 582 lines cut ...]
 populated config files : /Users/wasonj/miniforge3/.condarc
                          /Users/wasonj/.condarc
          conda version : 23.11.0
    conda-build version : not installed
         python version : 3.10.13.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=23.11.0=0
                          __osx=14.2.1=0
                          __unix=0=0
       base environment : /Users/wasonj/miniforge3  (writable)
      conda av data dir : /Users/wasonj/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/wasonj/miniforge3/pkgs
                          /Users/wasonj/.conda/pkgs
       envs directories : /Users/wasonj/miniforge3/envs
                          /Users/wasonj/.conda/envs
               platform : osx-64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.10.13 Darwin/23.2.0 OSX/14.2.1 solver/libmamba conda-libmamba-solver/23.12.0 libmambapy/1.5.5
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-03-05:
Hi @johnwason  - can you try with the osx-arm64 package instead please?

johnwason, 2024-03-05:
@Tobias-Fischer the osx-arm64 package does work. I don't know if the problem is Rosetta or if there is a problem with the x64 package. I do have an old 2012 Intel Mac Mini I can find if you don't have a test machine, but it runs an outdated version of Mac OS.

Tobias-Fischer, 2024-03-05:
Ok cool, that's good news. I've seen problems now a few times where trying to run osx-64 packages on osx-arm64 causes some library-related issues. If you could test with your old Intel Mac that'd be great.

johnwason, 2024-03-08:
@Tobias-Fischer gazebo opens normally on my 2012 x64 mac mini running mac osx 10.15.7.

Tobias-Fischer, 2024-03-09:
Thanks for checking! I’d said it’s not worth investigating this further, given that the packages work on their intended target platforms.