conda-forge/gazebo-feedstock #206: Windows build failure of 1.14.0=*_9 build

traversaro, 2024-03-16:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

~~~
2024-03-15T23:10:11.0496500Z TEST START: D:\bld\win-64\gazebo-11.14.0-h3902e0a_9.conda
2024-03-15T23:10:13.2949803Z Renaming work directory 'D:\bld\gazebo_1710535323199\work' to 'D:\bld\gazebo_1710535323199\work_moved_gazebo-11.14.0-h3902e0a_9_win-64'
2024-03-15T23:10:13.2952865Z shutil.move(work)=D:\bld\gazebo_1710535323199\work, dest=D:\bld\gazebo_1710535323199\work_moved_gazebo-11.14.0-h3902e0a_9_win-64)
2024-03-15T23:10:15.3302066Z Channels:
2024-03-15T23:10:15.3303827Z  - local
2024-03-15T23:10:15.3306629Z  - conda-forge
2024-03-15T23:10:15.3307188Z Platform: win-64
2024-03-15T23:10:40.0807919Z Collecting package metadata (repodata.json): ...working... done
2024-03-15T23:10:40.9708901Z Solving environment: ...working... failed
2024-03-15T23:10:40.9875986Z WARNING: failed to get install actions, retrying.  exception was: Unsatisfiable dependencies for platform win-64: {'nothing provides openssl >=3.2.1,<4.0a0 needed by libgdal-3.8.4-h7c2897a_0'}
2024-03-15T23:10:40.9903174Z WARNING: Tests failed for gazebo-11.14.0-h3902e0a_9.conda - moving package to D:\bld\broken
~~~

### Installed packages

```shell
.
```

### Environment info

```shell
.
```

Tobias-Fischer, 2024-03-16:
It looks like a restart of the build fixed the issue?

traversaro, 2024-04-06:
Yes.

closed 2024-04-06 by traversaro