conda-forge/gazebo-feedstock #223: gazebo (gzclient) with empty world crashes with "free(): invalid next size (fast)" when installed in the same environment of gazebo-yarp-plugins

traversaro, 2024-06-26:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Launching `gazebo --verbose` in an environment created with `conda create -n gazeboyarp gazebo yarp` works perfectly fine. Doing the same in an environment launched with `conda create -n gazebogyp gazebo yarp gazebo-yarp-plugins` results in a non-deterministic crash ~60% of the runs, with failures like `free(): invalid next size (fast)` . As the simulation is not loading anything related to gazebo-yarp-plugins, that sounds like an ABI break in some library on which gazebo depends, and is installed to an older version due to gazebo-yarp-plugins being installed.

### Installed packages

```shell
`gazebogyp` environment:
~~~
traversaro@IITBMP014LW012:~/gazebovanilla$ pixi list
Package                               Version       Build                  Size       Kind   Source
_libgcc_mutex                         0.1           conda_forge            2.5 KiB    conda  _libgcc_mutex-0.1-conda_forge.tar.bz2
_openmp_mutex                         4.5           2_gnu                  23.1 KiB   conda  _openmp_mutex-4.5-2_gnu.tar.bz2
ace                                   8.0.0         hac33072_1             5.2 MiB    conda  ace-8.0.0-hac33072_1.conda
alsa-lib                              1.2.12        h4ab18f5_0             542.8 KiB  conda  alsa-lib-1.2.12-h4ab18f5_0.conda
aom                                   3.9.1         hac33072_0             2.6 MiB    conda  aom-3.9.1-hac33072_0.conda
assimp                                5.4.1         h8343317_0             3.3 MiB    conda  assimp-5.4.1-h8343317_0.conda
atk-1.0                               2.38.0        h04ea711_2             347.6 KiB  conda  atk-1.0-2.38.0-h04ea711_2.conda
attr                                  2.5.1         h166bdaf_1             69.4 KiB   conda  attr-2.5.1-h166bdaf_1.tar.bz2
aws-c-auth                            0.7.22        hf36ad8f_6             103.5 KiB  conda  aws-c-auth-0.7.22-hf36ad8f_6.conda
aws-c-cal                             0.6.15        h816f305_1             45.7 KiB   conda  aws-c-cal-0.6.15-h816f305_1.conda
aws-c-common                          0.9.23        h4ab18f5_0             230.1 KiB  conda  aws-c-common-0.9.23-h4ab18f5_0.conda
aws-c-compression                     0.2.18        he027950_7             18.8 KiB   conda  aws-c-compression-0.2.18-he027950_7.conda
aws-c-event-stream                    0.4.2         hb72ac1a_14            52.8 KiB   conda  aws-c-event-stream-0.4.2-hb72ac1a_14.conda
aws-c-http                            0.8.2         h75ac8c9_3             190.8 KiB  conda  aws-c-http-0.8.2-h75ac8c9_3.conda
aws-c-io                              0.14.9        hd3d3696_3             154.8 KiB  conda  aws-c-io-0.14.9-hd3d3696_3.conda
aws-c-mqtt                            0.10.4        hb0abfc5_7             159.6 KiB  conda  aws-c-mqtt-0.10.4-hb0abfc5_7.conda
aws-c-s3                              0.5.10        h44b787d_4             107.4 KiB  conda  aws-c-s3-0.5.10-h44b787d_4.conda
aws-c-sdkutils                        0.1.16        he027950_3             53.7 KiB   conda  aws-c-sdkutils-0.1.16-he027950_3.conda
aws-checksums                         0.1.18        he027950_7             49 KiB     conda  aws-checksums-0.1.18-he027950_7.conda
aws-crt-cpp                           0.26.12       he940a02_1             332.2 KiB  conda  aws-crt-cpp-0.26.12-he940a02_1.conda
aws-sdk-cpp                           1.11.329      h0f5bab0_6             3.4 MiB    conda  aws-sdk-cpp-1.11.329-h0f5bab0_6.conda
azure-core-cpp                        1.12.0        h830ed8b_0             335.3 KiB  conda  azure-core-cpp-1.12.0-h830ed8b_0.conda
azure-identity-cpp                    1.8.0         hdb0d106_1             193.1 KiB  conda  azure-identity-cpp-1.8.0-hdb0d106_1.conda
azure-storage-blobs-cpp               12.11.0       ha67cba7_1             509.8 KiB  conda  azure-storage-blobs-cpp-12.11.0-ha67cba7_1.conda
azure-storage-common-cpp              12.6.0        he3f277c_1             133.5 KiB  conda  azure-storage-common-cpp-12.6.0-he3f277c_1.conda
blosc                                 1.21.6        hef167b5_0             47.7 KiB   conda  blosc-1.21.6-hef167b5_0.conda
bullet-cpp                            3.25          hfb8ada1_2             40.4 MiB   conda  bullet-cpp-3.25-hfb8ada1_2.conda
bzip2                                 1.0.8         hd590300_5             248.3 KiB  conda  bzip2-1.0.8-hd590300_5.conda
c-ares                                1.28.1        hd590300_0             164.9 KiB  conda  c-ares-1.28.1-hd590300_0.conda
ca-certificates                       2024.6.2      hbcca054_0             152.4 KiB  conda  ca-certificates-2024.6.2-hbcca054_0.conda
cairo                                 1.18.0        h3faef2a_0             959.3 KiB  conda  cairo-1.18.0-h3faef2a_0.conda
cfitsio                               4.4.1         hf8ad068_0             902.5 KiB  conda  cfitsio-4.4.1-hf8ad068_0.conda
console_bridge                        1.0.2         h924138e_1             18 KiB     conda  console_bridge-1.0.2-h924138e_1.tar.bz2
cppzmq                                4.10.0        h2e2a08d_1             28.4 KiB   conda  cppzmq-4.10.0-h2e2a08d_1.conda
dartsim                               6.13.2        hdf3b901_3             13.7 MiB   conda  dartsim-6.13.2-hdf3b901_3.conda
dav1d                                 1.2.1         hd590300_0             742.4 KiB  conda  dav1d-1.2.1-hd590300_0.conda
dbus                                  1.13.6        h5008d03_3             604.1 KiB  conda  dbus-1.13.6-h5008d03_3.tar.bz2
double-conversion                     3.3.0         h59595ed_0             76.8 KiB   conda  double-conversion-3.3.0-h59595ed_0.conda
eigen                                 3.4.0         h00ab1b0_0             1 MiB      conda  eigen-3.4.0-h00ab1b0_0.conda
ergocub-software                      0.7.4         ha770c72_0             7.9 KiB    conda  ergocub-software-0.7.4-ha770c72_0.conda
expat                                 2.6.2         h59595ed_0             134.4 KiB  conda  expat-2.6.2-h59595ed_0.conda
fcl                                   0.7.0         hadc09e8_4             1.5 MiB    conda  fcl-0.7.0-hadc09e8_4.conda
ffmpeg                                6.1.1         gpl_he44c6f3_112       9.3 MiB    conda  ffmpeg-6.1.1-gpl_he44c6f3_112.conda
flann                                 1.9.2         h2b5ea80_0             1.5 MiB    conda  flann-1.9.2-h2b5ea80_0.conda
fmt                                   10.2.1        h00ab1b0_0             189.3 KiB  conda  fmt-10.2.1-h00ab1b0_0.conda
font-ttf-dejavu-sans-mono             2.37          hab24e00_0             388.1 KiB  conda  font-ttf-dejavu-sans-mono-2.37-hab24e00_0.tar.bz2
font-ttf-inconsolata                  3.000         h77eed37_0             94.3 KiB   conda  font-ttf-inconsolata-3.000-h77eed37_0.tar.bz2
font-ttf-source-code-pro              2.038         h77eed37_0             684.4 KiB  conda  font-ttf-source-code-pro-2.038-h77eed37_0.tar.bz2
font-ttf-ubuntu                       0.83          h77eed37_2             1.5 MiB    conda  font-ttf-ubuntu-0.83-h77eed37_2.conda
fontconfig                            2.14.2        h14ed4e7_0             265.6 KiB  conda  fontconfig-2.14.2-h14ed4e7_0.conda
fonts-conda-ecosystem                 1             0                      3.6 KiB    conda  fonts-conda-ecosystem-1-0.tar.bz2
fonts-conda-forge                     1             0                      4 KiB      conda  fonts-conda-forge-1-0.tar.bz2
freeglut                              3.2.2         hac7e632_2             139.6 KiB  conda  freeglut-3.2.2-hac7e632_2.conda
freeimage                             3.18.0        h4b96d29_20            450.6 KiB  conda  freeimage-3.18.0-h4b96d29_20.conda
freetype                              2.12.1        h267a509_2             620.1 KiB  conda  freetype-2.12.1-h267a509_2.conda
[... 269 lines cut ...]
                  : __linux=5.15.153.1=0
                  : __glibc=2.35=0
                  : __cuda=12.3=0
                  : __archspec=1=skylake
         Cache dir: /home/traversaro/.cache/rattler/cache
      Auth storage: /home/traversaro/.rattler/credentials.json

Project
------------
              Name: gazebovanilla
           Version: 0.1.0
     Manifest file: /home/traversaro/gazebovanilla/pixi.toml
  Config locations:
      Last updated: 27-06-2024 00:17:22

Environments
------------
       Environment: default
          Features: default
          Channels: conda-forge
  Dependency count: 4
      Dependencies: gazebo, ergocub-software, yarp, gazebo-yarp-plugins
  Target platforms: linux-64
~~~
```

traversaro, 2024-06-26:
Environment that instead works fine:

`gazeboyarp` (working) environment:
~~~
traversaro@IITBMP014LW012:~/gazebovanilla$ cat working_list
Package                               Version       Build                  Size       Kind   Source
_libgcc_mutex                         0.1           conda_forge            2.5 KiB    conda  _libgcc_mutex-0.1-conda_forge.tar.bz2
_openmp_mutex                         4.5           2_gnu                  23.1 KiB   conda  _openmp_mutex-4.5-2_gnu.tar.bz2
ace                                   8.0.0         hac33072_1             5.2 MiB    conda  ace-8.0.0-hac33072_1.conda
alsa-lib                              1.2.12        h4ab18f5_0             542.8 KiB  conda  alsa-lib-1.2.12-h4ab18f5_0.conda
aom                                   3.9.1         hac33072_0             2.6 MiB    conda  aom-3.9.1-hac33072_0.conda
assimp                                5.4.1         h8343317_0             3.3 MiB    conda  assimp-5.4.1-h8343317_0.conda
atk-1.0                               2.38.0        h04ea711_2             347.6 KiB  conda  atk-1.0-2.38.0-h04ea711_2.conda
attr                                  2.5.1         h166bdaf_1             69.4 KiB   conda  attr-2.5.1-h166bdaf_1.tar.bz2
aws-c-auth                            0.7.22        hf36ad8f_6             103.5 KiB  conda  aws-c-auth-0.7.22-hf36ad8f_6.conda
aws-c-cal                             0.6.15        h816f305_1             45.7 KiB   conda  aws-c-cal-0.6.15-h816f305_1.conda
aws-c-common                          0.9.23        h4ab18f5_0             230.1 KiB  conda  aws-c-common-0.9.23-h4ab18f5_0.conda
aws-c-compression                     0.2.18        he027950_7             18.8 KiB   conda  aws-c-compression-0.2.18-he027950_7.conda
aws-c-event-stream                    0.4.2         hb72ac1a_14            52.8 KiB   conda  aws-c-event-stream-0.4.2-hb72ac1a_14.conda
aws-c-http                            0.8.2         h75ac8c9_3             190.8 KiB  conda  aws-c-http-0.8.2-h75ac8c9_3.conda
aws-c-io                              0.14.9        hd3d3696_3             154.8 KiB  conda  aws-c-io-0.14.9-hd3d3696_3.conda
aws-c-mqtt                            0.10.4        hb0abfc5_7             159.6 KiB  conda  aws-c-mqtt-0.10.4-hb0abfc5_7.conda
aws-c-s3                              0.5.10        h44b787d_4             107.4 KiB  conda  aws-c-s3-0.5.10-h44b787d_4.conda
aws-c-sdkutils                        0.1.16        he027950_3             53.7 KiB   conda  aws-c-sdkutils-0.1.16-he027950_3.conda
aws-checksums                         0.1.18        he027950_7             49 KiB     conda  aws-checksums-0.1.18-he027950_7.conda
aws-crt-cpp                           0.26.12       he940a02_1             332.2 KiB  conda  aws-crt-cpp-0.26.12-he940a02_1.conda
aws-sdk-cpp                           1.11.329      h0f5bab0_6             3.4 MiB    conda  aws-sdk-cpp-1.11.329-h0f5bab0_6.conda
azure-core-cpp                        1.12.0        h830ed8b_0             335.3 KiB  conda  azure-core-cpp-1.12.0-h830ed8b_0.conda
azure-identity-cpp                    1.8.0         hdb0d106_1             193.1 KiB  conda  azure-identity-cpp-1.8.0-hdb0d106_1.conda
azure-storage-blobs-cpp               12.11.0       ha67cba7_1             509.8 KiB  conda  azure-storage-blobs-cpp-12.11.0-ha67cba7_1.conda
azure-storage-common-cpp              12.6.0        he3f277c_1             133.5 KiB  conda  azure-storage-common-cpp-12.6.0-he3f277c_1.conda
blosc                                 1.21.6        hef167b5_0             47.7 KiB   conda  blosc-1.21.6-hef167b5_0.conda
bullet-cpp                            3.25          hfb8ada1_2             40.4 MiB   conda  bullet-cpp-3.25-hfb8ada1_2.conda
bzip2                                 1.0.8         hd590300_5             248.3 KiB  conda  bzip2-1.0.8-hd590300_5.conda
c-ares                                1.28.1        hd590300_0             164.9 KiB  conda  c-ares-1.28.1-hd590300_0.conda
ca-certificates                       2024.6.2      hbcca054_0             152.4 KiB  conda  ca-certificates-2024.6.2-hbcca054_0.conda
cairo                                 1.18.0        hbb29018_2             961.2 KiB  conda  cairo-1.18.0-hbb29018_2.conda
cfitsio                               4.4.0         hbdc6101_1             892.9 KiB  conda  cfitsio-4.4.0-hbdc6101_1.conda
console_bridge                        1.0.2         h924138e_1             18 KiB     conda  console_bridge-1.0.2-h924138e_1.tar.bz2
cppzmq                                4.10.0        h2e2a08d_1             28.4 KiB   conda  cppzmq-4.10.0-h2e2a08d_1.conda
dartsim                               6.13.2        hdf3b901_3             13.7 MiB   conda  dartsim-6.13.2-hdf3b901_3.conda
dav1d                                 1.2.1         hd590300_0             742.4 KiB  conda  dav1d-1.2.1-hd590300_0.conda
dbus                                  1.13.6        h5008d03_3             604.1 KiB  conda  dbus-1.13.6-h5008d03_3.tar.bz2
double-conversion                     3.3.0         h59595ed_0             76.8 KiB   conda  double-conversion-3.3.0-h59595ed_0.conda
eigen                                 3.4.0         h00ab1b0_0             1 MiB      conda  eigen-3.4.0-h00ab1b0_0.conda
ergocub-software                      0.7.4         ha770c72_1             7.8 KiB    conda  ergocub-software-0.7.4-ha770c72_1.conda
expat                                 2.6.2         h59595ed_0             134.4 KiB  conda  expat-2.6.2-h59595ed_0.conda
[... cut ...]
ycm-cmake-modules                     0.16.4        hac33072_0             138.2 KiB  conda  ycm-cmake-modules-0.16.4-hac33072_0.conda
zeromq                                4.3.5         h75354e8_4             345 KiB    conda  zeromq-4.3.5-h75354e8_4.conda
zlib                                  1.3.1         h4ab18f5_1             90.8 KiB   conda  zlib-1.3.1-h4ab18f5_1.conda
zstd                                  1.5.6         ha6fb4c9_0             541.8 KiB  conda  zstd-1.5.6-ha6fb4c9_0.conda
zziplib                               0.13.69       he45264a_2             104.4 KiB  conda  zziplib-0.13.69-he45264a_2.conda
~~~

traversaro, 2024-06-26:
Diff between working and non-working:

~~~diff
--- <unnamed>
+++ <unnamed>
@@ -1,4 +1,4 @@
-traversaro@IITBMP014LW012:~/gazeboyarp$ cat working
+traversaro@IITBMP014LW012:~/gazeboyarp$ pixi list
 Package                               Version       Build                  Size       Kind   Source
 _libgcc_mutex                         0.1           conda_forge            2.5 KiB    conda  _libgcc_mutex-0.1-conda_forge.tar.bz2
 _openmp_mutex                         4.5           2_gnu                  23.1 KiB   conda  _openmp_mutex-4.5-2_gnu.tar.bz2
@@ -30,8 +30,8 @@
 bzip2                                 1.0.8         hd590300_5             248.3 KiB  conda  bzip2-1.0.8-hd590300_5.conda
 c-ares                                1.28.1        hd590300_0             164.9 KiB  conda  c-ares-1.28.1-hd590300_0.conda
 ca-certificates                       2024.6.2      hbcca054_0             152.4 KiB  conda  ca-certificates-2024.6.2-hbcca054_0.conda
-cairo                                 1.18.0        hbb29018_2             961.2 KiB  conda  cairo-1.18.0-hbb29018_2.conda
-cfitsio                               4.4.0         hbdc6101_1             892.9 KiB  conda  cfitsio-4.4.0-hbdc6101_1.conda
+cairo                                 1.18.0        h3faef2a_0             959.3 KiB  conda  cairo-1.18.0-h3faef2a_0.conda
+cfitsio                               4.4.1         hf8ad068_0             902.5 KiB  conda  cfitsio-4.4.1-hf8ad068_0.conda
 console_bridge                        1.0.2         h924138e_1             18 KiB     conda  console_bridge-1.0.2-h924138e_1.tar.bz2
 cppzmq                                4.10.0        h2e2a08d_1             28.4 KiB   conda  cppzmq-4.10.0-h2e2a08d_1.conda
 dartsim                               6.13.2        hdf3b901_3             13.7 MiB   conda  dartsim-6.13.2-hdf3b901_3.conda
@@ -41,7 +41,7 @@
 eigen                                 3.4.0         h00ab1b0_0             1 MiB      conda  eigen-3.4.0-h00ab1b0_0.conda
 expat                                 2.6.2         h59595ed_0             134.4 KiB  conda  expat-2.6.2-h59595ed_0.conda
 fcl                                   0.7.0         hadc09e8_4             1.5 MiB    conda  fcl-0.7.0-hadc09e8_4.conda
-ffmpeg                                7.0.1         gpl_h3589c0a_102       9.6 MiB    conda  ffmpeg-7.0.1-gpl_h3589c0a_102.conda
+ffmpeg                                6.1.1         gpl_he44c6f3_112       9.3 MiB    conda  ffmpeg-6.1.1-gpl_he44c6f3_112.conda
 flann                                 1.9.2         h2b5ea80_0             1.5 MiB    conda  flann-1.9.2-h2b5ea80_0.conda
 fmt                                   10.2.1        h00ab1b0_0             189.3 KiB  conda  fmt-10.2.1-h00ab1b0_0.conda
 font-ttf-dejavu-sans-mono             2.37          hab24e00_0             388.1 KiB  conda  font-ttf-dejavu-sans-mono-2.37-hab24e00_0.tar.bz2
@@ -51,12 +51,13 @@
 fontconfig                            2.14.2        h14ed4e7_0             265.6 KiB  conda  fontconfig-2.14.2-h14ed4e7_0.conda
 fonts-conda-ecosystem                 1             0                      3.6 KiB    conda  fonts-conda-ecosystem-1-0.tar.bz2
 fonts-conda-forge                     1             0                      4 KiB      conda  fonts-conda-forge-1-0.tar.bz2
-freeglut                              3.2.2         ha6d2627_3             140.6 KiB  conda  freeglut-3.2.2-ha6d2627_3.conda
+freeglut                              3.2.2         hac7e632_2             139.6 KiB  conda  freeglut-3.2.2-hac7e632_2.conda
 freeimage                             3.18.0        h4b96d29_20            450.6 KiB  conda  freeimage-3.18.0-h4b96d29_20.conda
 freetype                              2.12.1        h267a509_2             620.1 KiB  conda  freetype-2.12.1-h267a509_2.conda
 freexl                                2.0.0         h743c826_0             58.4 KiB   conda  freexl-2.0.0-h743c826_0.conda
 fribidi                               1.0.10        h36c2ea0_0             111.7 KiB  conda  fribidi-1.0.10-h36c2ea0_0.tar.bz2
-gazebo                                11.14.0       orignameh4988b5b_214   55.9 MiB   conda  gazebo-11.14.0-orignameh4988b5b_214.conda
+gazebo                                11.14.0       orignameh8358474_214   52.4 MiB   conda  gazebo-11.14.0-orignameh8358474_214.conda
+gazebo-yarp-plugins                   4.11.2        ha770c72_2             9.8 KiB    conda  gazebo-yarp-plugins-4.11.2-ha770c72_2.conda
 gdbm                                  1.18          h0a1914f_2             190.2 KiB  conda  gdbm-1.18-h0a1914f_2.tar.bz2
 gdk-pixbuf                            2.42.12       hb9ae30d_0             515.8 KiB  conda  gdk-pixbuf-2.42.12-hb9ae30d_0.conda
 geos                                  3.12.1        h59595ed_0             1.7 MiB    conda  geos-3.12.1-h59595ed_0.conda
@@ -70,8 +71,8 @@
 gnutls                                3.7.9         hb077bed_0             1.9 MiB    conda  gnutls-3.7.9-hb077bed_0.conda
 graphite2                             1.3.13        h59595ed_1003          94.6 KiB   conda  graphite2-1.3.13-h59595ed_1003.conda
 graphviz                              11.0.0        hc68bbd7_0             2.2 MiB    conda  graphviz-11.0.0-hc68bbd7_0.conda
-gst-plugins-base                      1.24.5        hbaaba92_0             2.7 MiB    conda  gst-plugins-base-1.24.5-hbaaba92_0.conda
-gstreamer                             1.24.5        haf2f30d_0             1.9 MiB    conda  gstreamer-1.24.5-haf2f30d_0.conda
[... cut ...]
+yarp-python                           3.9.0         py312h9b7a4f6_3        1.1 MiB    conda  yarp-python-3.9.0-py312h9b7a4f6_3.conda
 ycm-cmake-modules                     0.16.4        hac33072_0             138.2 KiB  conda  ycm-cmake-modules-0.16.4-hac33072_0.conda
 zeromq                                4.3.5         h75354e8_4             345 KiB    conda  zeromq-4.3.5-h75354e8_4.conda
 zlib                                  1.3.1         h4ab18f5_1             90.8 KiB   conda  zlib-1.3.1-h4ab18f5_1.conda
~~~

traversaro, 2024-06-26:
There are several key dependencies at different versions (`libxcb`, `libgdal`, `ffmpeg`, `qt-main`). Probably it is better to first make sure that gazebo-yarp-plugins can be installed with the latest version of all these dependencies before investigating more.

traversaro, 2024-06-26:
> There are several key dependencies at different versions (`libxcb`, `libgdal`, `ffmpeg`, `qt-main`). Probably it is better to first make sure that gazebo-yarp-plugins can be installed with the latest version of all these dependencies before investigating more.

These fails with:

~~~
traversaro@IITBMP014LW012:~$ conda create -n gazebo yarp gazebo-yarp-plugins libopencv==4.10.* libxcb==1.16 ffmpeg==7.*
Channels:
 - conda-forge
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package gazebo-yarp-plugins-4.6.0-ha770c72_2 requires libgazebo-yarp-plugins 4.6.0 h44e5fea_2, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ gazebo-yarp-plugins is installable with the potential options
│  ├─ gazebo-yarp-plugins [4.10.0|4.11.0|4.11.2|4.9.0] would require
│  │  └─ libgazebo-yarp-plugins [4.10.0 h107fe55_0|4.10.0 h89d8660_1|...|4.9.0 h492fe42_3], which requires
│  │     └─ libopencv >=4.9.0,<4.9.1.0a0  with the potential options
│  │        ├─ libopencv 4.9.0, which can be installed;
│  │        └─ libopencv 4.9.0 would require
│  │           └─ qt6-main [>=6.6.2,<6.7.0a0 |>=6.6.3,<6.7.0a0 ], which requires
│  │              └─ libxcb >=1.15,<1.16.0a0 , which can be installed;
│  ├─ gazebo-yarp-plugins [4.6.0|4.7.0|4.8.0] would require
│  │  ├─ libgazebo-yarp-plugins [4.6.0 h44e5fea_2|4.6.0 h44e5fea_3|4.7.0 h0c703cf_1|4.7.0 h86e38c8_0|4.8.0 h0c703cf_0], which requires
│  │  │  └─ libopencv >=4.7.0,<4.7.1.0a0 , which can be installed;
│  │  └─ libopencv >=4.7.0,<4.7.1.0a0 , which can be installed;
│  ├─ gazebo-yarp-plugins 4.8.0 would require
│  │  └─ libgazebo-yarp-plugins [4.8.0 hc9bcf84_1|4.8.0 hffe6568_2], which requires
│  │     └─ libopencv >=4.8.0,<4.8.1.0a0 , which can be installed;
│  └─ gazebo-yarp-plugins [4.8.0|4.9.0] would require
│     └─ libgazebo-yarp-plugins [4.8.0 h4494b4a_4|4.8.0 hb0c3a5d_5|...|4.9.0 hb0c3a5d_0], which requires
│        └─ libopencv >=4.8.1,<4.8.2.0a0 , which can be installed;
├─ libopencv 4.10**  is not installable because it conflicts with any installable versions previously reported;
└─ libxcb 1.16  is not installable because it conflicts with any installable versions previously reported.
~~~

traversaro, 2024-06-26:
So better solve https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/pull/28 before.

traversaro, 2024-06-26:
Reproduced the error without gyp via:

~~~
conda create -n gazeboolddeps gazebo ffmpeg==6.* libopencv==4.9.* libxcb==1.15.* libgdal==3.8.5
~~~

traversaro, 2024-06-26:
> Reproduced the error without gyp via:
> 
> ```
> conda create -n gazeboolddeps gazebo ffmpeg==6.* libopencv==4.9.* libxcb==1.15.* libgdal==3.8.5
> ```

Even better:

~~~
conda create -n gazebooldgdal gazebo libgdal==3.8.5
~~~

fails, while:

~~~
conda create -n gazebooldgdal gazebo libgdal==3.9.1
~~~

works, probably it is something `libxcb` related or libgdal related, the diff is:

~~~diff
--- <unnamed>
+++ <unnamed>
@@ -1,5 +1,5 @@
-(gazebogdalold) traversaro@IITBMP014LW012:~$ conda list
-# packages in environment at /home/traversaro/miniforge3/envs/gazebogdalold:
+(gazebogdalnew) traversaro@IITBMP014LW012:~$ conda list
+# packages in environment at /home/traversaro/miniforge3/envs/gazebogdalnew:
 #
 # Name                    Version                   Build  Channel
 _libgcc_mutex             0.1                 conda_forge    conda-forge
@@ -31,8 +31,8 @@
 bzip2                     1.0.8                hd590300_5    conda-forge
 c-ares                    1.28.1               hd590300_0    conda-forge
 ca-certificates           2024.6.2             hbcca054_0    conda-forge
-cairo                     1.18.0               h3faef2a_0    conda-forge
-cfitsio                   4.4.1                hf8ad068_0    conda-forge
+cairo                     1.18.0               hbb29018_2    conda-forge
+cfitsio                   4.4.0                hbdc6101_1    conda-forge
 console_bridge            1.0.2                h924138e_1    conda-forge
 cppzmq                    4.10.0               h2e2a08d_1    conda-forge
 dartsim                   6.13.2               hdf3b901_3    conda-forge
@@ -41,7 +41,7 @@
 eigen                     3.4.0                h00ab1b0_0    conda-forge
 expat                     2.6.2                h59595ed_0    conda-forge
 fcl                       0.7.0                hadc09e8_4    conda-forge
-ffmpeg                    7.0.1           gpl_ha6482d7_101    conda-forge
+ffmpeg                    7.0.1           gpl_h3589c0a_102    conda-forge
 flann                     1.9.2                h2b5ea80_0    conda-forge
 fmt                       10.2.1               h00ab1b0_0    conda-forge
 font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
@@ -51,12 +51,12 @@
 fontconfig                2.14.2               h14ed4e7_0    conda-forge
 fonts-conda-ecosystem     1                             0    conda-forge
 fonts-conda-forge         1                             0    conda-forge
-freeglut                  3.2.2                hac7e632_2    conda-forge
+freeglut                  3.2.2                ha6d2627_3    conda-forge
 freeimage                 3.18.0              h4b96d29_20    conda-forge
 freetype                  2.12.1               h267a509_2    conda-forge
 freexl                    2.0.0                h743c826_0    conda-forge
 fribidi                   1.0.10               h36c2ea0_0    conda-forge
-gazebo                    11.14.0         orignameh5ffa8c3_214    conda-forge
+gazebo                    11.14.0         orignameh4988b5b_214    conda-forge
 gdbm                      1.18                 h0a1914f_2    conda-forge
 gdk-pixbuf                2.42.12              hb9ae30d_0    conda-forge
 geos                      3.12.1               h59595ed_0    conda-forge
@@ -70,8 +70,8 @@
 gnutls                    3.7.9                hb077bed_0    conda-forge
 graphite2                 1.3.13            h59595ed_1003    conda-forge
 graphviz                  11.0.0               hc68bbd7_0    conda-forge
[... 57 lines cut ...]
 x264                      1!164.3095           h166bdaf_2    conda-forge
 x265                      3.5                  h924138e_3    conda-forge
-xcb-util                  0.4.0                hd590300_1    conda-forge
-xcb-util-image            0.4.0                h8ee46fc_1    conda-forge
-xcb-util-keysyms          0.4.0                h8ee46fc_1    conda-forge
-xcb-util-renderutil       0.3.9                hd590300_1    conda-forge
-xcb-util-wm               0.4.1                h8ee46fc_1    conda-forge
+xcb-util                  0.4.1                hb711507_2    conda-forge
+xcb-util-image            0.4.0                hb711507_2    conda-forge
+xcb-util-keysyms          0.4.1                hb711507_0    conda-forge
+xcb-util-renderutil       0.3.10               hb711507_0    conda-forge
+xcb-util-wm               0.4.2                hb711507_0    conda-forge
 xerces-c                  3.2.5                hac6953d_0    conda-forge
 xkeyboard-config          2.42                 h4ab18f5_0    conda-forge
 xorg-fixesproto           5.0               h7f98852_1002    conda-forge
@@ -273,7 +275,7 @@
 xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
 xorg-libice               1.1.1                hd590300_0    conda-forge
 xorg-libsm                1.2.4                h7391055_0    conda-forge
-xorg-libx11               1.8.9                h8ee46fc_0    conda-forge
+xorg-libx11               1.8.9                hb711507_1    conda-forge
 xorg-libxau               1.0.11               hd590300_0    conda-forge
 xorg-libxaw               1.0.14               h7f98852_1    conda-forge
 xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
~~~

traversaro, 2024-06-27:
Ok, after https://github.com/conda-forge/gazebo-yarp-plugins-feedstock/pull/28/ `conda create -n test gazebo gazebo-yarp-plugins` is working, because there exists of a build of gazebo-yarp-plugins compatible with libxcb==1.16 . 

However, `conda create -n test gazebo libxcb==1.15` still fails. Part of the problem (or at least, something that makes the analysis more complex) is that apparently since https://github.com/conda-forge/gazebo-feedstock/pull/208 the rebuild did not bumped the build number, and I checked the tick without checking if it actually bumped it, so we have an endless number of `_14` builds that actually refer to different configurations.

traversaro, 2024-08-14:
If you are trying to install gazebo with some robotology package and you end up with the problematic 214 build, apparently the 213 builds seems to work fine:

~~~
conda create -n loremorework -c conda-forge -c robotology gazebo=*=*_213 yarp icub-main gazebo-yarp-plugins icub-models whole-body-estimators robots-configuration idyntree-yarp-tools
~~~

fyi @LoreMoretti @xela-95

Tobias-Fischer, 2026-01-18:
Is this still relevant @traversaro?